john30/ebusd-configuration #209: VRC 720f - Temperature Setpoint

splashmak, 2021-01-18:
Hello,
I did the exact same thing with the 720f regulator and even if a lot of things are ok, everything is not working :

The main issues for me are related to the temperature setpoint modification.
To achieve that I see 2 posibilities : (1) set the regulator to manual mode & (2) automatic mode "veto" temp command

1. Ok to set the mode with **z1OpMode** parameter but can't modify the **z1ActualRoomTempDesired**
2. Can not modify the temperature mode **s1SFMode** until I make a physical change on the 720f. After that even if I cancel the action it's accessible for some hours and everything work like a charm( I don't know how lang precisely yet). Difference is that when manually (with regulator action) doing it I have to set the action duration which is not done when using ebusd

By the way the **z1DayTemp** is not accessible when the  **z1NightTemp** is. I check the first memory zones to find it but yeah ... maybe you have a trick to find it ?

[readall_result.txt](https://github.com/john30/ebusd-configuration/files/5826642/readall_result.txt)

_Originally posted by @splashmak in https://github.com/john30/ebusd-configuration/issues/196#issuecomment-761864812_

paddenstoeltje, 2022-11-28:
@splashmak did you ever solved the issue. I'm running into the exact same problem as you describe, see post [234](https://github.com/john30/ebusd-configuration/issues/234).

splashmak, 2022-11-30:
@paddenstoeltje  unfortunately not. I tried it years ago on a raspberry pi but SD card crashed and I had no backup. So I retried recently and I have seen that the new configuration file is now added so I'm using it with the url link ...without all the functionnalities. It's a mess because at the beginning I didn't wanted this new model of VRC ... So I'm just waiting/hoping a miracle on this thread

daclerk, 2022-12-16:
Hoi @paddenstoeltje and @splashmak, have a look to post #234 , there I posted a solution for my cabled VRC720. Would be curious if this also works for your wireless VRC720f.

ppnacho, 2023-08-31:
I have a VRC720f module controller and with ebusctl i don't appear in the answer console.
I only can see the next items,
address 03: master https://github.com/john30/ebusd-configuration/pull/11
address 08: slave https://github.com/john30/ebusd-configuration/pull/11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu00.full.csv"
address 10: master https://github.com/john30/ebusd-configuration/pull/2
address 15: slave https://github.com/john30/ebusd-configuration/pull/2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"
address 31: master https://github.com/john30/ebusd-configuration/issues/8, ebusd
address 36: slave https://github.com/john30/ebusd-configuration/issues/8, ebusd
address 71: master https://github.com/john30/ebusd-configuration/issues/9
address 76: slave https://github.com/john30/ebusd-configuration/issues/9, scanned "MF=Vaillant;ID=VWZIO;SW=0607;HW=5103"

What .csv files must link to slave https://github.com/john30/ebusd-configuration/pull/2 and https://github.com/john30/ebusd-configuration/issues/9?