john30/ebusd-configuration #212: ELCO Aerotop Split

ck-coke, 2021-02-05:
Hi,

I'm thinking about buying an ebus adapter for my Elco Aerotop Split Heat Pump. How high is the chance that I can scan and find out all codes to integrate the heatpump later on in Iobroker? Does it make sense to buy an adapter or not?

john30, 2021-02-20:
to be honest: the chances are almost zero to get matching configuration files for this device immediately.
However, with an [adapter](https://adapter.ebusd.eu/) you can sniff into the eBUS traffic and if all goes well, the device might just use the standard messages from the eBUS PDFs. In this case, it is rather easy to add the definitions. Otherwise you'd have to find the meaning of the grabbed messages one by one.

wrongisthenewright, 2022-05-23:
Hi CK-Coke,
I'm working on a configutation settings for a similar machine. I've got an Ariston Nimbus 70/m, from what I've seen online this is a similar machine if not the same with another brand, I'll let you know how it works...