john30/ebusd-configuration #213: please help for Vailant Solar system

christian-pb, 2021-02-11:
I can not see the solar values.
Operating time solar pump, total sum in kWh and operating time solar time.
Can you please help? I have the following output:

ebusctl scan result
15;Vaillant;UI   ;0324;6201;;;;;;;
23;Vaillant;SOLSY;0306;6301
25;Vaillant;SOLSY;0306;6301
26;Vaillant;SOLSY;0306;6301
44;Vaillant;SOLSY;0306;6301
50;Vaillant;SOLSY;0306;6301
ec;Vaillant;SOLSY;0306;6301

ebusctl info
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0324;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 3f: master #23
address 44: slave #23, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301"
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/ec.solsy.sc.csv"

ebusctl grab result
1025b504010b / 0400030000 = 1
1025b5040117 / 0101 = 5200
1026b5040101 / 09140200000580000100 = 2548
1026b504010d / 050000008000 = 1040
1026b5090118 / 0a00000000000000000000 = 869
1044b5070104 / 0100 = 1581
10ecb5040101 / 09000300000007000100 = 1040
10ecb504010d / 050000008000 = 1040
10ecb504010f / 0a40040b02200321ff0000 = 2807
10ecb5040111 / 06550703550703 = 2650
10ecb5040112 / 0ae4ff0012130080000000 = 2807
10ecb5040113 / 09150f003c0000010103 = 1
1026b509030d7c00 / 0100 = 1575
10ecb50903298100 / 03810001 = 1
10ecb50903298200 / 0382008c = 1
10ecb50903298300 / 03830000 = 2804
1023b505072b0f0100000000 / 00 = 1040
1025b505072b0f0100000000 / 00 = 1040
1026b505072b0f0100000000 / 00 = 1040
103f050105aa1919fa00 = 1035
103f050105aa2626fa00 = 941
103f050105aa1919fb00 = 875
103f050105aa2525fb00 = 9
103f050105aa2626fb00 = 344
103f050105aa2626f900 = 133
103f050105aa2424fe00 = 88
103f050105aa2525fe00 = 7
103f050105aa2323ff00 = 12
103f050105aa2424ff00 = 90
103f050105aa2525fc00 = 158
103f050105aa2626fc00 = 21
103f050105aa2424fd00 = 6
103f050105aa2525fd00 = 106
103f050105aa1d1d0a00 = 24
103f050105aa1c1c0b00 = 32
103f050105aa1d1d0b00 = 3
103f050105aa1d1d0800 = 1
103f050105aa1e1e0800 = 29
103f050105aa1d1d0900 = 37
103f050105aa1e1e0900 = 2
103f050105aa1b1b0e00 = 53
103f050105aa1a1a0f00 = 64
103f050105aa1b1b0f00 = 4
103f050105aa1b1b0c00 = 2
103f050105aa1c1c0c00 = 38
103f050105aa1b1b0d00 = 36
103f050105aa21210200 = 10
103f050105aa21210300 = 10
103f050105aa21210000 = 8
103f050105aa23230000 = 152
103f050105aa24240000 = 6
103f050105aa21210100 = 83
103f050105aa23230100 = 6
103f050105aa1e1e0600 = 2
103f050105aa1f1f0600 = 25
103f050105aa1e1e0700 = 27
103f050105aa21210400 = 7
103f050105550000fa00 = 315
103f050105aa1f1f0500 = 11
103f050105aa21210500 = 2
103f050105aa19191200 = 11
103f050105aa1a1a1000 = 76
103f050105aa19191100 = 134
103f050105aa1a1a1100 = 40
103f050105000000fa00 = 124
103fb509040e7d0000 = 5198
1050b505072b0f0100000000 / 00 = 1040
10ecb505072b0f0100000000 / 00 = 1040

pagaille, 2021-03-05:
What's kind of solar system is it ?

christian-pb, 2021-03-05:
Hi, it is the "auroMATIC 620" from Vailant as the main Control unit. The unit controlls the solar pump, temperature sensors, etc. 
The gas burner is a "ecovit vkk" from Vailant.

I found out, that B5h 04h - Block 12h is Solar1. Byte 6 and 7 have to be "Collector array 1 run-time solar circuit pump".

I am sorry, but I have no idea, how I can implement this in code. 
If you can help me, it would be really really great.
Thank you very much in advance. No you need further informations?

Best regards

pagaille, 2021-03-05:
That system looks supported. https://github.com/john30/ebusd-configuration/blob/0fe1b4cd1767d41fde457ad06f470298ae4d0cfa/ebusd-1.x.x/vaillant_de/solsy.mc.csv

Did you try `ebusctl read StatPowerOn ` for example ?

christian-pb, 2021-03-05:
No I did not try it:
I tired:
-f --configpath http://ebusd.eu/config/ --scanconfig=full -d tcp:192.168.1.11:5000 --latency=80 -p 8890 --httpport 8889
for online configs

and 
-f --configpath=/etc/ebusd --scanconfig=full -d tcp:192.168.1.11:5000 --latency=80 -p 8890 --httpport 8889 --loglevel=info

for manuall configs 

(I use the "eBus Koppler Ethernet" with the IP 192.168.1.11)
Where I have to use "ebusctl read StatPowerOn" ?

pagaille, 2021-03-05:
At first sight your config is working. The devices are recognized. Using the ebusctl you can query the various registers of those devices. Please read the wiki to get more information as to how use the tools.