john30/ebusd-configuration #216: [HOWTO] Wolf CGB-11, SM-1 and eBUs adapter 3

ngblume, 2021-03-28:
Hey everyone,

since I recently started working with eBus and the Wolf heating systems, I would like to post my findings / how far I got and how I got there (for everyone looking to get started).

**System:**
- Wolf CGB-11
- Wolf SM-1

**eBus-Adapter and tools**
- John30's eBus adapter V3 (https://adapter.ebusd.eu/, RPi-version)
- RPi 4 (connected via wireless to router)
- ebusd running on RPi (standard installation from repository; see below)

**Status**
1. receiving messages from system
2. (most) messages properly decoded
3. missing solar

### HowTo (get there)
- Expertise required: **Beginner with some linux command line comfort** 
- Duration: **20 min excluding download times for image and updates**
1. Setup RPi with "Raspberry Pi OS Lite" via balena Etcher (https://www.raspberrypi.org/software/operating-systems/)
2. Eject and reinsert SD card on your computer (the one used for transfering OS to SD card
2. Place wpa_supplicant.conf in root folder of "boot" containing the following info (edited for your specifics, ensure UTF-8 encoding in your editor, i.e. Notepad++ under "Encoding" > "UTF-8"):
```
country=DE
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
ap_scan=1
network={
 scan_ssid=1
 ssid="YOURSSID"
 psk="YOURPASSWD"
key_mgmt=WPA-PSK
}
```
3. Place completely empty file with the filename "ssh" (NO extension) in root folder of "boot" drive as well
4. Eject SD-card and insert into RPi
5. Start RPi without eBUs adapter V3 attached (might be unnecessary)
6. Connect to it via SSH using PuTTY
7. Execute neccessary steps as described by @john30 here: [https://adapter.ebusd.eu/raspberrypi.en](https://adapter.ebusd.eu/raspberrypi.en)
8. Restart RPi as mentioned
9. Login via SSH
10. Run "sudo apt-get update"
`sudo apt-get update`
11. Run "sudo apt-get upgrade"
`sudo apt-get upgrade`
12. Reboot
13. Install **ebusd** from repository as described by @john30 here: https://github.com/john30/ebusd-debian/blob/master/README.md (if MQTT not required, use different LIST file; don't execute cmd without looking at next lines in readme ;-) )
14. Reboot RPi
15. Change eBusd config as follows using "sudo nano /etc/default/ebusd"
`sudo nano /etc/default/ebusd`
DO NOT use "scanconfig" since the Wolf systems apparrently do not adhere to the specification in this context, at least the CGB-11 does not, see here: https://github.com/john30/ebusd-configuration/issues/71#issuecomment-557872344)
```
# /etc/default/ebusd:
# config file for ebusd service.

# Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="-c /etc/ebusd/ -d enh:/dev/ttyAMA0 --latency=50 --enablehex"
```
16. Change to your home directory via "cd /home/pi/"
`cd /home/pi/`
17. Create folder "ebus-configs" using "mkdir ebus-configs"
`mkdir ebus-configs`
18. Change to newly created folder via "cd ebus-configs"
`cd ebus-configs`
19. Clone GIT repository into this path using "git clone https://github.com/john30/ebusd-configuration.git"
`git clone https://github.com/john30/ebusd-configuration.git`
20. Create folder "/etc/ebusd/" using "sudo mkdir /etc/ebusd/"
`sudo mkdir /etc/ebusd/`
[... 192 lines cut ...]
2021-03-28 13:41:16.091 [update notice] received unknown MM cmd: 30715023094c660000005d010000
2021-03-28 13:41:16.417 [update notice] received unknown MM cmd: 3071502309b01300f4015d010000
2021-03-28 13:41:16.746 [update notice] received unknown MM cmd: 3071502309b8720000005d010000
2021-03-28 13:41:17.049 [update notice] received write feuerung betrd QQ=10: Brauchwasser_Heizen;6;31.38;-;-;45.0;-
2021-03-28 13:41:17.338 [update notice] received update-read broadcast betrd QQ=f1: 1;00;-;36.0;-;53;15
2021-03-28 13:41:18.573 [update notice] received unknown BC cmd: 71fe50230908620900005d010000
>> unknown BC > SM-1
```
As per this link somewhen / somehow this ussed to work with these config files: https://forum.fhem.de/index.php?topic=64039.0
But when I try "ebusctl r solar kollektortemp" or "ebusctl r solar wwsolartemp" or "ebusctl r solar pumpe", I get an error:
```
pi@ebus-rpi:/etc/ebusd $ ebusctl r solar kollektortemp
ERR: element not found
```
Could anyone give me a pointer as to how to get those values decoded as well again?
I don't think I need to do "Creating new message definition files" (https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files), since the definitions are there, but not working properly... but I might be wrong..

And second question:
These values (or some of them) should go into a database for logging and plotting... (Grafana und InfluxDB)
Is the best way to poll for data from the ebus daemon at specific intervals or can I register to an event that triggers once a broadcast is received (for all data broadcasted)? And is there any work in this direction in Python that someone is aware of?

Thanks !

Cheers
Niels Göran

bdariusb, 2021-10-01:
is one able to change the field names in the config files?

john30, 2021-10-03:
sure. each field may have a name associated with it in the columns titles "field..."

ngblume, 2021-10-04:
Any response to the question, why no solar data shows up?

bdariusb, 2021-10-04:
you don't have the fields defined that you are querying, e.g. `ebusctl r solar kollektortemp` 
Basically all you can query is listed in `ebusctl f -F name,comment`
> 
> 
> Any response to the question, why no solar data shows up?

bdariusb, 2021-10-04:
you can define a field with read command but instead of field type `r` use `r[1-9]`, e.g. `r3`. This will poll the field values with priority 3 (more on that in the docs) 
> And second question:

teeceeell, 2022-02-05:
I get stuck with:
ebusct scan full

It's getting stuck with: 
1 scan(s) still running

ebusctl i brings this:
root@loxberry:/etc/ebusd# ebusctl i
version: ebusd 21.3.v21.3
device: /dev/ttyAMA0, enhanced
signal: acquired
symbol rate: 48
max symbol rate: 189
reconnects: 0
masters: 1
messages: 56
conditional: 0
poll: 0
update: 6
address 31: master #8, ebusd
address 36: slave #8, ebusd

Any ideas?

MacvonUnruh, 2024-02-12:
Hallo,
ich habe eine neues Bedienmodul an meine Mehrfamilienhaus-Gastherme (Wolf) bekommen und versuche dieses über den ebus auszulesen.
Habe ein Raspi-Zero2 mit eBUS Adapter Shield v5
Anschluss und Konfiguration passt soweit. Ich bekomme auch Daten, kann sie nur nicht "Identifizieren"

`2024-02-12` 02:15:45.690 [bus notice] device status: transport opened
2024-02-12 02:15:45.748 [main notice] ebusd 23.3.23.3 started on device: /dev/ttyAMA0, serial high speed, enhanced
2024-02-12 02:15:45.784 [main error] error reading config files from /etc/ebusd/: ERR: element not found, last error: 08.csv:3: ERR: element not found, field type TEMP in field 2
2024-02-12 02:15:45.785 [bus notice] bus started with own address 31/36
2024-02-12 02:15:45.785 [bus error] device status: missing enhanced byte 2
2024-02-12 02:15:45.785 [bus error] device status: unexpected enhanced byte 2
2024-02-12 02:15:45.785 [bus notice] signal acquired
2024-02-12 02:15:45.857 [bus notice] device status: reset, supports info
2024-02-12 02:15:45.920 [bus notice] device status: extra info: firmware 8.1[4106].1[4106], jumpers 0x03
2024-02-12 02:15:46.959 [bus notice] new master 10, master count 2
2024-02-12 02:15:46.976 [bus notice] new master 03, master count 3
2024-02-12 02:15:46.976 [update notice] received unknown MM cmd: 1003080008331900088000000a
2024-02-12 02:15:47.291 [bus notice] new master f1, master count 4
2024-02-12 02:15:47.291 [update notice] received unknown BC cmd: f1fe080008331900080000000a
2024-02-12 02:15:47.657 [update notice] received unknown BC cmd: f1fe50230990d20100005d010000
2024-02-12 02:15:48.024 [update notice] received unknown BC cmd: f1fe502309f07e0101005d010000
2024-02-12 02:15:48.925 [update notice] received unknown BC cmd: 03fe050308010058646ad82d08
2024-02-12 02:15:51.933 [update notice] received unknown MM cmd: 1003050709aa0693010080ff14ff
2024-02-12 02:15:52.249 [update notice] received unknown BC cmd: f1fe050308010110ff6aff2d08
2024-02-12 02:15:55.785 [main notice] found messages: 0 (0 conditional on 0 conditions, 0 poll, 0 update)
2024-02-12 02:15:56.887 [update notice] received unknown MM cmd: 1003080008331900088000000a
2024-02-12 02:15:57.202 [update notice] received unknown BC cmd: f1fe080008331900080000000a
2024-02-12 02:15:57.568 [update notice] received unknown BC cmd: f1fe50230990d20100005d010000
2024-02-12 02:15:57.935 [update notice] received unknown BC cmd: f1fe502309f07e0101005d010000
2024-02-12 02:15:58.846 [bus notice] new master 30, master count 5
2024-02-12 02:15:58.846 [update notice] received unknown BC cmd: 30fe070009008010170212020124
2024-02-12 02:15:59.060 [update notice] received unknown MM cmd: 03f108000800324d080002000a
2024-02-12 02:16:01.825 [update notice] received unknown MM cmd: 1003050709aa0193010080ff14ff
2024-02-12 02:16:02.140 [update notice] received unknown BC cmd: f1fe050308010110ff6aff2d08
2024-02-12 02:16:06.779 [update notice] received unknown MM cmd: 1003080008331900088000000a
2024-02-12 02:16:07.094 [update notice] received unknown BC cmd: f1fe080008331900080000000a
2024-02-12 02:16:07.460 [update notice] received unknown BC cmd: f1fe50230990d20100005d010000
2024-02-12 02:16:07.826 [update notice] received unknown BC cmd: f1fe502309f07e0101005d010000
2024-02-12 02:16:08.678 [update notice] received unknown BC cmd: 03fe050308010058646dd82d08
2024-02-12 02:16:11.736 [update notice] received unknown MM cmd: 1003050709aa0393010080ff14ff``

` ebusctl i
version: ebusd 23.3.23.3
update check: OK
device: /dev/ttyAMA0, serial high speed, enhanced, firmware 8.1[4106].1[4106]
access: *
signal: acquired
symbol rate: 34
max symbol rate: 244
min arbitration micros: 4
max arbitration micros: 18
min symbol latency: 4
max symbol latency: 9
reconnects: 0
masters: 7
messages: 6
conditional: 0
poll: 0
update: 0
address 03: master #11
address 07: master #16
address 08: slave #11, scanned "MF=Kromschroeder;ID=  ;SW=0230;HW=-"
address 0c: slave #16, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 10: master #2
address 15: slave #2, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 36: slave #8, ebusd, scanning
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address f1: master #10
address f6: slave #10, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
`
` ebusctl f -F name,comment
,
,
,
,
,
,

`

leonundjulie, 2024-02-13:
Hi. Great description! I followed the way defined in the opening and in consequence I am able to read the fires from my WOLF gas heating system, however, just in task 22 it needs a small correction. The correct command is cp **-r** /home/pi/ebus-configs/ebusd-configuration/ebusd-2.x.x/de/* /etc/ebusd/  … the “-r” was missing.

Next task: to get the data via MQTT into my iOBroker.

Dr-Gigavolt, 2024-10-06:
Hello guys,

I have a CGB-11 too which was running for years without any communication with wall thermostats. Now I would like to modernize the old hardware and make it running with Home Assistant. The target is to set warm water temperature and mixer set point by Home Assistant. Or at least to switch between day and night set points that are adjusted by the BM + control the circulating pump.

I managed to set up the ebusd, thanks John for the software and Niels for the detailed instructions. I can read some values, but writing does not work. I made a simple trial with the HG01 parameter.

reading works:
_localhost: r hg01
20.0_

writing is accepted:
_localhost: w -c feuerung hg01 21
done_

but the value is not changed:
_localhost: r hg01
20.0_

Is my write command, or the line in 08.csv wrong?
_w,feuerung,hg01,HG 01 - Hysterese Vorlauftemperatur,,03,5023,002200,hysterese_vorlauf,,UIN,10,temp,Hysterese Vorlauftemperatur (5-30),suffix,,HEX:4,=9d010000,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,_

I tried to change parameter by the BM and got the following log entry:

_2024-10-07 00:02:21.431 [update notice] received unknown MS cmd: 0708502203423e02 / 025dd1
2024-10-07 00:02:21.730 [update notice] received read feuerung hg01 QQ=07: 20.0
2024-10-07 00:02:22.040 [update notice] received write feuerung betrd QQ=10: aus;Kesselpumpeaus;5.00;-;-;10.0;-
2024-10-07 00:02:22.378 [update notice] received update-read broadcast betrd QQ=f1: 0;00;-;26.0;-;47;14
2024-10-07 00:02:22.719 [update notice] received write feuerung sollw QQ=10: 5.000;14.000;-;50;10.000
2024-10-07 00:02:23.014 [update notice] received update-read broadcast sollw QQ=f1: 5.000;13.898;0;40;10.000
**2024-10-07 00:02:23.363 [update notice] received unknown MS cmd: 3008502203658601 / 020080
2024-10-07 00:02:23.712 [update notice] received unknown MM cmd: 0703502309c42200d2005d010000
2024-10-07 00:02:23.944 [update notice] received unknown MM cmd: 030750020100
2024-10-07 00:02:24.428 [update notice] received unknown MS cmd: 70515022032af402 / 020080
2024-10-07 00:02:24.805 [update notice] received unknown MS cmd: 7051501407500005001a0014 / 090040b31600d80a640a**
2024-10-07 00:02:25.421 [update notice] received read feuerung hg01 QQ=07: 21.0_

Some of the unknown commands was the write operation. Do you have an idea how to fix it?

Thanks a lot.

Best regards,
Anton