john30/ebusd-configuration #220: Desired temperature overwritten by the heater

marcinwadon, 2021-04-15:
Hello :)

I own a Saunier Duval Thelia Condens heater and Exacontrol E7C room controller.
That room controller has a really slow reaction to temp change. I decided to simulate it using ebusd - this is how I started.

I took a look into the grab results and logs and I have figured out that the room controller frequently reads: `Status01` and `Status02` as well as it sends `SetMode` and one unknown message. I used configuration from one of the Vaillant heaters which work 1:1 with my heater.

I assume `Status01` and `Status02` are read-only to get info about the heater state - am I right?
I wasn't able to write the `SetMode` command, because it's update-write but because I decided to simulate the room controller and I will turn it off, I wrote a `SetModeOverride` and I'm using automation to re-send the desired temperatures every minute. It works pretty well - I can see changes of desired flow temperature, as well as storage desired temperature.

The second command which is unknown to me is `1008b5070108 / 026400`. Response `026400` can be decoded as `64.00` but I have totally no idea what it is. I see that the e7c controller sends it exactly every minute and the response is always the same.

What's more - I'm not able to scan the room controller in the ebus, I receive ACK error every time. (but I don't need this).

**And now my issue** - everything is working really great, except one thing. I'm able to set all parameters, max temperatures, and so on. I read the `ModulationTempDesired` parameter which is a percentage power of a heater and it's always 20%.
When the room controller is connected to the ebus, the `ModulationTempDesired` changes (~50-60%, then 20%, and so on). When the room controller is disconnected, the modulation is always set to 20%.

I tried to change `ModulationTempDesired` from read-only to write command and write a percentage modulation to the heater. I succeeded when the flame was off. The modulation changed according to my command. When the flame is on, modulation is again always 20% and I cannot override it.

What I'm missing? I thought that the heater will modulate the flame according to the desired flow temperature but looks it's a wrong assumption.

 There is nothing else being sent between the heater and room controller (according to grab results).

Thanks in advance :) When I will fix this issue I'm going to make a pull request with all my changes for future use.

marcinwadon, 2021-04-16:
I have figured out that the heater will not detect `ebusd` as an external controller till the first `SetMode` command sent.
When I power on the heater I am able to select desired water and flow temperatures on the heater itself, the heater uses modulation properly then. When I send `SetMode` command, the heater detects is as an ebus controller and automatically blocks controlling the temperature from the heater itself, allowing only to set it up via that ebus.

Is there any way to write a desired flow temperature other than `SetMode` command?
I took a look into `FlowTempDesired` and `FlowSetPotmeter` parameters and both are not working in the writable mode.

wizu, 2021-09-03:
@marcinwadon have you dealt with this problem?

marcinwadon, 2021-09-03:
@wizu hey, still no idea.

wizu, 2021-09-06:
@marcinwadon have you seen this https://github.com/john30/ebusd-configuration/issues/170?

marcinwadon, 2021-09-06:
Oh, thank you @wizu, I didn't see that. Probably because "modulation" isn't mentioned anywhere there, but they are writing about "stagnation" and very slow react on the desired flow temp, so that must be the same behavior.
I will check the solution today or tomorrow.

marcinwadon, 2021-09-23:
Hey @wizu - I can confirm it works, thank you very much :)

closed 2021-09-23 by marcinwadon