john30/ebusd-configuration #221: Help support Protherm 23MOV

itProfi, 2021-06-20:
I have Protherm 23MOV  https://fixbroken.ru/oshibka-f20-na-kotle-proterm-gepard/  
https://teplopoint.ru/kotly/kotel-protherm-gepard-23-mov  with EBUS support
Also, I have Thermolink P thermostat  - https://tdkomfort.ru/shop/komnatnyi-regulyator-protherm-thermolink-p-4529.html
I connect to EBUS via Orange PI Zero board extender - https://www.ebay.com/itm/274313895394?hash=item3fde61dde2:g:zF4AAOSwlBtedkgT
EBUD version 21.2
When I connect ,I see: 
````
ebusd -f -d /dev/ttyUSB1 --configpath=/etc/ebusd/ --scanconfig=08 --loglevel=debug --httpport=8889
2021-06-20 09:40:04.192 [main notice] ebusd 21.2.v21.2 started with single scan on device /dev/ttyUSB1
2021-06-20 09:40:04.193 [main info] loading configuration files from /etc/ebusd/
2021-06-20 09:40:04.193 [main info] reading templates /
2021-06-20 09:40:04.196 [main info] read templates in /
2021-06-20 09:40:04.196 [main info] reading file broadcast.csv
2021-06-20 09:40:04.199 [main info] successfully read file broadcast.csv
2021-06-20 09:40:04.199 [main info] reading file memory.csv
2021-06-20 09:40:04.201 [main info] successfully read file memory.csv
2021-06-20 09:40:04.201 [main info] read config files
2021-06-20 09:40:04.217 [bus notice] bus started with own address 31/36
2021-06-20 09:40:04.218 [main info] registering data handlers
2021-06-20 09:40:04.218 [main info] registered data handlers
2021-06-20 09:40:04.247 [bus debug] ERR: SYN received during no signal, switching to ready
2021-06-20 09:40:04.247 [bus notice] signal acquired
2021-06-20 09:40:05.139 [bus notice] new master 10, master count 2
2021-06-20 09:40:05.205 [bus notice] new master 03, master count 3
2021-06-20 09:40:05.206 [update info] received MS cmd: 1008b5040100 / 0a00ffffffffffffff0080
2021-06-20 09:40:05.206 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffff0080
2021-06-20 09:40:05.459 [update info] received MS cmd: 1008b5110102 / 05004c964c74
2021-06-20 09:40:05.460 [update notice] received unknown MS cmd: 1008b5110102 / 05004c964c74
2021-06-20 09:40:05.709 [update info] received MS cmd: 1008b51302050b / 0101
2021-06-20 09:40:05.710 [update notice] received unknown MS cmd: 1008b51302050b / 0101
2021-06-20 09:40:08.267 [update info] received MS cmd: 1008b510090000146cffff05ff00 / 0101
2021-06-20 09:40:08.268 [update notice] received unknown MS cmd: 1008b510090000146cffff05ff00 / 0101
2021-06-20 09:40:13.184 [update info] received MS cmd: 1008b5110101 / 0979770080ffff0000ff
2021-06-20 09:40:13.185 [update notice] received unknown MS cmd: 1008b5110101 / 0979770080ffff0000ff
2021-06-20 09:40:14.218 [main debug] performing regular tasks
2021-06-20 09:40:14.219 [main notice] starting initial scan for 08
2021-06-20 09:40:14.219 [bus info] scan 08 cmd: 3108070400
2021-06-20 09:40:14.222 [bus debug] start request 31
2021-06-20 09:40:14.222 [bus debug] arbitration start with 31
2021-06-20 09:40:18.097 [bus debug] arbitration lost
2021-06-20 09:40:18.097 [bus debug] ERR: arbitration lost during ready, retry
2021-06-20 09:40:18.191 [update info] received MS cmd: 1008b510090000146cffff05ff00 / 0101
2021-06-20 09:40:18.191 [update notice] received unknown MS cmd: 1008b510090000146cffff05ff00 / 0101
2021-06-20 09:40:18.192 [bus debug] start request 31
2021-06-20 09:40:18.192 [bus debug] arbitration start with 31
2021-06-20 09:40:23.109 [bus debug] arbitration lost
2021-06-20 09:40:23.109 [bus debug] ERR: arbitration lost during ready, retry
2021-06-20 09:40:23.197 [update info] received MS cmd: 1008b5110101 / 0978770080ffff0000ff
2021-06-20 09:40:23.197 [update notice] received unknown MS cmd: 1008b5110101 / 0978770080ffff0000ff
2021-06-20 09:40:23.198 [bus debug] start request 31
2021-06-20 09:40:23.198 [bus debug] arbitration start with 31

root@orangepizero:~# ebusctl i
version: ebusd 21.2.v21.2
signal: acquired
symbol rate: 23
max symbol rate: 42
reconnects: 0
masters: 3
messages: 12
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
````
How I can support my Prtotherm?

john30, 2021-11-30:
as the PB byte seems to be b5 (=Vaillant), you could just try one of the 08.*.csvs.
what is the scan result? depending on the adapter used, you might have to add "--latency=50" or so to the ebusd start params

itProfi, 2023-08-26:
I decide back to my boiler )
Same problem:
`
zerolts@orangepizero:~$ ebusd -f -d /dev/ttyUSB1 --configpath=/etc/ebusd/ --scanconfig=08 --loglevel=debug --latency=50
2023-08-26 18:13:02.042 [main notice] ebusd 23.2.23.2 started with single scan on device: /dev/ttyUSB1
2023-08-26 18:13:02.043 [main info] loading configuration files from /etc/ebusd/
2023-08-26 18:13:02.043 [main debug] reading directory /etc/ebusd/
2023-08-26 18:13:02.043 [main debug] file type of /etc/ebusd/mqtt-integration.cfg is file
2023-08-26 18:13:02.043 [main debug] file type of /etc/ebusd/memory.csv is file
2023-08-26 18:13:02.044 [main debug] file type of /etc/ebusd/_templates.csv is file
2023-08-26 18:13:02.044 [main debug] file type of /etc/ebusd/vaillant is dir
2023-08-26 18:13:02.044 [main debug] file type of /etc/ebusd/broadcast.csv is file
2023-08-26 18:13:02.044 [main info] reading templates /
2023-08-26 18:13:02.049 [main info] read templates in /
2023-08-26 18:13:02.049 [main info] reading file memory.csv
2023-08-26 18:13:02.052 [main info] successfully read file memory.csv
2023-08-26 18:13:02.052 [main info] reading file broadcast.csv
2023-08-26 18:13:02.056 [main info] successfully read file broadcast.csv
2023-08-26 18:13:02.056 [main info] read config files, got 11 messages
2023-08-26 18:13:02.077 [bus notice] bus started with own address 31/36
2023-08-26 18:13:02.077 [main info] registering data handlers
2023-08-26 18:13:02.077 [main info] registered data handlers
2023-08-26 18:13:02.109 [bus debug] ERR: SYN received during no signal, switching to ready
2023-08-26 18:13:02.109 [bus notice] signal acquired
2023-08-26 18:13:05.673 [bus notice] new master 10, master count 2
2023-08-26 18:13:05.735 [bus notice] new master 03, master count 3
2023-08-26 18:13:05.735 [update info] received MS cmd: 1008b5110101 / 093c3c0080ffff0000ff
2023-08-26 18:13:05.735 [update notice] received unknown MS cmd: 1008b5110101 / 093c3c0080ffff0000ff
2023-08-26 18:13:07.700 [update info] received BC cmd: 10feb5160800ffffffffffffff
2023-08-26 18:13:07.700 [update notice] received unknown BC cmd: 10feb5160800ffffffffffffff
2023-08-26 18:13:10.817 [update info] received MS cmd: 1008b510090000144cffff05ff00 / 0101
2023-08-26 18:13:10.817 [update notice] received unknown MS cmd: 1008b510090000144cffff05ff00 / 0101
2023-08-26 18:13:12.078 [main debug] performing regular tasks
2023-08-26 18:13:12.078 [main notice] starting initial scan for 08
2023-08-26 18:13:12.079 [bus info] scan 08 cmd: 3108070400
2023-08-26 18:13:12.114 [bus debug] start request 31
2023-08-26 18:13:12.114 [bus debug] arbitration start with 31
2023-08-26 18:13:12.200 [bus debug] arbitration lost
2023-08-26 18:13:12.200 [bus debug] ERR: arbitration lost during ready, retry
2023-08-26 18:13:12.200 [bus debug] start request 31
2023-08-26 18:13:12.200 [bus debug] arbitration start with 31
2023-08-26 18:13:12.286 [bus debug] arbitration lost
2023-08-26 18:13:12.286 [bus debug] ERR: arbitration lost during ready, retry
2023-08-26 18:13:12.286 [bus debug] start request 31
2023-08-26 18:13:12.286 [bus debug] arbitration start with 31
2023-08-26 18:13:12.372 [bus debug] arbitration lost
2023-08-26 18:13:12.372 [bus debug] ERR: arbitration lost during ready, retry
2023-08-26 18:13:12.372 [bus debug] start request 31
2023-08-26 18:13:12.372 [bus debug] arbitration start with 31
2023-08-26 18:13:12.458 [bus debug] arbitration lost
2023-08-26 18:13:12.458 [bus debug] notify request: ERR: arbitration lost
2023-08-26 18:13:12.458 [main error] initial scan failed: ERR: arbitration lost
2023-08-26 18:13:15.818 [update info] received MS cmd: 1008b5110101 / 093c3c0080ffff0000ff
2023-08-26 18:13:15.818 [update notice] received unknown MS cmd: 1008b5110101 / 093c3c0080ffff0000ff
2023-08-26 18:13:17.817 [update info] received MS cmd: 1008b5040100 / 0a00ffffffffffffff0080
2023-08-26 18:13:17.817 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffff0080
2023-08-26 18:13:18.070 [update info] received MS cmd: 1008b5110102 / 05004c964c78
2023-08-26 18:13:18.070 [update notice] received unknown MS cmd: 1008b5110102 / 05004c964c78
2023-08-26 18:13:20.801 [update info] received MS cmd: 1008b510090000144cffff05ff00 / 0101
2023-08-26 18:13:20.801 [update notice] received unknown MS cmd: 1008b510090000144cffff05ff00 / 0101
2023-08-26 18:13:22.459 [main debug] performing regular tasks
2023-08-26 18:13:22.459 [main notice] starting initial scan for 08
2023-08-26 18:13:22.459 [bus info] scan 08 cmd: 3108070400
2023-08-26 18:13:22.486 [bus debug] start request 31
2023-08-26 18:13:22.486 [bus debug] arbitration start with 31
2023-08-26 18:13:22.574 [bus debug] arbitration lost
2023-08-26 18:13:22.574 [bus debug] ERR: arbitration lost during ready, retry
2023-08-26 18:13:22.574 [bus debug] start request 31
2023-08-26 18:13:22.574 [bus debug] arbitration start with 31
2023-08-26 18:13:22.660 [bus debug] arbitration lost
2023-08-26 18:13:22.660 [bus debug] ERR: arbitration lost during ready, retry
2023-08-26 18:13:22.660 [bus debug] start request 31
2023-08-26 18:13:22.660 [bus debug] arbitration start with 31
2023-08-26 18:13:22.745 [bus debug] arbitration lost
2023-08-26 18:13:22.745 [bus debug] ERR: arbitration lost during ready, retry
2023-08-26 18:13:22.745 [bus debug] start request 31
2023-08-26 18:13:22.745 [bus debug] arbitration start with 31
2023-08-26 18:13:22.831 [bus debug] arbitration lost
2023-08-26 18:13:22.831 [bus debug] notify request: ERR: arbitration lost
2023-08-26 18:13:22.831 [main error] initial scan failed: ERR: arbitration lost
2023-08-26 18:13:25.007 [network info] [00001] client connection opened 127.0.0.1
2023-08-26 18:13:25.008 [network debug] [00001] wait for result
2023-08-26 18:13:25.008 [main debug] >>> i
2023-08-26 18:13:25.008 [main debug] <<< version: ebusd 23.2.23.2
device: /dev/ttyUSB1
signal: acquired
symbol rate: 24
max symbol rate: 44
r ...

`