john30/ebusd-configuration #224: Support 0010021887 Vaillant ecoTEC plus 206/5-5 (H-PL) R6

adam-kulagowski, 2021-07-19:
Hi,

I can confirm that it's working ok w/ `bai.0010015600.inc` instead of default `bai.308523.inc`

Regards,
Adam

john30, 2021-11-30:
are you referring to a pull request? or what's the requested action here?

adam-kulagowski, 2021-12-23:
I'm kindly requesting to assign  206/5-5 (H-PL) R6 to bai.0010015600.inc template instead of bai.308523.inc.

I've modified 08.bai.csv
```
[PROD='0010014629';'0010014630';'0010014631';'0010014632';'0010014633';'0010015596';'0010015597';'0010015598';'0010015599';'0010015600';'0010015601';'0010015602';'0010015603';'0010015604';'0010015605';'0010015606';'0010015607';'0010015608';'0010015609';'0010015610';'0010015611';'0010015612';'0010015613';'0010015614';'0010019268';'0010021888';'0010021887']!load,bai.0010015600.inc,,,
```
So PROD  0010021887  is correctly assigned to better template. I can confirm it is working correctly.

adam-kulagowski, 2023-08-05:
I've created PR for this device: https://github.com/john30/ebusd-configuration/pull/349

closed 2023-09-29 by adam-kulagowski