john30/ebusd-configuration #225: 15.700.csv extract current room humidity

kgeree, 2021-07-19:
Is it possible to somehow extract the current room humidity value from a VRC700? In the menu its available, both the current humidity and also current dew point.

mwildbolz, 2021-08-18:
would also be interesting for me - haven't found a solution yet.
Does anybody know?

rmeekers, 2022-01-10:
I'm also interested if anybody manages to read this out

Wireheadbe, 2022-10-26:
Running 22.4 here:
```
address 03: master #11
address 04: slave #25, ebusd
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0105;HW=7803", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
address ff: master #25, ebusd
```

Grab result shows:
```
1008b5110100 / 08010211001f100000 = 9033
1008b5160111 / 00 = 6
1052b5160111 / 090103010e0130010301 = 6
10feb510020601 = 1809
1008b5120204ff / 0101 = 1805
1008b513020508 / 00 = 1808
1008b5100305ff01 / 0101 = 1809
1052b5030c0700ffffffffffffffffffff / 0101 = 603
1052b516081000ffff0304212c / 0b2000ff0304592d00000000 = 18
1052b516081002ffff0304212d / 0b2201ff0304212d00000000 = 167
1052b52309000000000202000000 / 0101 = 4
```

But that doesn't seem like it contains Humidity data, we probably need to poll that directly from the registers.
Is there some way to do this? Eager to help out - can we read registers directly from ebus? Didn't find it immediately in the Wiki