john30/ebusd-configuration #230: readallvaillantregisters.sh

grossfr, 2021-09-17:
@john30
bei der Verwendung von https://github.com/john30/ebusd/blob/master/contrib/scripts/readallvaillantregisters.sh besteht hier nicht die Möglichkeit, dass ungewollte Schaltbefehle oder Sollwerte gesendet werden?
Gibt es eine Möglichkeit alle Register nur zu lesen ohne zu schreiben?

LukasGrebe, 2021-10-14:
mit dem `-r, --readonly` Flag beim start des ebusd sollte das verhindert werden können. Siehe auch: Only read from device, never write to ithttps://github.com/john30/ebusd/wiki/2.-Run#device-options

john30, 2021-11-30:
warum macht man mehrere [Einträge](https://github.com/john30/ebusd-configuration/issues/72#issuecomment-921949359) für den selben Inhalt? Das kostet jeden der es liest unnötig Zeit. Verstehe ich nicht.

closed 2021-11-30 by john30