john30/ebusd-configuration #231: wanted: data log for SetMode of Vaillant VWL 87/4 (or similar)

Th3Link, 2021-09-22:
Hi,
we would like to implement a control for my VWL heat pump. Can someone provide some grabbed data between the hmu00 and 70000? We do not have a VRC700 or similar ourselves and do not want to buy one.
Until now we run the heatpump in "Notlauf" but we would like to have it a bit more automated. The ebus communication is running so far (we can read a lot of statistics) but setting the flow temperature via the SetMode does not work out of the box and we do not have enough input data to find out what is needed. After changing from `uw` to `w` in hcmode.inc we can successfully send something like
`ebusctl write -c hmu setmode 0;23.0;55.0;55;0;0;0;0;1;1;1`
and played around with the mode (first number) and the different disable/release bits. however, the heatpump still remain solid...

`$ ebusctl f -f -w -c hmu SetMode
w,hmu,SetMode,Operation Mode,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,boiler mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,`

`$ ebusctl info
version: ebusd 21.2.v21.2
update check: OK
access: *
signal: acquired
symbol rate: 24
max symbol rate: 97
min arbitration micros: 11
max arbitration micros: 13
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 2
messages: 96
conditional: 0
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0308;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2, ebusd
address 15: slave #2, ebusd`

Software version of the VWF 87/4 HMU is 0304.03.08

Cheers :)

moritzschieder, 2022-08-22:
@Th3Link, do you still need help?
I have a Vaillant VWL 87/4 and VRC 720 up and running with ebusd…

Th3Link, 2022-08-23:
Hi @moritzschieder,

yes we are still manual controlling the heat pump, we made a little progress by getting an "unconnected to system controller" while experimenting. It would be very helpful if you can provide some data logging, best case with timestamps and some hints when you changed different parameters (enable/disable heating, changing heating temperature, enable/disable hot water, change hot water temperature, enable cooling, change cooling temperature).

Th3Link, 2022-12-05:
@moritzschieder I'd like to kindly ask you if you can grab some data from the ebus and paste it somewhere.
Cheers :)