john30/ebusd-configuration #234: VRC720f cannot set z1ActualRoomTempDesired

roman1528, 2021-10-16:
Hi.
I have an issue on setting the z1ActualRoomTempDesired.
Any change in any situation (SFmode/OpMode/Hc1RoomTempSwitchOn) doesn't set this value.

z1DayTemp seems not to work for this thermostat because the read is ERR: invalid position in decode.
(basv z1DayTemp = ff15b52406020003000700 / 00)

How can I set the desired RoomTemp to manually control the heater? Maybe in z1OpMode = day. Tryed anything...

basv z1ActualRoomTempDesired = ff15b52406020003001400 / 08010314000000cc41

r;w,,z1ActualRoomTempDesired,Raumsolltemperatur Zone 1,,,,1400,,,tempv,,,current room setpoint considering all basic conditions passed to the control algorithms

VRC 720f Room Controller (address 15: slave #2, scanned "MF=Vaillant;ID=BASV0;SW=0217;HW=8803", loaded "vaillant/15.basv.csv") -> 15.700.csv

atmoTEC plus VCW 194/4-5 (address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv") 

Thanks so far.

marcoutcho, 2021-11-02:
Hi,
please read my issue and see if it help : https://github.com/john30/ebusd-configuration/issues/209
At this time I'm using home assistant and I can modify the desired temperature with it only when a put the VRC720f in manual mode ( which is durating only x hours )

kgeree, 2022-10-28:
I have the exact same issue for a wired version of VRC720 (SensoComfort) z?DayTemp values not available. I haven't looked into it deeper yet, but let me know if I can help anything debugging / solving it.

paddenstoeltje, 2022-11-12:
Actually I have the same problem: I'm unable to write the z1ActualRoomTempDesired and z1DayTemp  is unable to be read ERR: invalid position for 3115b52406020003000700 / 00). 

Using: vaillant/15.basv.csv for ID "basv0", SW0217, HW8803 -> 15.700.csv

I tried already a lot of stuff to get it working, so far without succes. Any help would be appreciated.

paddenstoeltje, 2022-11-14:
Actually after first writing a value in z1ActualRoomTempDesired/set; I'm able to read and write the value. However, it seems the z1DayTemp is never used in any mode? I would expect to see it copied to z1ActualRoomTempDesired whenever the mode is in day.

Any help / suggestions would be appriciated.

paddenstoeltje, 2022-11-14:
Thanks, however the z1DayTemp is never loaded in the z1ActualRoomTempDesired; even when the z1OpMode is in 'day'. The same is true for the night mode and z1NightTemp.

- Are you also able to set multiple heating temperatures per day in the VRC700? I _assume_ the VRC720f directly takes the temperature as set in the device for that timeslot. Furthermore, I _assume_ this value is then written to z1ActualRoomTempDesired. This would explain why the z1DayTemp would do nothing. 
- If the above assumptions are true, I would love to be able to write directly on the ebus the temperatures for certain timeslots (eg z1timer.MondayTemp). However, as far as I can tell these values are not shared on the ebus (probably internal variables in the VRC720f?. 
- Other question, what is the meaning of 'z1SFMode'? I know this can be either veto or auto. 

Bottom line, I'm still missing some tags to control the temperature in auto /night /day mode. The only way to control it is to put it in veto mode and change the z1QuickVetoTemp.

I cannot image other people having the same problem. Is the VRC720f that new?

daclerk, 2022-11-23:
> Actually after first writing a value in z1ActualRoomTempDesired/set; I'm able to read and write the value. However, it seems the z1DayTemp is never used in any mode? I would expect to see it copied to z1ActualRoomTempDesired whenever the mode is in day.

Hey paddenstoeltje, so you managed to write z1ActualRoomTempDesired of the VRC720f? I am still not able to write this value such that it is actually accepted by my 720. While I can update other values (e.g. z1NightTemp), I did not manage to update the z1ActualRoomTempDesired value. Do you have any further hints? Thanks!

paddenstoeltje, 2022-11-24:
No, I'm not able to write the z1ActualRoomTempDesired tag. I'm able to write the z1NightTemp and the z1DayTemp though. However, none of these are copied in the z1ActualRoomTemp. So it seems a bit useless...

For the moment I use 3 modes; auto / veto / off. These can be set in OpMode in combination with SFMode.
- Auto mode (OpMode: auto, SFMode: auto). The heater takes the values as set in the VRC720f.
- Veto mode (OpMode: auto, SFMode: veto). I use this for overwriting the programmed setup. z1QuickVetoTemp is now used as setpoint; which can be written.
- Off mode (OpMode: off, SFMode: Auto). The setpoint is turned to 5degC, no clue where this is value is taken from. Also this 'off'-mode is not the same as turning the heating of in the VRC720f, as then there is just no SP anymore.

It is not ideal, but at least I have some degree of freedom to play with.

daclerk, 2022-12-16:
Ok, I think I have a solution. The csv file for the VRC720 has to be adapted. In particular the line

r;w,,z1ActualRoomTempDesired,room temperature desired zone 1,,,,1400,,,tempv,,,current room setpoint considering all basic conditions passed to the control algorithms

has to be replaced by

r;w,,z1ActualRoomTempDesired,room temperature desired zone 1,,,,2200,,,tempv,,,current room setpoint considering all basic conditions passed to the control algorithms

(so the "1400" has to be replaced by "2200").

For my cabled VRC720 this works (and I hope also for the wireless VRC720f). At least the value is updated on the display of the VRC720 if I change it via ebusd.

Since the csv file is stored on the github server (and is a link to the csv file of the VRC700), the quick solution is 
1. copying the csv files from the github server to a local directory (apparently also some of the other *.csv files and *.inc files from the directory are necessary)
2. renaming the file 15.700.csv to 15.720.csv and adapting the line mentioned above (change 1400 to 2200)
3. starting ebusd with the --configpath option set to this local directory

If somebody could confirm that this solution is working for the VRC720, we could ask @john30 to adapt the online version of the file 15.720.csv such that all future users do not have to bother with this problem.

Z-pi, 2022-12-16:
> Ok, I think I have a solution. The csv file for the VRC720 has to be adapted. In particular the line
>
> r;w,,z1ActualRoomTempDesired,room temperature desired zone 1,,,,1400,,,tempv,,,current room [...]

Thank you! 🎉 I changed my csv file with your instruction and this works for my wired VRC720.

paddenstoeltje, 2022-12-19:
Tried to test it but I always get the following error when trying to use my own config iso the github seerver:

[main error] error reading scan config file vaillant/15.basv.csv for ID "basv0", SW0217, HW8803: ERR: end of input reached, vaillant/15.basv.csv:1: ERR: end of input reached, missing message type/name/pbsb

Actually quite strange, as the local files are just a copy of https://github.com/john30/ebusd-configuration/

daclerk, 2022-12-19:
> [main error] error reading scan config file vaillant/15.basv.csv for ID "basv0", SW0217, HW8803: ERR: end of input reached, vaillant/15.basv.csv:1: ERR: end of input reached, missing message [...]

Ok, so maybe the csv file for the MRC720f should not be named 15.720.csv as for the wired version but 15.basv.csv.

Concerning the error message: Did you make sure that the local csv files have the right contents after copying them (e.g. by opening the local copies)? I got in the beginning a similar error because the csv files just had GitHub html content (I used the wget command for downloading, but apparently for getting the files with the real csv content I had to use a slightly different URL (maybe this is logic for someone that is used to work with GitHub, but for me as newbie to Github I didn’t know).

paddenstoeltje, 2022-12-19:
> Concerning the error message: Did you make sure that the local csv files have the right contents after copying them (e.g. by opening the local copies)? I got in the beginning a similar error because [...]

I did a git clone, so the files should be a 1on1. I checked with vi, contents is ok. 

> Ok, so maybe the csv file for the MRC720f should not be named 15.720.csv as for the wired version but 15.basv.csv.

Actually I tried changing manually 15.700.csv / 15.720.csv / 15.basv.csv without any luck. I kept getting the error. Even when nothing was modified and I just use a clone of Johns Repo I get the error. So the problem is somwerhere in my configuration. Will have a look later on.

paddenstoeltje, 2022-12-19:
Gave it another shot. Now it works, no clue what went wrong last time :).

I can confirm I'm able to write the z1actualRoomTempDesired when making the changes as discussed above. So the change is not only valid for the VRC720 but also for the VRC720**f**. It works beautifully! How did you found out it had to be 2200?

@daclerk do you make a pull request?

daclerk, 2022-12-19:
Perfect :-)! To figure out the 2200, I used a Vaillant VR900 and watched the ebus traffic (via the ebusctl grab command) while doing changes to the desired room temperature in the SensoAPP. After this experience with the VR900 and SensoAPP I am quite happy to have ebusd :-). Apart from the fact that only a few parameters are visible in this App, it is running quite unstable, lots of restarts of the VR900 were necessary and during a few hours the Vaillant-Server was not reachable. Luckily I bought the VR900 second hand for a good price…

Concerning the pull request: Maybe someone else could do it who knows how this works. I would first have to figure out how to do a pull request and for this I probably don’t have time during the next days.

paddenstoeltje, 2022-12-22:
The only porblem that I've got now is that the z1ActualRoomTempDesired is not updated anymore when the SFMode is in 'auto' (thus the VRC720f just uses the sheduled temperatures). It remains on the last set value I've done in SFMode 'day'. 

I would expect the z1ActualRoomTempDesired to follow the setpoint in auto mode as indicated on the VRC720f; so I can display this on my graph what the wished temperature is. This was the case when the value of z1ActualRoomTempDesired in the config was 1400. That's why I would like to find out what the 1400 / 2200 / xxx values actually mean.

Anyone else seeing the above result?

RotorMotor42, 2022-12-24:
I think 2200 is for setting and getting the day temperature and 1400 is only for reading in the actual/resulting desired temperature.

This works quiet fine for me:

`r,,z1ActualRoomTempDesired,Raumsolltemperatur Zone 1,,,,1400,,,tempv,,,current room setpoint considering all basic conditions passed to the control algorithms`

`r;w,,z1DayTemp,Solltemperatur Zone 1,,,,2200,,,tempv,,,day setpoint for zone 1`

Desmoyo, 2023-01-31:
> r;w,,z1DayTemp,Solltemperatur Zone 1,,,,2200,,,tempv,,,day setpoint for zone 1

you add that to csv 15.700.csv ?

chrispi68, 2023-04-28:
"z1ActualRoomTempDesiredWrite" is working for me on zone 2. I also added "z2ActualRoomTempDesiredWrite" for zone 2, but this seems not to work. Any ideas?

gasman1844, 2023-07-22:
Thank you to paddenstoeltje and others - I have exactly duplicated the steps that he took to get manual control of z1ActualRoomTempDesiredWrite and it works. Coming from Windows, this is my first time with linux and it has been a very steep learning curve. But I've now got key info coming through and I can usefully write as well.

mainmind83, 2024-07-14:
> Ok, I think I have a solution. The csv file for the VRC720 has to be adapted. In particular the line
>
> r;w,,z1ActualRoomTempDesired,room temperature desired zone 1,,,,1400,,,tempv,,,current room [...]

You can not duplicate ID 2200:

2024-07-15 01:07:46.111 [main error] error reading scan config file vaillant/15.ctlv2.csv for ID "ctlv2", SW0514, HW1104: ERR: duplicate entry, vaillant/15.ctlv2.csv:168: ERR: duplicate entry, duplicate ID

r;w,,z1DayTemp,Solltemperatur Zone 1,,,,2200,,,tempv,,,day setpoint for zone 1

mainmind83, 2025-01-02:
https://github.com/john30/ebusd-configuration/blob/cafba50f874c2d37d035c7260adfdcbffe6bfdad/ebusd-2.1.x/en/vaillant/15.ctlv2.csv?plain=1#L163

same 1400 value

vjonas, 2025-01-29:
> Ok, I think I have a solution. The csv file for the VRC720 has to be adapted. In particular the line
>
>
>
> r;w,,z1ActualRoomTempDesired,room temperature desired zone 1,,,,1400,,,tempv,,,current room [...]

Thank you so much!! I was searching for days to modify this param remotely!
This worked for me as well with the 720 wired. 
Should we modify the repo, and or atleast comment it to the repo next to it when to use 1400 VS 2200?