john30/ebusd-configuration #239: Help with decoding E7c

blanosko, 2021-11-16:
Hi, 

Yesterday my ebus adapter 3 (Wifi) arrived so I connected everything and tried to set it up. My enviroment is Proxmox host on which are HomeAssistant and ebusd as virtual machines. I have MQTT broker on HomeAssistant. Here is some basic info:

Boiler: Protherm Tiger Condens
Thermostat: Thermolink P (Wired) + Outside temperature sensor

Start up options for EBUSD:
`EBUSD_OPTS="-d enh:192.168.0.117:9999 -l /var/log/ebusd.log  --scanconfig --mqtthost=192.168.0.10 --mqttport=1883 --mqttuser=ebusd --mqttpass=pass"`

Result of ebusctl info command:
[image]

So, problem is that there is very little information on ebus bus. And messages that are decoded does not make any sense:

[image]

Is it correct configuration csv that it automaticaly loaded?

Thanks for any answers

john30, 2021-11-30:
for the bai there should be more values. just check with `ebusctl find` which messages are available

blanosko, 2021-11-30:
Thanks for advice. I found out that other values arent actively polled. I edited load config file with r1 on the lines I want and they started to be populated in mqtt messages. So loaded config for BAI is OK. Another peoblem is that my thermostat (controller) is E7C00 and from this device I cant get anything. Tried mapping config for e7f but values are not decoded corectly.