john30/ebusd-configuration #241: [~MISSING] VRC 720/2 (CTLV2) Configuration

mklooss, 2021-12-27:
Hey there,

we got an new VRC720/2 (CTLV2) - (0020260914) and it looks like not compatible right now.
an symlink to 15.700.csv went wrong. like

```
ebusctl r -f hwcTimer.Saturday
ERR: invalid position in decode
```
```
2021-12-27 21:21:11.802 [update error] unable to parse read ctlv2 hwcTimer.Monday from 3115b524050301000100 / 00: ERR: invalid position
2021-12-27 21:21:11.802 [main error] read ctlv2 hwcTimer.Monday: decode ERR: invalid position
```
so i looks like i have to create an new "translation"? for those commands

```
root@rpi2ebusd:~# ebusctl i
error connecting to localhost:8888
root@rpi2ebusd:~# ebusctl i
version: ebusd 21.3.v21.3
device: /dev/ttyAMA0, enhanced
access: *
signal: acquired
symbol rate: 31
max symbol rate: 70
min arbitration micros: 7
max arbitration micros: 15
min symbol latency: 10
max symbol latency: 28
reconnects: 0
masters: 3
messages: 225
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0804;HW=7603", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

so i have to take a look at: https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files

mklooss, 2021-12-27:
okay, it looks like just the timer configurations can not be read.

mklooss, 2021-12-28:
the symlink to 15.700.csv works for like ~60 %... of the commands i've tried.

So all "[xz]Timer.Monday" etc does not work.
```
ebusctl r -c ctlv2 -f -V z1Timer.Monday
ERR: invalid position in decode
```

how can i debug, to get more command / informations?

mklooss, 2022-01-04:
https://github.com/john30/ebusd/issues/487

meceo, 2022-05-14:
@john30 is there any instruction on how one can prepare a configuration file for a new device. I also have ctlv2 and would like to try to finalize the configuration.

I am receiving quite a lot of unknown commands, such as:

```
2022-05-14 19:39:37.414 [update notice] received unknown MS cmd: 1076b5110101 / 09ff27a011ff5b0000ff
2022-05-14 19:39:37.681 [update info] received MS cmd: 1076b512030f0201 / 07db020000800d00
2022-05-14 19:39:37.681 [update notice] received unknown MS cmd: 1076b512030f0201 / 07db020000800d00
2022-05-14 19:39:37.893 [update info] received BC cmd: 10feb505025c00
2022-05-14 19:39:37.893 [update notice] received unknown BC cmd: 10feb505025c00
2022-05-14 19:39:38.184 [update info] received MS cmd: 1052b5230103 / 0f00800080008000801f010080c87d04
2022-05-14 19:39:38.185 [update notice] received read vr_70 SensorData QQ=10: -;-;-;-;17.94;-;c8 7d 04
2022-05-14 19:39:38.458 [update info] received MS cmd: 1008b51009000000ffffff070000 / 0101
2022-05-14 19:39:38.459 [update notice] received update-write hmu SetMode QQ=10: auto;0.0;-;-;1;1;1;0;0;0
2022-05-14 19:39:38.719 [update info] received MS cmd: 1076b51009000000ffffff010000 / 0101
2022-05-14 19:39:38.719 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2022-05-14 19:39:38.960 [update info] received MS cmd: 1008b507020900 / 021703
2022-05-14 19:39:38.960 [update notice] received unknown MS cmd: 1008b507020900 / 021703
```

john-o-matic, 2022-05-25:
Hello,
My name is John, but not John30 who owns this repo
I also recently changed for a VRC 720F and I'm also trying to figure out how to adapt the timer parameters not working.
Main difference (and enhancement) I noticed in timer programming compared to VRC 700 is:
- each day is no longer limited to 3 slots
- each timer slot has a defined temperature (if I remember correctly VRC700 applies "confort" temperature to all slots)
So I think this is why the csv need to be adapted. Looking at the log below it looks the bus is replying, but the message only contains 0. 
```
2022-05-08 22:14:01.470 [update error] unable to parse read basv z1Timer.Tuesday from 3115b524050303000201 / 00: ERR: invalid position
2022-05-08 22:14:01.470 [main error] read basv z1Timer.Tuesday: decode ERR: invalid position
2022-05-08 22:14:01.471 [bus notice] >3115b52405030300020176<000000>00
```
I actually removed z1timer.Monday (to z1timer.Saturday) from my 15.700.csv file and then attempted to send the following read command in hex
ebusctl read -h -f 15b524050303000201
But I did not see more info using "ebusctl grab result decode" and "ebusctl grab result all".
I hope @john30, or someone else, can help us out on this topic.

jakolion, 2024-10-04:
Have you found a solution to the problem? I am stuck at the same point and would like to set the timers via ebusd to avoid the inconvenient input on the Vaillant console.