john30/ebusd-configuration #242: Vaillant ecoTEC pure automatic modulation and SetMode

pawelmuszynski, 2021-12-29:
Hello,

I have Vaillant ecoTEC pure VC 226/7-2. It's detected as a MF=Vaillant;ID=BAI00;SW=0709;HW=1303. Configuration loaded: "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
I'd like to emulate a room controller (like VRC700, etc.). I read that it sends SetMode command every 10 minutes and I checked when I stop sending SetMode, settings will be reseted after 20 minutes. According to https://github-com.translate.goog/john30/ebusd/issues/179?_x_tr_sl=de&_x_tr_tl=pl&_x_tr_hl=en-US&_x_tr_pto=op,sc#issuecomment-434628658 I created SetModeOverride to have ability to send proper message. It turns out when I send "0;60.0;45.0;-;0;0;0;0;0;0" value my gas boiler keeps water at 60 degrees but the automatic modulation doesn't work. It works at minimal power (25%). When I send "-;60.0;45.0;-;0;0;0;0;0;0" value, ebus controller is not in use (d.90 register shows 'off') and automatic modulation starts working fine but the boiler doesn't use the values and set the water temperature to 55 (it's also showed in d.05 register) It looks like it's a default value when you stop sending SetMode (SetModeOverride). I don't even know why it is 55. I set FlowsetHcMax to 70. Can someone show me what values of SetMode are sent by room controller? Or how to enable automatic modulation during sending SetModeOverride?

wizu, 2022-03-30:
https://github.com/john30/ebusd-configuration/issues/220

pawelmuszynski, 2022-07-25:
Hallo,

based on https://github.com/john30/ebusd-configuration/issues/170#issuecomment-709592197 I redefined the record for SetModeOverride to:

```
wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,setmode1,,UCH,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,setmode2,,UCH,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2
```

I also redefined the invocation to:

```
#!/bin/sh

hcmode="0"
flowtempdesired="65"
hwctempdesired="45"
hwcflowtempdesired="-"
disablehc="0"
disablehwcload="0"

ebusctl w -c bai SetModeOverride "$hcmode;$flowtempdesired;$hwctempdesired;$hwcflowtempdesired;-;$disablehc;0;$disablehwcload;-;0;0;0"
```

and the boler started increasing its power after several minutes:

```
$ ebusctl read -V ModulationTempDesired
bai ModulationTempDesired =57.7 % [Modulationssollwert]
```

I will check during the winter how long it take the water to heat up. I have old charts to compare. Now it's too hot outside to turn heaters on for longer time.

Thank you @wizu to guide me to the right topic.

HiQual, 2023-04-12:
Hello @pawelmuszynski 
Many thanks for your "SetModeOverride" settings - I can now write my "FlowTempDesired" and modulation remains working.
I tried to write "PartloadHcKW" by changing the third field in the SetModeOverride command, to set the bai maximum output power - so far without success.
If you have any suggestions I would be very pleased to hear them - thank you.

pawelmuszynski, 2023-04-13:
Hello @HiQual 
The 3rd value from SetModeOverride is hwctempdesired. It's responsible for the target temperature in your hot water tank. For max power setting I set it manually in the boiler. I can see that writing to PartloadHcKW and PartloadHwcKW registers don't change the max boiler power. I just tested it on hot water tank heating. I didn't dive into it because manual setting is enough for me.

Answering to my previous question it looks like my change on SetModeOverride works fine with modulation.

HiQual, 2023-04-14:
Hello @pawelmuszynski 
Many thanks for your reply.  I am doing the same as you - and it is working well.
Just in case anyone else is reading this, I should explain that my question meant to say I had altered the bai.nnnnn.inc file SetModeOverride line by replacing hwctempdesired with PartloadHcKW - without success.

closed 2023-04-14 by pawelmuszynski

dulek, 2025-10-01:
@pawelmuszynski: Hey, I'm trying to get this working, but I found out that using `SetModeOverride` I can only lower the desired temperatures compared to the values set on the boiler display. I can't get them higher.

I could set the high values on the boiler and keep a loop of messages lowering them constantly, but for the sake of possible ebusd or Home Assistant breakage I'd rather have the boiler reset down, not up.

Had you observed a similar effect? Or maybe I'm running the boiler in a wrong mode?

dulek, 2025-10-24:
> > > I have a glowworm 35c, but it presents with the same hw/sw codes as you guys.
> > > I for the life of my cannot find the `SetModeOverride`
> > > I just get a element not found.
> > > Do I need to add it manually to the definitions?
> > 
> > 
> > Yes. Use these: [#242 (comment)](https://github.com/john30/ebusd-configuration/issues/242#issuecomment-1194481837)
> 
> Thanks for reply.
> 
> Are you continously resending the commands to make sure they stick?

Yup, I use Home Assistant to publish MQTT message every 5 minutes + have an alert when heating curve value doesn't match value on the heater for more than 10 minutes. So far this had run without issues, but I give no guarantees. ;) 

BTW - if you have return sensor enabled, then the magic formula to compare heater value with heat curve value is T_curve = T_heater * 2 - 20.

ghosalmartin, 2025-10-26:
@dulek ah cracking thanks. 

its good to know then its all fine for this boiler. 

huh ill have to take a look at return sensor.

awalland, 2026-01-08:
i don't have `ModulationTempDesired` anywhere, i do have `FlowTempDesired` though. this is what i use to set the water temp for heating.

```
-> % grep TempDesired /etc/ebusd/08.bai.csv
r,,,FlowTempDesired,d.05 Flow/return set point: Flow temperature target value or return target value when return regulation is set,,,b509,0d3900,value,,D2C,,°C,temperature
r,,,HwcTempDesired,d.06 DHW setpoint: Hot water temperature target value,,,b509,0dea03,value,,D2C,,°C,temperature
r,,,StorageTempDesired,"d.07 Storage setpoint: Warm start temperature value (combination boiler plus only), Storage temperature target value (system boiler only)",,,b509,0d0400,value,,D2C,,°C,temperature
r,,,ExtFlowTempDesiredMin,d.09 ext flowsetpoint: minimum out of Kl.7 and eBus flow setpoint,,,b509,0d6e04,value,,D2C,,°C,temperature
r,,,ExternalFlowTempDesired,external flowsetpoint: Flow target temperature from external regulator to terminal 7-8-9,,,b509,0d2500,value,,D2C,,°C,temperature
```

bobrippling, 2026-02-02:
Interesting, just `hcmode`, `flowtempdesired` and `hwctempdesired` if I've read that right - also `setmode1` looks to be set (`0`)?

What's the Tado's heating-off command look like? And what do you get when you repeat those commands via a `ebusctl write` ?

jakeycrx, 2026-02-02:
> Interesting, just `hcmode`, `flowtempdesired` and `hwctempdesired` if I've read that right - also `setmode1` looks to be set (`0`)?
> 
> What's the Tado's heating-off command look like? And what do you get when you repeat those commands via a `ebusctl write` ?

It looks like it doesn't use setmode1 & 2. This is what the Tado shows for setmode when in control:

 {
  "hcmode": {
    "value": "auto"
  },
  "flowtempdesired": {
    "value": 80
  },
  "hwctempdesired": {
    "value": 45
  },
  "hwcflowtempdesired": {
    "value": null
  },
  "disablehc": {
    "value": 0
  },
  "disablehwctapping": {
    "value": 0
  },
  "disablehwcload": {
    "value": 0
  },
  "remoteControlHcPump": {
    "value": 0
  },
  "releaseBackup": {
    "value": 0
  },
  "releaseCooling": {
    "value": 0
  }
}

The command for heating off is the same, but just with disablehc set to 1 and flowtemp to 0

I get an end of input reached error when trying to send the same command, I think because it only has 10 parameters and it's expecting 12?

/ # ebusctl write -c bai SetModeOverride '0;75;45.0;-;0;0;0;0;0;0'
ERR: end of input reached

jakeycrx, 2026-02-03:
> Hello [@jakeycrx](https://github.com/jakeycrx) Please take a look at [#537 (comment)](https://github.com/john30/ebusd-configuration/issues/537#issue-3549659003) You also need to know that you have to repeat send the Setmode command (< every 10 minutes typical) otherwise the boiler reverts to its internal control. Also try with first the Tado disconnected because having other controllers on the bus can interfere with each other.

Thanks! Modulation appears to be working now after I used your command. setmode1 & 2 need to be set to -

ebusctl write -c bai SetModeOverride '0;80;50;-;-;0;0;0;-;0;0;0'

My next issue is setting SetModeOverride via MQTT. Is anybody sucessfully using MQTT for SetModeOverride?

bobrippling, 2026-02-03:
> It looks like it doesn't use setmode1 & 2. This is what the Tado shows for setmode when in control:

Do you have the line emitted by `ebusctl listen` for those tado commands?

dulek, 2026-02-03:
> Thanks! Modulation appears to be working now after I used your command. setmode1 & 2 need to be set to -
>
> ebusctl write -c bai SetModeOverride '0;80;50;-;-;0;0;0;-;0;0;0'
>
> My next issue is setting [...]

I use such action in Home Assistant:
```
metadata: {}
data:
  qos: "1"
  retain: false
  topic: ebusd/bai/SetModeOverride/set
  payload: >-
    0;{{ states('sensor.heating_curve') }};{{
    states('input_number.hwc_setpoint') }};-;-;0;0;0;-;0;0;0
action: mqtt.publish
```

bobrippling, 2026-02-04:
> Thank you for all the help!
> 
> > Do you have the line emitted by `ebusctl listen` for those tado commands?
> 
> How do I check that?

Good to hear! For that, are you able to run `ebusctl listen` on the machine that's running `ebusd`? And then make the changes via Tado to see what commands it emits (via the `listen` command)?

jakeycrx, 2026-02-04:
> Thanks - the `setmode1` and `setmode2` parameters I'm trying to figure out what they mean, how do you use them to control the boiler? (As we've seen, in the spec they're just ignored)

I'm not sure exactly what they do, but on my boiler if I send a value of 0 or 1 to `setmode1` or `setmode2` then the boiler gets stuck at minimum modulation.

The SetMode cmd for my boiler has to have `setmode1` and `setmode2` set to `-` which I believe to translate to `null`

I send `0;50;45;-;-;0;0;0;-;0;0;0` and the modulation has been ramping up and down perfectly so far.