john30/ebusd-configuration #248: VRC 720 SFMode 1 boost ventilation (stoßlüften) not working

snooze83, 2022-01-09:
Hello,
i have a problem with the VRC 720. The funktion boost ventilation (stoßlüften) is not enableable by ebusd. With the VRC 700 it does work with command "ebusctl write -c 700 SFMode 1". SFMode 1 is not working with VRC 720. 
If i write command  "ebusctl write -c 700 **z1**SFMode 1" to the ebus, Display from 720 shows "stoßlüten" the heating stop, but the recovair (ventilation system) does not rule and no speed increase.
I bought a vr921 and logged the data. Unfortunately I don't get any further. Can someone take a look at that? That is the period from the triggering of "burst ventilation" of the app to the reaction of the ventilation system.

Thanks, 
Andreas 

2022-01-08 16:44:12.395 [update notice] received unknown MS cmd: 1008b5110101 / 09524f20fe4b680000ff
2022-01-08 16:44:12.666 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2022-01-08 16:44:12.920 [update notice] received unknown MS cmd: 1008b5110102 / 06033c9646826e
2022-01-08 16:44:14.001 [update notice] received unknown MS cmd: 0015b5240702010000f90001 / 020000
2022-01-08 16:44:14.114 [update notice] received unknown MS cmd: 0306b511040a470300 / 0100
2022-01-08 16:44:14.425 [update notice] received unknown MS cmd: 0015b5240602000000f900 / 050300f90000
2022-01-08 16:44:15.228 [update notice] received read ctlv2 z1ActualRoomTempDesired QQ=00: 0.0
2022-01-08 16:44:15.409 [update notice] received unknown MS cmd: 0306b511020900 / 0700800000ff1020
2022-01-08 16:44:15.497 [update notice] received unknown MS cmd: 0015b52406020003001b00 / 0601031b000100
2022-01-08 16:44:16.245 [update notice] received unknown MS cmd: 0015b52406020009000400 / 06010904000100
2022-01-08 16:44:22.401 [update notice] received unknown MS cmd: 1008b5110101 / 09524f20fe4b680000ff
2022-01-08 16:44:22.672 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2022-01-08 16:44:24.213 [update notice] received unknown MS cmd: 0306b511040a470300 / 0100
2022-01-08 16:44:25.254 [update notice] received unknown MS cmd: 0306b511020900 / 0700800000ff1020
2022-01-08 16:44:31.120 [update notice] received read ctlv2 HwcHolidayStartPeriod QQ=00: 01.01.2020
2022-01-08 16:44:31.567 [update notice] received read ctlv2 HwcHolidayEndPeriod QQ=00: 02.01.2020
2022-01-08 16:44:31.925 [update notice] received read ctlv2 HwcOpMode QQ=00: auto
2022-01-08 16:44:32.407 [update notice] received unknown MS cmd: 1008b5110101 / 09524f20fe4b680000ff
2022-01-08 16:44:32.678 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2022-01-08 16:44:34.436 [update notice] received unknown MS cmd: 0306b511040a470300 / 0100
2022-01-08 16:44:34.748 [update notice] received read ctlv2 HwcSFMode QQ=00: auto
2022-01-08 16:44:34.985 [update notice] received read ctlv2 HwcStorageTemp QQ=00: 52
2022-01-08 16:44:35.122 [update notice] received unknown MS cmd: 0306b511020900 / 0700800000ff1020
2022-01-08 16:44:35.215 [update notice] received read ctlv2 HwcTempDesired QQ=00: 55
2022-01-08 16:44:35.447 [update notice] received unknown MS cmd: 0015b52406020001000f00 / 0601010f000000
2022-01-08 16:44:35.680 [update notice] received unknown MS cmd: 0015b52406020001001100 / 0703011100000000
2022-01-08 16:44:35.903 [update notice] received unknown MS cmd: 0015b52406020001001200 / 050301120000
2022-01-08 16:44:36.223 [update notice] received unknown MS cmd: 0015b52406020001001000 / 0703011000000000
2022-01-08 16:44:36.753 [update notice] received unknown MS cmd: 0015b52406020000004800 / 06010048000000
2022-01-08 16:44:37.292 [update notice] received unknown MS cmd: 0015b52406020003002200 / 08020322000000a841
2022-01-08 16:44:37.528 [update notice] received unknown MS cmd: 0015b52406020003002800 / 080103280000001c42
2022-01-08 16:44:37.765 [update notice] received read ctlv2 z1HolidayEndPeriod QQ=00: 02.01.2020
2022-01-08 16:44:38.033 [update notice] received unknown MS cmd: 0015b52406020003002700 / 050003270000
2022-01-08 16:44:38.265 [update notice] received unknown MS cmd: 0015b52406020003001e00 / 0703031e00000000
2022-01-08 16:44:38.497 [update notice] received read ctlv2 z1OpMode QQ=00: auto
2022-01-08 16:44:38.682 [update error] unable to parse read ctlv2 z1Name2 from 0015b52406020003001800 / 06030318003100: ERR: invalid position
2022-01-08 16:44:38.926 [update notice] received read ctlv2 z1Name1 QQ=00: Zone
2022-01-08 16:44:39.162 [update notice] received read ctlv2 z1NightTemp QQ=00: 21.5
2022-01-08 16:44:39.387 [update notice] received read ctlv2 z1SFMode QQ=00: ventilation
2022-01-08 16:44:39.611 [update notice] received unknown MS cmd: 0015b52406020003000100 / 06020301000100
2022-01-08 16:44:39.847 [update notice] received read ctlv2 z1RoomTemp QQ=00: 21.7875
2022-01-08 16:44:40.087 [update notice] received read ctlv2 z1CoolingTemp QQ=00: 24
2022-01-08 16:44:40.324 [update notice] received read ctlv2 z1QuickVetoTemp QQ=00: 22.5
2022-01-08 16:44:40.556 [update notice] received read ctlv2 z1HolidayStartPeriod QQ=00: 01.01.2020
2022-01-08 16:44:40.776 [update notice] received unknown MS cmd: 0015b52406020003001100 / 050203110001
2022-01-08 16:44:41.017 [update notice] received read ctlv2 z1HolidayTemp QQ=00: 20
2022-01-08 16:44:41.250 [update notice] received unknown MS cmd: 0015b52406020003002600 / 080303260000004040
2022-01-08 16:44:41.474 [update notice] received unknown MS cmd: 0015b52406020003001c00 / 0501031c0000
2022-01-08 16:44:41.706 [update notice] received unknown MS cmd: 0015b52406020003002400 / 0703032400010113
2022-01-08 16:44:42.026 [update notice] received unknown MS cmd: 0015b52406020003002300 / 08020323000000c041
2022-01-08 16:44:42.361 [update notice] received unknown MS cmd: 0015b52406020003002100 / 0703032100000000
2022-01-08 16:44:42.440 [update notice] received unknown MS cmd: 1008b5110101 / 09524f20fe4b680000ff
2022-01-08 16:44:42.719 [update notice] received unknown MS cmd: 0015b52406020003002500 / 050303250000
2022-01-08 16:44:42.804 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2022-01-08 16:44:43.049 [update notice] received unknown MS cmd: 0015b52406020003002000 / 0703032000000000
2022-01-08 16:44:43.134 [update notice] received read bai DateTime QQ=10: valid;16:44:44;08.01.2022;-1.875
2022-01-08 16:44:43.391 [update notice] received unknown MS cmd: 1008b5110102 / 06033c9646826e
2022-01-08 16:44:43.630 [update notice] received update-read broadcast vdatetime QQ=10: 16:44:43;08.01.2022
2022-01-08 16:44:43.887 [update notice] received unknown MS cmd: 1008b5110100 / 08950212001f100080
2022-01-08 16:44:44.246 [update notice] received unknown MS cmd: 1038b522050003002a00 / 0100
2022-01-08 16:44:44.612 [update notice] received unknown MS cmd: 1038b50903298503 / 0485039902
2022-01-08 16:44:44.672 [update notice] received unknown MS cmd: 0306b511040a470300 / 0100
2022-01-08 16:44:44.989 [update notice] received unknown MS cmd: 1038b50903298603 / 0486030080
2022-01-08 16:44:45.079 [update notice] received unknown MS cmd: 0306b511020900 / 0700800000ff1020
2022-01-08 16:44:45.814 [update notice] received unknown MS cmd: 1038b50903298803 / 0488030080
2022-01-08 16:44:46.209 [update notice] received unknown MS cmd: 1038b51303048903 / 0a8903b0042003d007b004
2022-01-08 16:44:46.497 [update notice] received read ctlv2 Time QQ=00: 16:44:47
2022-01-08 16:44:46.707 [update notice] received unknown MS cmd: 1038b51303048a03 / 0a8a030300010006000400
2022-01-08 16:44:47.131 [update notice] received unknown MS cmd: 1038b51303048b03 / 0a8b030100010006000100
2022-01-08 16:44:47.219 [update notice] received unknown MS cmd: 0015b52406020009000100 / 06030901000100
2022-01-08 16:44:47.717 [update notice] received unknown MS cmd: 0015b52406020009000d00 / 0603090d000300
2022-01-08 16:44:47.960 [update notice] received unknown MS cmd: 0015b52406020009000800 / 0703090800000000
2022-01-08 16:44:47.991 [update notice] received update-read broadcast outsidetemp QQ=10: -1.875
2022-01-08 16:44:48.268 [update notice] received unknown MS cmd: 0015b52406020009000a00 / 0703090a00000000
2022-01-08 16:44:48.492 [update notice] received unknown MS cmd: 0015b52406020009000f00 / 0503090f0000

kgeree, 2023-01-08:
Hi, did you get any further on this? I'm also missing the "quick" functions on the VRC 720..

snooze83, 2023-01-08:
Hi, unfortunately no solution until now