john30/ebusd-configuration #249: B511 message not yet decoded?

brucemiranda, 2022-01-11:
After many years I recompiled a local version of ebusd and noted that it now supported an remote configuration from http://cfg.ebusd.eu/. So decided to give it a try, 

Noticed that the B511 message has still not yet been formally decoded and nor is my Vaillant boiler added to the list, which is why the defaults are loaded.

Can my boiler be included in the common files? And the definition for the B511 message?

`
 ebusctl info
version: ebusd 21.3.v21.3-135-g29e9633
update check: revision v21.3 available
device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9M9DV3R-if00-port0
access: *
signal: acquired
symbol rate: 62
max symbol rate: 133
min arbitration micros: 111
max arbitration micros: 212
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 3
messages: 227
conditional: 3
poll: 0
update: 11
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 64: slave, scanned "MF=Vaillant;ID=V6500;SW=0007;HW=8301", loaded "vaillant/64.v65.csv"
address 7f: master #24
address 84: slave #24, scanned "MF=Vaillant;ID=OTS00;SW=0205;HW=9101"

ebusctl scan result
08;Vaillant;BAI00;0518;7401;21;10;08;0010002725;1300;005719;N9
64;Vaillant;V6500;0007;8301
84;Vaillant;OTS00;0205;9101

ebusctl grab result
7f08b5110100 / 08f803ff22040f0081 = 690
`

The closest configuration I have for the B511 message which I used to use locally was. 

`r,,FlowTempStatus,Status Heizkreis,,,,00,FlowTemp,,temp,,,,WaterPressure,,UCH,10,bar,,unknown4,,UCH,,,,Status,,UCH,,,,Fan,,BI0,1=Fan,,,Gas1,,BI1,1=Gas1,,,Gas2,,BI2,1=Gas2,,,HcPump,,BI3,1=HcPump,,,HwcPump1,,BI4,1=HwcPump1,,,HwcPump2,,BI5,1=HwcPump2,,,Ignition,,BI6,1=Ignition,,,Bit7,,BI7,1=Bit7,,,ErrorStatus,,UCH,,,,HcDemand,,BI0,1=HcDemand,,,BlockTime,,BI1,1=BlockTime,,,Bit2,,BI2,1=Bit2,,,Bit3,,BI3,1=Bit3,,,Bit4,,BI4,1=Bit4,,,Bit5,,BI5,1=Bit5,,,Bit6,,BI6,1=Bit6,,,HwcDemand,,BI7,1=HwcDemand,,`

brucemiranda, 2022-01-11:
Is there are a way to supplement the remote configuration files with a local configuration file for a single message that isn't being decoded by the remote ones?

cyberthom42, 2022-10-25:
That should work with the include option. I am not sure if there is a combination possible local and remote using the include option.

djd1983, 2024-02-18:
Hi there, this is my first contribution on git hub and I hope I can offer a little progress regarding this issue.

I found out that there was one Massage 1008B5110100 is not decoded in my system and in the hcmode.inc. After some trys and errors I thought about, that there is one more "StatusChain" than Status 01 (B5110101) or Status02 (B5110102).

So when i type

ebusctl hex 08B5110100 i geht the following strings as periodically bradcast, one time in a minute

`0826031426040f0081`

So for a better overview I manipulated the hcmode.inc with the additional Entry

`r,,StatusX,Wert1/Wert2/Wert3/Wert4/Wert5/Wert6/Wert7/Wert8,,,,00,,,HEX;HEX;HEX;HEX;HEX;HEX;HEX;HEX,,,`

After that I recorded, for an half an hour, the results and thats what it is:

```
18.02.2024 13:51:54(-60.27 seconds) b4;02;13;26;04;0f;00;81
18.02.2024 13:50:54(-60.08 seconds) af;02;13;26;04;0f;00;81
18.02.2024 13:49:54(-60.07 seconds) a5;02;13;26;04;0f;00;81
18.02.2024 13:48:55(-59.07 seconds) 9a;02;13;26;04;0f;00;81
18.02.2024 13:47:54(-60.1 seconds)  8b;02;13;26;04;0f;00;81
18.02.2024 13:46:54(-60.26 seconds) 7d;02;13;26;04;0f;00;81
18.02.2024 13:45:54(-60.25 seconds) 6c;02;13;26;04;0f;00;81
18.02.2024 13:44:54(-60.37 seconds) 6e;02;13;26;04;0f;00;81
18.02.2024 13:43:54(-59.11 seconds) 6c;02;13;26;04;0f;00;81
18.02.2024 13:42:49(-65.09 seconds) c1;02;13;26;04;0f;00;81
18.02.2024 13:41:54(-55.22 seconds) 39;03;13;00;00;08;00;81
18.02.2024 13:40:49(-65.22 seconds) af;04;13;55;18;3f;00;80
18.02.2024 13:39:49(-60.07 seconds) af;04;13;5a;18;3f;00;80
18.02.2024 13:38:49(-59.37 seconds) b1;04;13;5e;18;3f;00;80
18.02.2024 13:37:49(-60.26 seconds) af;04;13;65;18;3f;00;80
18.02.2024 13:36:49(-60.35 seconds) aa;04;13;6a;18;3f;00;80
18.02.2024 13:35:49(-60.27 seconds) bd;04;13;90;18;3f;00;80
18.02.2024 13:34:49(-59.36 seconds) 3a;04;12;c8;18;3f;00;80
18.02.2024 13:33:49(-60.36 seconds) 85;02;10;8c;17;76;00;80
18.02.2024 13:32:49(-60.16 seconds) 80;02;0f;00;08;00;00;83
18.02.2024 13:31:49(-60.13 seconds) 85;02;10;00;08;00;00;83
18.02.2024 13:30:48(-60.16 seconds) 8f;02;12;00;07;08;00;83
18.02.2024 13:29:49(-59.34 seconds) 9a;02;12;00;07;08;00;83
18.02.2024 13:28:49(-60.37 seconds) a5;02;12;00;07;08;00;83
18.02.2024 13:27:48(-60.26 seconds) af;02;12;00;07;08;00;83
18.02.2024 13:26:44(-64.36 seconds) 0c;03;12;00;07;08;00;83
18.02.2024 13:25:44(-60.36 seconds) 1e;03;12;26;04;0f;00;81
18.02.2024 13:24:44(-60.17 seconds) 12;03;12;26;04;0f;00;81
18.02.2024 13:23:43(-60.18 seconds) 03;03;12;26;04;0f;00;81
18.02.2024 13:22:44(-59.08 seconds) f0;02;12;26;04;0f;00;81
18.02.2024 13:21:44(-60.3 seconds)  e4;02;12;26;04;0f;00;81
18.02.2024 13:20:44(-60.21 seconds) de;02;12;26;04;0f;00;81
18.02.2024 13:19:44(-60.08 seconds) 7e;02;12;00;02;08;00;81
18.02.2024 13:18:44(-60.13 seconds) 69;02;0f;00;08;00;00;83
18.02.2024 13:17:43(-60.16 seconds) 6c;02;0f;00;08;00;00;83
18.02.2024 13:16:43(-60.13 seconds) 6e;02;0f;00;08;00;00;83
18.02.2024 13:15:44(-59.22 seconds) 70;02;0f;00;08;00;00;83
18.02.2024 13:14:44(-60.21 seconds) 73;02;0f;00;08;00;00;83
18.02.2024 13:13:44(-60.22 seconds) 76;02;0f;00;08;00;00;83
18.02.2024 13:12:43(-60.21 seconds) 7b;02;10;00;08;00;00;83
18.02.2024 13:11:44(-59.22 seconds) 86;02;12;00;07;08;00;83
18.02.2024 13:10:39(-65.41 seconds) 92;02;12;00;07;08;00;83
18.02.2024 13:09:38(-60.36 seconds) a2;02;12;00;07;08;00;83
18.02.2024 13:08:39(-59.36 seconds) af;02;12;00;07;08;00;83
18.02.2024 13:07:39(-60.31 seconds) 1e;03;12;00;07;08;00;83
18.02.2024 13:06:38(-60.31 seconds) 1e;03;12;26;04;0f;00;81
18.02.2024 13:05:38(-60.21 seconds) 15;03;12;26;04;0f;00;81
18.02.2024 13:04:39(-59.17 seconds) 0a;03;12;26;04;0f;00;81
18.02.2024 13:03:39(-60.12 seconds) fb;02;12;26;04;0f;00;81
18.02.2024 13:02:39(-60.18 seconds) ee;02;12;26;04;0f;00;81
18.02.2024 13:01:38(-60.35 seconds) e3;02;12;26;04;0f;00;81
18.02.2024 13:00:39(-59.36 seconds) e6;02;12;26;04;0f;00;81
18.02.2024 12:59:39(-60.27 seconds) 76;02;0f;00;08;00;00;83
[... 26 lines cut ...]
18.02.2024 12:32:29(-60.3 seconds)  a5;02;13;00;07;08;00;83
18.02.2024 12:31:28(-60.28 seconds) ee;02;13;00;07;08;00;83
18.02.2024 12:30:28(-60.12 seconds) 19;03;13;26;04;0f;00;81
18.02.2024 12:29:29(-59.36 seconds) 08;03;13;26;04;0f;00;81
18.02.2024 12:28:29(-60.31 seconds) f6;02;13;26;04;0f;00;81
18.02.2024 12:27:28(-60.36 seconds) ee;02;13;26;04;0f;00;81
18.02.2024 12:26:28(-60.26 seconds) 7b;02;0f;00;08;00;00;83
18.02.2024 12:25:29(-59.22 seconds) 7b;02;0f;00;08;00;00;83
18.02.2024 12:24:28(-60.36 seconds) 7b;02;0f;00;08;00;00;83
18.02.2024 12:23:28(-60.13 seconds) 7d;02;0f;00;08;00;00;83
18.02.2024 12:22:28(-60.31 seconds) 80;02;10;00;08;00;00;83
18.02.2024 12:21:29(-59.41 seconds) 83;02;10;00;08;00;00;83
18.02.2024 12:20:23(-65.41 seconds) 86;02;10;00;08;00;00;83
18.02.2024 12:19:23(-60.17 seconds) 8d;02;13;00;07;08;00;83
18.02.2024 12:18:24(-59.22 seconds) 95;02;13;00;07;08;00;83
18.02.2024 12:17:23(-60.36 seconds) 9d;02;13;00;07;08;00;83
18.02.2024 12:16:23(-60.37 seconds) a4;02;13;00;07;08;00;83
18.02.2024 12:15:24(-59.31 seconds) c9;02;13;00;07;08;00;83
18.02.2024 12:14:24(-60.12 seconds) 23;03;13;26;04;0f;00;81
18.02.2024 12:13:23(-60.13 seconds) 0f;03;13;26;04;0f;00;81
```

The first four Bytes seem to be somethin like counting up and down. If the first Byte finished counting to "ff" for example, than the second byte changes from "02" to "03" and so on. The last Byte seems to be always "81", "83"

Maybe there is someone who can interprete the data into some heatinglanguage :)

chrizzzp, 2024-02-19:
@djd1983 It has been decoded (Status00), see here: https://github.com/john30/ebusd-configuration/pull/330#issuecomment-1859292111

```
*r,,,,,,"B511",,,,,,,,,,,,,,,,,,,,,,,,,
r,,Status00,,,,,00,S00_SupplyTemp,,temp,,,,WaterPressure,,UCH,10,bar,,S00_CompressorModulation,,percent1,,,,CompressorState,,UCH,0=off;1=Heating Prerun;4=Heating;5=Heating Overrun;24=Hot Water;110=Defrosting,,,HeatingState,,UCH,8=off;9=Heating,,,field6,,UCH,,,,Defrost,,UCH,0=inactive;32=active,,,S00_CompressorPower,,percent1,,,
```

djd1983, 2024-02-20:
> @djd1983 It has been decoded (Status00), see here: [#330 (comment)](https://github.com/john30/ebusd-configuration/pull/330#issuecomment-1859292111)
> 
> ```
> *r,,,,,,"B511",,,,,,,,,,,,,,,,,,,,,,,,,
> r,,Status00,,,,,00,S00_SupplyTemp,,temp,,,,WaterPressure,,UCH,10,bar,,S00_CompressorModulation,,percent1,,,,CompressorState,,UCH,0=off;1=Heating Prerun;4=Heating;5=Heating Overrun;24=Hot Water;110=Defrosting,,,HeatingState,,UCH,8=off;9=Heating,,,field6,,UCH,,,,Defrost,,UCH,0=inactive;32=active,,,S00_CompressorPower,,percent1,,,
> ```

Ah great thank u. But some questions because I think I have to customize this code for my system:
-what ist the supply temp?
-For What kind of compressor are the Numbers for field 3, 4 and 8?

chrizzzp, 2024-02-20:
>  -what ist the supply temp?

It is the current (measured) flow temp of the supply, i.e. the temp the water is heated to by the heat pump

> -For What kind of compressor are the Numbers for field 3, 4 and 8?

Arotherm (split) heat pump compressor, supposed to work also with monoblock machines. The compressor modulation is also given in the 'Live Menu' of the display, the compressor power (or load, in %) relates to the bar on the left of the hydraulic station display (if present).

chrizzzp, 2024-02-20:
@djd1983 
In my case the flow temps of 'Status00' and 'Status01' are identical, just polled at different intervals (1 min vs. ~15 mins) that's why there are differences sometimes. If you superimpose them graphically you can check this better than comparing just the values.

> And I dont know why I have some data for an compressor, because I haven't one.

Good point... :-) so if you have a gas boiler they will probably represent some other value? I can only speak for the Arotherm heat pumps. It's alway good to visualize them to figure out what they could mean.

djd1983, 2024-02-21:
> It's alway good to visualize them to figure out what they could mean

I hope the following graphic will help:

[image]

In Status00 there are 7 fields. The first and the second one are the same like in ur config. FlowTemp and waterPressure. The next fields are like u see documentated. Feld = Field :)
Additional to this, I inserted the lines
green = returntemp
orange = FlowTemp
violett = Temp of Water in watertank

[image]

Some things were noticable:
Field 6 has obvious no function
Field 3 (blue) is maybe something like the amount of power? Because, as the temp of the water in the tank drops to 45°C(see violettt graphic) , the line went from 0 to 199 fast. After the heating was constant, the power decends slowly to minimum of the necessary amount. (I hope I interprete this understandable)

Have fun and thank you :P

djd1983, 2024-02-21:
The brightblue line -field7- seems to be something that indicates night and day mode. It turns from Daymode, always periodically from 129 to 131, at 22h to night mode, where it switches periodically between 1 and 3

chrizzzp, 2024-02-21:
@djd1983 OK... a gas boiler. I'm more familiar with heat pumps, so I'm probably not the right person to help...

However, for gas boiler systems you should consider also the parameters mentioned in the top post regarding:

gas, fans, pumps, ignition...

- field 3: yes, seems to be power related... (numerical value, UCH)
- field 4: a numerical (decimal) status code (UCH) (for the different states of the heater, have to be decoded)
- field 5: binary status info of several boiler components (bit0 = Fan, bit1 = Gas1, bit2 = Gas2, bit4 = HcPump etc. (see below, I think the spike is the actication of the Hwc pump(s)
- field 6: numerical (decimal) error status (UCH), always 0 in your case, so no error I guess...
- field 7: several binary-encoded states concerning Hc/Hcw demand, activation of time programs (BlockTime?)

`r,,FlowTempStatus,Status Heizkreis,,,,00,FlowTemp,,temp,,,,WaterPressure,,UCH,10,bar,,unknown4,,UCH,,,,Status,,UCH,,,,Fan,,BI0,1=Fan,,,Gas1,,BI1,1=Gas1,,,Gas2,,BI2,1=Gas2,,,HcPump,,BI3,1=HcPump,,,HwcPump1,,BI4,1=HwcPump1,,,HwcPump2,,BI5,1=HwcPump2,,,Ignition,,BI6,1=Ignition,,,Bit7,,BI7,1=Bit7,,,ErrorStatus,,UCH,,,,HcDemand,,BI0,1=HcDemand,,,BlockTime,,BI1,1=BlockTime,,,Bit2,,BI2,1=Bit2,,,Bit3,,BI3,1=Bit3,,,Bit4,,BI4,1=Bit4,,,Bit5,,BI5,1=Bit5,,,Bit6,,BI6,1=Bit6,,,HwcDemand,,BI7,1=HwcDemand,,`

So obviously 'Status00' is quite different for the different heating systems. Sorry for not being more helpful...  :-)

djd1983, 2024-02-22:
> Sorry for not being more helpful... :-)

u were very helpful. thank u for ur information :) I think studying, trying and connect to each other and talking about different issues are part of this big ebusd project. And ur details will help me to get the right values. :)

I will report when I found out more for my systemconfig.

djd1983, 2025-01-26:
Hey everybody, its nearly one year ago and I found some time to try to decode my Status00 in my gasboiler again. Maybe some of u have time to help to decode.

To record some values in my home assistant i gave the following Status00 in my 08.bai.csv:

`r,,,Status00,Vorlauftemperatur/Wasserdruck/Leistung/8Bit/1xHEX/1xHEX/8Bit,,,b511,00,Vorlauftemperatur,,D2C,,°C,,WaterPressure,,UCH,10,bar,,Leistung,,UCH,2,%,,verschStati1,,BI0;BI1;BI2;BI3;BI4;BI5;BI6;BI7,,,,PumpStatus,,HEX,,,,FehlerStatus,,UCH,,,,HcDemand,,BI0,1=HcDemand,,,BlockTime,,BI1,1=BlockTime,,,Bit2,,BI2,1=Bit2,,,Bit3,,BI3,1=Bit3,,,Bit4,,BI4,1=Bit4,,,Bit5,,BI5,1=Bit5,,,Bit6,,BI6,1=Bit6,,,HwcDemand,,BI7,1=HwcDemand,,`

To decode the 8 Bits in the 4th field, I will show u the following graphic:

[image]

parallel to that, the temperatures of hc and hwc:

[image]

My findings:
#unbekannt = unknown

unknown5: is "1" when hc is working
unknown7: is "1" when hwc is working, u can sie this parallel in the flowtemp and leistung (power) diagramm
unknown6: is "1" when when heating stops, but parallel its "1" too, when hwc in on
unknown3 and 4: they going to be "1" some minutes before unknown5 going to "0", so before hc is turning off.

@chrizzzp : in ur definition u have some "gasvalues" in this place. for what could they be?

unknown8 till 10 and 12 are always "0". 12 is maybe for errorcodes?
unknown11: any value of the water pump? when hwc is working it's 3F - DEC 63. kind of power in %?
unknown13: i gues is the activation for the timeschedule hc (night and daytime)
unknown14: 1= BlocktimeActive
unknown15-19: are always 0
unknown20: 0=hwc nightschedule; 1=hwc dayschedule

I know that looks weird, but maybe some of u have some ideas to this sttaus00. :)

chrisbushy, 2026-11-05:
Adding data from a similar setup, which may help close this one.

**Setup:** Vaillant ecoTEC plus 832 (LPG combi), `bai` SW0405 HW9602, polled by a V37/VR33-type OpenTherm→eBUS converter (master 7f, tado X on the OT side). Same message: `7f08b5110100 / 08cf020b0008000003`

**New this time:** I filmed the boiler LCD alongside the ebusd telegram log and checked the boiler's Live monitor, so byte 3 and the status bytes are calibrated against what the boiler itself displays.

| Byte | Meaning |
|---|---|
| 0-1 | Flow temp, D2C (/16) |
| 2 | unknown |
| 3 | Burner load, raw 0-138 (load % = byte3/138) |
| 4 | Status code S.x (decimal), active states |
| 5 | Component bits (fan / gas / pumps) |
| 6 | Error, always 0x00 |
| 7 | Demand / block bits |

**Evidence**

- **Bytes 0-1:** match Status01 flow within 0.2 °C.
- **Byte 2:** 0x0b / 0x0c idle or anti-cycle, 0x0e CH, 0x0f DHW / after DHW.
- **Byte 3:** 0x22 (34) at light-off = LCD flame bar about 1/4. 0x67 (103) about 3/4. 0x8a (138) during DHW = bar full.
- **Byte 4:** 0x02 = S.2 pump pre-run, 0x04 = S.4 burner on CH, 0x07 = S.7 pump overrun, 0x0e = S.14 burner on DHW, 0x11 = S.17, 0x00 idle.
- **Anti-cycle (confirmed):** byte 4 = 0x00 and byte 7 = 0x03 while the boiler's Live monitor showed "S.08 Heating mode: Anti-cycling time".
- **Byte 5:** 0x08 pump only, 0x0f fan + gas + pump (CH firing), 0x3f with DHW bits, 0x38 DHW ending. Agrees with @chrizzzp's fan/gas/pump idea.
- **Byte 6:** always 0x00, agrees with the earlier finding.
- **Byte 7:** 0x01 CH demand, 0x02 anti-cycle block (S.08 confirmed), 0x03 = both, 0x05 CH + DHW. Fits @djd1983's "BlockTime" observation. No 0x80 day/night bit here, probably because there's no VRC time programme on an OT setup.

**Example sequence** (heating start → DHW draw → back to heating):

```
20:28:19 08bf020b0008000003   idle, anti-cycle block
20:28:43 08de020e0002080001   S.2 pump pre-run
20:29:21 0844020e22040f0001   S.4 flame lit, load 34
20:35:35 08f0020e67040f0001   S.4 load 103
20:39:14 0819040f8a0e3f0005   S.14 DHW, load 138 (bar full)
20:39:38 0844040c0011380001   tap closed, flame off
20:40:39 08e9020f22040f0001   S.4 relit
20:41:15 08d9020f0007080002   S.7 pump overrun
21:54:50 08ca020c0008000003   Live monitor: S.08 anti-cycling time (confirmed)
```

**Still unknown:** byte 2.

**Suggested names:** `Status00` → `flowtemp` (D2C), `unknown` (UCH), `burnerload` (UCH), `statuscode` (UCH), `components` (bits), `error` (UCH), `demand` (bits: 0x01 CH demand, 0x02 block/anti-cycle, 0x04 DHW).

To others with Status00 data: does byte 4 match the S.x in your Live monitor? And is 138 your maximum byte-3 value during a hot water draw too, or does it vary by model?

_Decoded by Claude (Anthropic's AI assistant) from my ebusd telegram log, a video of the boiler display and Live monitor checks. Happy to log more or test a definition._