john30/ebusd-configuration #250: Protherm Skat support

moskovskiy82, 2022-01-11:
Have a protherm skat (ekectrical). No external thermostat (but getting one if it can help - advice needed on Thermolink)
The only needed option was to control the flowtempdesired setpoint.

$ebusctl f -d -v -c scan.08 id
scan.08 id = prefix=21;year=20;week=38;product=0010023648;supplier=3100;counter=013480;suffix=N1

```
$ ebusctl info
version: ebusd 21.3.v21.3
update check: OK, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.bai.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available
device: /dev/ttyUSB0, enhanced
access: *
signal: acquired
symbol rate: 23
max symbol rate: 96
min arbitration micros: 327
max arbitration micros: 1602
min symbol latency: 11
max symbol latency: 16
reconnects: 0
masters: 3
messages: 144
conditional: 0
poll: 0
update: 9
address 00: master #1
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0108;HW=7503", loaded "vaillant/bai.prothermskat.inc" ([PROD='0010023648']), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

bai.prothermskat.inc is a csv from the  https://github.com/john30/ebusd-configuration/pull/205
Only changed the naming for ExtFlowTempDesiredMin to a standard

Also modified the 08.bai.csv to include 
[PROD='0010023648']!load,bai.prothermskat.inc,,,

I have tried setting directly *changin the register to writable) but with no luck
ebusctl write -c bai flowtempdesired 65

Akso have included the solution from [https://github.com/john30/ebusd/issues/179](https://github.com/john30/ebusd/issues/179)

Adding the line to bai.prothermskat.csv

`wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,,,IGN:1,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,,,IGN:1,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2`

Afterwards

 $ ebusctl write -c bai SetModeOverride '55;55;-;-;0;0;0;0;0;0'
done

But still

```
 $ ebusctl read -f flowtempdesired
64.00
```

```
 $ ebusctl grab result all
3108070400 / 0ab5424149303001087503 = 1: scan.08
3108b5090127 / 094e313c3c3c3c3c3c3c = 48: scan.08 id
3108b5040100 / 0a00ffffffffffffff0080 = 10: bai DateTime
3108b5040116 / 00 = 10: bai Status16
3108b5110101 / 098200008000710100ff = 10: bai Status01
3108b5110102 / 060332a0466e6e = 10: bai Status02
3108b5110103 / 00 = 10: bai Status
3108b5100900376effff00000000 / 0100 = 4: BAI SetModeOverride
3108b503020001 / 0affffffffffffffffffff = 10: bai currenterror
3108b509030d0000 / 0100 = 10: bai externalHwcSwitch
3108b509030d0100 / 027003 = 11: bai HwcTemp
3108b509030d0200 / 03d90600 = 10: bai WaterPressure
3108b509030d0500 / 010f = 11: bai Flame
3108b509030d0c00 / 0101 = 10: bai ChangesDSN
3108b509030d0e00 / 0101 = 10: bai DCRoomthermostat
3108b509030d1000 / 01f0 = 10: bai VolatileLockout
3108b509030d1100 / 0100 = 10: bai TempGradientFailure
3108b509030d1200 / 0100 = 10: bai TempDiffBlock
3108b509030d1300 / 0100 = 10: bai TempDiffFailure
3108b509030d1500 / 02ffff = 10: bai HcPumpStarts
3108b509030d1700 / 03910300 = 10: bai StorageTemp
3108b509030d1800 / 03260400 = 11: bai FlowTemp
3108b509030d1e00 / 0101 = 10: bai OverflowCounter
3108b509030d2000 / 0100 = 10: bai DeactivationsTemplimiter
3108b509030d2100 / 0100 = 10: bai BlockTimeHcMax
3108b509030d2200 / 020000 = 10: bai HwcHours
3108b509030d2300 / 020000 = 10: bai HwcStarts
3108b509030d2800 / 020000 = 10: bai HcHours
3108b509030d2900 / 020000 = 10: bai HcStarts
3108b509030d2a00 / 0100 = 10: bai ACRoomthermostat
3108b509030d2e00 / 021502 = 11: bai ModulationTempDesired
3108b509030d3000 / 0100 = 10: bai HcUnderHundredStarts
3108b509030d3004 / 010a = 10: bai DSNOffset
3108b509030d3100 / 0100 = 10: bai HwcUnderHundredStarts
3108b509030d3104 / 029065 = 10: bai DSNStart
3108b509030d3400 / 0101 = 10: bai EbusSourceOn
3108b509030d3500 / 020000 = 10: bai maintenancedata_HwcTempMax
3108b509030d3600 / 020000 = 10: bai StorageTempMax
3108b509030d3700 / 020000 = 10: bai FlowTempMax
3108b509030d3800 / 0100 = 10: bai RemainingBoilerblocktime
3108b509030d3900 / 020004 = 15: bai flowtempdesired
3108b509030d3a04 / 0101 = 10: bai WarmstartDemand
3108b509030d3e00 / 0100 = 10: bai ExternalFaultmessage
3108b509030d3e04 / 0104 = 10: bai ProductionByte
3108b509030d3f00 / 0101 = 10: bai extWP
3108b509030d3f04 / 00 = 10: bai Night-timeOffset
3108b509030d4000 / 0100 = 10: bai HeatingDemand
3108b509030d4400 / 0101 = 10: bai WP
3108b509030d4900 / 01f0 = 10: bai Ignitor
3108b509030d4b00 / 0120 = 10: bai HwcTypes
3108b509030d5400 / 0100 = 10: bai PositionValveSet
3108b509030d5500 / 020000 = 10: bai HwcWaterflow
3108b509030d5600 / 020000 = 10: bai HwcWaterflowMax
3108b509030d5700 / 0100 = 10: bai HwcImpellorSwitch
3108b509030d5800 / 0100 = 10: bai HwcDemand
3108b509030d5b00 / 0101 = 10: bai WarmstartEnable
3108b509030d5e00 / 0102 = 10: bai BoilerType
3108b509030d6000 / 0103 = 10: bai ParamToken
3108b509030d6400 / 010a = 10: bai WPPostrunTime
3108b509030d6804 / 0100 = 10: bai SetFactoryValues
3108b509030d6b00 / 03e3ffaa = 10: bai expertlevel_ReturnTemp
3108b509030d6c00 / 010c = 10: bai PartloadHcKW
3108b509030d7000 / 0100 = 10: bai FloorHeatingContact
3108b509030d7100 / 0101 = 10: bai StoragereleaseClock
3108b509030d7400 / 0100 = 10: bai EBusHeatcontrol
3108b509030d7600 / 0339fcaa = 10: bai OutdoorstempSensor
3108b509030d7700 / 01f0 = 10: bai Templimiter
3108b509030d7b00 / 0100 = 10: bai CirPump
3108b509030d7f00 / 0101 = 10: bai EbusVoltage
3108b509030d8900 / 0100 = 10: bai FluegasvalveOpen
3108b509030d9800 / 050000ffff00 = 10: bai ReturnTemp
3108b509030d9900 / 0103 = 10: bai Testbyte
3108b509030d9a00 / 049a65659a = 10: bai DSN
3108b509030da000 / 0103 = 10: bai SecondPumpMode
3108b509030da500 / 020005 = 10: bai FlowsetHcMax
3108b509030daa00 / 061102640c640c = 10: bai PowerValue
3108b509030dab00 / 0104 = 10: bai Statenumber
3108b509030daf00 / 0101 = 10: bai WaterpressureBranchControlOff
3108b509030db300 / 0100 = 10: bai ReturnRegulation
3108b509030db700 / 0101 = 10: bai HcPumpMode
3108b509030db900 / 0101 = 10: bai AccessoriesOne
3108b509030dba00 / 0102 = 10: bai AccessoriesTwo
3108b509030dbb00 / 01f0 = 10: bai Gasvalve
3108b509030dbf00 / 0100 = 10: bai ExtStorageModulCon
3108b509030dc000 / 050020260268 = 10: bai PartnumberBox
3108b509030dd200 / 01f0 = 10: bai TemplimiterWithNTC
3108b509030dd300 / 01f0 = 10: bai VolatileLockoutIFCGV
3108b509030dd900 / 027003 = 10: bai HwcTempMax
3108b509030dda00 / 0102 = 10: bai DisplayMode
3108b509030ddc00 / 0100 = 10: bai InitialisationEEPROM
3108b509030dde00 / 0101 = 10: bai TimerInputHc
3108b509030de400 / 01f0 = 10: bai ExternGasvalve
3108b509030de500 / 0727131203020101 = 10: bai DCFTimeDate
3108b509030deb00 / 0100 = 10: bai SpecialAdj
3108b509030dec00 / 0102 = 10: bai AntiCondensValue
3108b509030ded00 / 0101 = 10: bai WPSecondStage
3108b509030df003 / 020000 = 10: bai FlowSetPotmeter
3108b509030df103 / 020000 = 10: bai HwcSetPotmeter
3108b509030df203 / 0101 = 11: bai HeatingSwitch
3108b509030df303 / 0101 = 10: bai HwcSwitch
3108b509030dfb00 / 02d007 = 10: bai PrimaryCircuitFlowRate
3108b509050e39001004 / 00 = 1: bai flowtempdesired
```

Any chances in my case or it's a dead end?

mmib, 2022-12-15:
Have the same boiler! It seems SetModeOverride '55;55;-;-;0;0;0;0;0;0' has too many params and in our case should be shorter,
something like '45.0;0.0;-;0.0;-;on'. I didn't try it yet, just an idea for experiment.

mmib, 2022-12-15:
https://www.pittnerovi.com/jiri/hobby/electronics/ebus/Vaillant_ebus.pdf
there is SetOperationMode (B5h 05h 02h)
seems we need to edit hcmode.inc and add it there, not familiar with this format yet

mmib, 2022-12-16:
This works [#170 ](https://github.com/john30/ebusd-configuration/issues/170#issuecomment-709592197)
Impossible to set temperature higher than set on display.

stevoh6, 2023-02-08:
> This works [#170 ](https://github.com/john30/ebusd-configuration/issues/170#issuecomment-709592197) Impossible to set temperature higher than set on display.

You keep the original SetMode command and add new one SetModeOverride?  Did yo ufind the solution to set values higher than on display?

mmib, 2026-02-02:
> BAI firmware characteristics:
> Does NOT support:
> Water pressure

This is NOT true, I have PROTHERM 12 KE / 14 RU,UA with BAE firmware and I can read water pressure

[image]

here is my alters to default config

/etc/ebusd/mqtt-hassio.cfg
> filter-name = status|temp|yield|power|runtime|hours|starts|mode|curve|^load$|^party$|sensor|pressure|press|mode|state

bai.0010008045.inc
> wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,setmode1,,UCH,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,setmode2,,UCH,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2

/etc/default/ebusd
> EBUSD_OPTS="--scanconfig --configpath=/etc/ebusd/ebusd-configuration/ebusd-2.1.x/en/ --device=enh:192.168.0.10:9999  --receivetimeout=70 --mqtthost='192.168.0.11' --mqttport=8883 --mqttuser='ebusd' --mqttpass='REDACTED' --mqttca=/etc/ssl/certs/ --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson --scanconfig=08"

mmib, 2026-02-20:
Do use it on linux