john30/ebusd-configuration #254: Vaillant eloblck current heating power and other parameters

AznKid45, 2022-02-18:
Hi!
I would like to take this opportunity to say a huge thank you to all the participants involved in this project.
I have a purchased circuit Vailant eloblock 14
It is currently defined as follows: 

> version: ebusd 22.1.28da22c
> device: /dev/ttyUSB1
> access: *
> signal: acquired
> symbol rate: 24
> max symbol rate: 133
> min arbitration micros: 112
> max arbitration micros: 359
> min symbol latency: 4
> max symbol latency: 10
> reconnects: 0
> masters: 2
> messages: 225
> conditional: 3
> poll: 19
> update: 10
> address 03: master #11
> address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0107;HW=7503", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
> address 31: master #8, ebusd
> address 36: slave #8, ebusd
> 

Correct current heating power:
`
r;wi,,PartloadHcKW,d.00 Heizungsteillast,,,,"6C00",,,power,,,Heizungsteillast
`
The following parameters are read (boiler and external temperature sensor not connected): 

> bai ACRoomthermostat = off
> bai averageIgnitiontime = 0.0
> bai BoilerType = 2
> bai ChangesDSN = 1
> bai CirPump = off
> bai CounterStartattempts1 = 0
> bai CounterStartattempts2 = 0
> bai CounterStartAttempts3 = 0
> bai CounterStartAttempts4 = 0
> bai currenterror = -;-;-;-;-
> bai DateTime = nosignal;-:-:-;-.-.-;-
> bai dcfState = nosignal
> bai DCRoomthermostat = on
> bai DeactivationsIFC = 0
> bai DeactivationsTemplimiter = 0
> bai DeltaFlowReturnMax = -0.06
> bai DisplayMode = 2
> bai DSN = 12511
> bai DSNOffset = 11
> bai DSNStart = 12500
> bai EbusSourceOn = on
> bai EbusVoltage = on
> bai expertlevel_ReturnTemp = -1.81;cutoff
> bai ExternalFaultmessage = off
> bai externalHwcSwitch = off
> bai ExternGasvalve = 240
> bai ExtStorageModulCon = no
> bai extWP = off | bai extWP = on
> bai FanHours = -
> bai FanMaxSpeedOperation = 11300
> bai FanMinSpeedOperation = 1400
> bai FanPWMSum = -
> bai FanPWMTest = -
> bai FanSpeed = 0
> bai FanStarts = -
> bai Flame = off
> bai FlameSensingASIC = 0
> bai FloorHeatingContact = off
> bai FlowsetHcMax = 80.00
> bai FlowTemp = 8.69;ok | bai FlowTemp = 9.56;ok
> bai FlowTempDesired = 0.50
> bai FlowTempMax = 0.00
> bai Fluegasvalve = off
> bai FluegasvalveOpen = off
> bai Gasvalve3UC = off
> bai Gasvalve = off
> bai GasvalveASICFeedback = off
> bai GasvalveUC = off
> bai HcHours = 0
> bai HcPumpStarts = -
> bai HcStarts = 0
> bai HcUnderHundredStarts = 0
> bai HeatingSwitch = on
> bai HwcDemand = no
> bai HwcHours = 0
> bai HwcImpellorSwitch = no
> bai HwcStarts = 0
> bai HwcSwitch = off
> bai HwcTemp = 0.00;ok
> bai HwcTypes = 32
> bai HwcUnderHundredStarts = 0
> bai HwcWaterflow = 0.00
> bai HwcWaterflowMax = 0.00
> bai Ignitor = off
> bai InitialisationEEPROM = no
> bai IonisationVoltageLevel = 0.0
> bai maintenancedata_HwcTempMax = 0.00
> bai maxIgnitiontime = 0.0
> bai minIgnitiontime = -
> bai ModulationTempDesired = 17.0 | bai ModulationTempDesired = 35.7
> bai OutdoorstempSensor = -60.44;cutoff
> bai OverflowCounter = yes
> bai ParamToken = 3
> bai PartloadHcKW = 5
> bai PartnumberBox = 00 20 26 02 68
> bai PositionValveSet = 0
> bai PowerValue = 11 02 64 0e 64 0e
> bai PrAPSCounter = -
> bai PrAPSSum = -
> bai PrEnergyCountHc1 = -
> bai PrEnergyCountHc2 = -
> bai PrEnergyCountHc3 = -
> bai PrEnergyCountHwc1 = -
> bai PrEnergyCountHwc2 = -
> bai PrEnergyCountHwc3 = -
> bai PrEnergySumHc1 = -
> bai PrEnergySumHc2 = -
> bai PrEnergySumHc3 = -
> bai PrEnergySumHwc1 = -
> bai PrEnergySumHwc2 = -
> bai PrEnergySumHwc3 = -
> bai ProductionByte = 4
> bai PrVortexFlowSensorValue = -1
> bai PumpHours = -
> bai PumpHwcFlowNumber = -
> bai PumpHwcFlowSum = -
> bai RemainingBoilerblocktime = 0
> bai ReturnTemp = 0.00;-;ok
> bai ReturnTempMax = 0.00
> bai SerialNumber = ff ff ff ff ff ff ff ff
> bai SetFactoryValues = no
> bai SetMode = auto;0.5;0.0;-;0;0;1;0;0;0
> bai SHEMaxDeltaHwcFlow = -0.06
> bai SHEMaxFlowTemp = -0.06
> bai Statenumber = 0 | bai Statenumber = 4
> bai Status01 = 8.5;0.0;-;0.0;-;off | bai Status01 = 14.0;0.0;-;0.0;-;on
> bai Status02 = on;50;80.0;70;70.0
> bai Storageloadpump = 0
> bai StorageLoadPumpHours = -
> bai StorageloadPumpStarts = -
> bai StorageTemp = -14.94;cutoff
> bai StorageTempDesired = 15.00
> bai StorageTempMax = 0.00
> bai TargetFanSpeed = 0
> bai TargetFanSpeedOutput = 0
> bai TempDiffBlock = 0
> bai TempDiffFailure = 0
> bai TempGradientFailure = 0
> bai Templimiter = off
> bai TemplimiterWithNTC = no
> bai TempMaxDiffExtTFT = -0.06
> bai Testbyte = 3
> bai TimerInputHc = on
> bai ValveStarts = -
> bai VolatileLockout = no
> bai VolatileLockoutIFCGV = no
> bai WarmstartDemand = no
> bai WaterHcFlowMax = -
> bai WaterPressure = 1.488;ok | bai WaterPressure = 1.461;ok
> bai WaterpressureBranchControlOff = on
> bai WaterpressureMeasureCounter = -
> bai WaterpressureVariantSum = -
> bai WP = off | bai WP = on
> bai WPSecondStage = off
> scan.08  = Vaillant;BAI00;0107;7503
> scan.08 id = 21;19;33;0010023657;3100;005492;N4

When the circuit is off (or standby) parameter Statenumber = 0 when on bai Statenumber = 4
I suspect that this parameter is not read correctly.

Plz tell me, is it possible to calculate the rest of the parameters on my circuit? And fix Statenumber and PartloadHcKW
Thanks

AznKid45, 2022-02-20:
Parameter write not working:

```
admin@homesrv01:~$ ebusctl r -f -v PartloadHcKW
bai PartloadHcKW power=5
admin@homesrv01:~$ ebusctl w -c bai PartloadHcKW 4
done
admin@homesrv01:~$ ebusctl r -f -v PartloadHcKW
bai PartloadHcKW power=5
```
Has anyone been able to change this setting?

Mitti10, 2022-02-22:
Hi, I see similar behaviour on my Vaillant Ecotec ("vaillant/bai.308523.inc") . On Write command I receive the "00" response that I assume means success. Strange is that the write operation on PartloadHcKW was working fine few days ago. I am investigating what could went wrong.

AznKid45, 2022-02-23:
Tell me, please, the "Statenumber" parameter, does it work for you normally?
As I understand it, it should take a value from 0 to 3, and for me, when it is on, it is 4

Mitti10, 2022-02-23:
Hi, for me "Statenumber" reflects the value of 'state code S.x' that  I can see on display on heater. As per documenation : S.0 - no need for heating, S.4 heating-fire on, S.31 Summer mode, etc. There are S.00-S.99 defined in my guide to heater.
So S=4 looks good when heating on in your system, I think.

AznKid45, 2022-02-24:
> Hi, for me "Statenumber" reflects the value of 'state code S.x' that I can see on display on heater. As per documenation : S.0 - no need for heating, S.4 heating-fire on, S.31 Summer mode, etc. [...]

Where did you get the description for your boiler? Thanks.

AznKid45, 2022-03-27:
I found a decryption for the Statenumber parameter, maybe it will come in handy for someone:

> 0=No heat demand
> 4=Heating on
> 7=Pump overrun
> 31=No heat demand, summer operating mode
> 34=Heating mode: frost protection

stalniy, 2023-07-27:
@AznKid45 where did you find that information? I also have the same heater, and trying to get more or  less useful details from ebusd :)

AznKid45, 2023-07-28:
> 

Hi. I found this information in the instruction manual. Status codes section

stalniy, 2023-07-28:
Can you please share whether you was to control it successfully share your config or at least where you stopped and what they main blockers?

I saw messages of people complaining that they can’t  set flow temperature above temperature on heater. 

 as far as I see this a challenging task. I’ve just started and need to learn/understand that message format.

stalniy, 2023-07-28:
In short, does this effort worth time spent investigating stuff? Are you satisfied with results?

stalniy, 2023-08-08:
I created an instruction for integration of Eloblock with ebusd, it's in Ukrainian -> https://gist.github.com/stalniy/5ec118d50f7b8c0a87514e9085418b64

It allowed me to read a lot more parameters from my boiler rather than with standard autodetection rules

Pavel-Demidyuk, 2024-11-30:
> I created an instruction for integration of Eloblock with ebusd, it's in Ukrainian -> https://gist.github.com/stalniy/5ec118d50f7b8c0a87514e9085418b64
> 
> It allowed me to read a lot more parameters from my boiler rather than with standard autodetection rules

@stalniy, thank you for sharing the instructions, I have the same bolier and now a lot more parameters are displayed in HA. have you managed to create a thermostat based on the available parameters?

stalniy, 2024-12-01:
I went a different way, I haven’t created thermostat in HA terminology. But created a dashboard to control heating. 

I have a collector based heating system and  configured it to work separately for every room in house. So, I have HA thermostat per room and all these thermostats change  heater’s behavior based on max difference between desired and current temp in a room. Higher difference - higher flow temp to quicker warm out the house/room. Additionally if current temp in a room >= desired one, the valve on collector automatically closes and excludes that room from heating group

Pavel-Demidyuk, 2024-12-01:
> I went a different way, I haven’t created thermostat in HA terminology. But created a dashboard to control heating.
>
> I have a collector based heating system and configured it to work separately for [...]

Thanks, @stalniy! Are you using SetModeOverride for changing flow temperature? Do you write any other ebus parameters?

stalniy, 2024-12-01:
Yes, you need to call it every few minutes in order for settings to be used by heater. Otherwise settings are reverted to what was set on display. Also you can’t set heating temperature over ebus higher than the value set on heater’s display. 

you can change all writable parameters, eg partload (partial load).

Gfermoto, 2025-01-08:
I spent a long time choosing the settings for the boilers **Vaillant eloBLOCK & atmoTEC** and decided to draw up the results and share them. https://github.com/Gfermoto/Vaillant/

Gfermoto, 2025-03-16:
> > Я пошел другим путем, не стал создавать термостат в терминологии HA. А создал панель управления отоплением.
> > У меня есть система отопления на основе коллектора, и я настроил ее на работу отдельно для каждой комнаты в доме. Итак, у меня есть термостат HA на комнату, и все эти термостаты изменяют поведение обогревателя на основе максимальной разницы между желаемой и текущей температурой в комнате. Больше разница - больше температура потока, чтобы быстрее прогреть дом/комнату. Кроме того, если текущая температура в комнате >= желаемой, клапан на коллекторе автоматически закрывается и исключает эту комнату из группы отопления
> 
> Спасибо,[@stalniy](https://github.com/stalniy)! Используете ли вы SetModeOverride для изменения температуры потока? Пишете ли вы какие-либо другие параметры ebus?

Увы. Я продвинулся в парсинге ошибок но еще дале от завершения
Вот в частности одна из исправленных строк:
r,,DCFTimeDate,DCF_TimeDate_DK,,,,E500,,,HEX:7,,,DCF Time / date stamp struct { uchar sec; uchar min; uchar std; uchar tag; uchar mon; uchar wtg; uchar jah; } Tdcf77_time;