john30/ebusd-configuration #258: vaillant/15.700.csv wird nicht mehr geladen

msfox, 2022-06-19:
Ich habe seit gut 08.06. 11:25 (laut FHEM-Log) das Problem, dass von der Vaillant bestimmte Werte nicht mehr gelesen werden können.
>ebusctl i

liefert nun:

> address 03: master #11
>address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0305;HW=0403", loaded "vaillant/08.hmu.csv"
>address 10: master #2
>address 31: master #8, ebusd
>address 36: slave #8, ebusd
>address 71: master #9
>address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0305;HW=0403"

hier fehlt nun plötzlich:
> address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"

--
Wie kann ich diese wieder ermitteln?

reboot half nicht.
 
Was ist 08.06. 11:25 geschehen:
- Ich habe das LAN-Kabel vom Ebus-RPi am Switch umgesteckt, weil ich den RJ45-Port für einen 2. Homematic-Gateway braucht. Nur sollte es doch den ebus-Service egal sein, wo der RPi am Switch reingesteckt ist. Der Ebus wird ja via GPIO (Platine aus dem FHEM-Forum) abegriffen und hat mit dem Netzwerk nix zu tun.

mf76130, 2022-08-30:
Hi,
Hast Du eine Lösung gefunden?

bei mir ädt es die 700 auch nur ab und zu.

mfG
Michael

msfox, 2022-08-30:
Nein, leider nicht. Irgendwann war mal Stromausfall, da waren RPi und Wärmepumpe vom Strom getrennt. Danach wurden wieder alle Werte gelesen (also vaillant/15.700.csv geladen). Nach einem weiteren Shutdown vom RPi wurde die vaillant/15.700.csv dann wieder nicht gefunden bzw. verwendet. Ich hatte am Wochenende an der WP ein Reset gemacht und beim RPi einen Reboot, aber das hat nicht geholfen. Ich will jetzt auch nicht, nur zum Versuch, die Wärmepumpe ständig hart (über Sicherung) vom Strom nehmen, nur damit ebus vielleicht mal wieder die vaillant/15.700.csv lädt. Ist ja keine Windows, was man immer mal Neustarten muss :).

msfox, 2022-09-04:
So, ich habe jetzt die WP und den RPi mit ebus noch einmal komplett vom Strom genommen und erst die WP und dann der RPi wieder in Betrieb genommen. Ich brauchte den Wert für den Gesamtstromverbrauch, da ich dies direkt an der WP nicht so genau finde.

`address ec: slave, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603"`

Wird jetzt gefunden. Fraglich bleibt immer noch, warum das nicht immer der Fall ist.

mf76130, 2022-09-14:
Ich hab jetzt scanconfig aus der config entfernt und nur noch die csv's die ich brauche im lokalen ordner jetzt gehts auch bei mir.
Und ebusd erkennt die Werte laufend, kein neustart der Wärmepumpe mehr notwendig.

john30, 2022-09-18:
[ebusd 22.4](https://github.com/john30/ebusd/releases/latest) is out and it continues to try scanning devices until successful, maybe that helps