john30/ebusd-configuration #263: Heatpump VWL 75/6 ebusd finds only a few values

rivertechnology, 2022-07-30:
Hi, Thanks for all the good work and many information about the ebusd software. Unfortunately I only get a few values form my ebus.

Installed hardware:
Heatpump aroTHERM VWL75/6 (06.07)
ControlUnit VWZ AI (02.02)
Thermostat  sensoCOMFORT VRC 720/2 (05.14)
ERERA eBus Coupler USB at RPi4 buster

**ebusctl info**

```
version: ebusd 22.3.p20220508
update check: revision v22.3 available
device: /dev/ttyUSB0
signal: acquired
symbol rate: 31
max symbol rate: 135
min arbitration micros: 798
max arbitration micros: 2538
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 4
messages: 70
conditional: 0
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"
```

**ebusctl find**

```
broadcast datetime = no data stored
broadcast error = no data stored
broadcast hwcStatus = no data stored
broadcast id = no data stored
broadcast id = no data stored
broadcast load = no data stored
broadcast outsidetemp = 23.312
broadcast signoflife = no data stored
broadcast vdatetime = 12:50:59;30.07.2022
general valuerange = no data stored
hmu ConsumptionThisYear10 = no data stored
hmu ConsumptionThisYear11 = no data stored
hmu ConsumptionThisYear12 = no data stored
hmu ConsumptionThisYear1 = no data stored
hmu ConsumptionThisYear2 = no data stored
hmu ConsumptionThisYear3 = no data stored
hmu ConsumptionThisYear4 = no data stored
hmu ConsumptionThisYear5 = no data stored
hmu ConsumptionThisYear6 = no data stored
hmu ConsumptionThisYear7 = no data stored
hmu ConsumptionThisYear8 = no data stored
hmu ConsumptionThisYear9 = no data stored
hmu ConsumptionTotal = no data stored
hmu CurrentCompressorUtil = no data stored
hmu CurrentConsumedPower = no data stored
hmu currenterror = no data stored
hmu CurrentYieldPower = no data stored
hmu DateTime = no data stored
hmu EnergyCool = no data stored
hmu EnergyHc = no data stored
hmu errorhistory = no data stored
hmu Hours = no data stored
hmu HoursCool = no data stored
hmu HoursHc = no data stored
hmu SetMode = 3;-;-;130;1;1;1;1;0;0
hmu State = 33;37;193;128
hmu Status01 = 70.5;66.0;-;-;-;hwc
hmu Status02 = no data stored
hmu Status16 = no data stored
hmu Status = no data stored
hmu StatusCirPump = on
hmu WaterThroughput = no data stored
hmu YieldThisYear10 = no data stored
hmu YieldThisYear11 = no data stored
hmu YieldThisYear12 = no data stored
hmu YieldThisYear1 = no data stored
hmu YieldThisYear2 = no data stored
hmu YieldThisYear3 = no data stored
hmu YieldThisYear4 = no data stored
hmu YieldThisYear5 = no data stored
hmu YieldThisYear6 = no data stored
hmu YieldThisYear7 = no data stored
hmu YieldThisYear8 = no data stored
hmu YieldThisYear9 = no data stored
hmu YieldTotal = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.08  = Vaillant;HMU00;0607;5103
scan.08 id = no data stored
scan.15  = Vaillant;CTLV2;0514;1104
scan.15 id = 21;21;41;0020260913;0953;006999;N9
scan.76  = Vaillant;VWZIO;0202;0103
scan.76 id = 21;21;49;0010031645;0082;005184;N3
```

It seems that only vaillant/08.hmu.csv has been loaded, but i don't get many values.

Do I something wrong with my config? My /etc/default/ebusdebusd = EBUSD_OPTS="--scanconfig=full -d /dev/ttyUSB0 -p 8888 --httpport=8889 --htmlpath=/var/ebeus/html"

Or is there no config file available yet? Can I hulp with contribute on this issue?

Thanks in advance

mf76130, 2022-08-30:
Hi,

Did you already find a solution?
I have a simialr issue.
I have following devices, + 2 more that are not used
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0306;HW=8802"
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"

Sometimes the csv is loaded, sometimes not.
most of the times it loaded for the HMU01 and i get data from it.
today after a reboot of the heatpump i got the first time the csv for 7000 loaded and i seem to get data.

thats how my config looks like
EBUSD_OPTS="--scanconfig  --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtthost=192.168.0.17 --mqttport=1883 --mqttuser=USER --mqttpass=passwort --mqttjson"

rivertechnology, 2022-08-30:
Hi,

Yes, I found a solution. But in your config I don't see any device configured like in my case an USB ebus device: --scanconfig=full -d /dev/ttyUSB0?

My thermostat: VRC720 was not recognized, so I manually cloned this configuration to /etc/ebusd : [GitHub - janphkre/ebusd-configuration at feature/support-ctlv2](https://github.com/janphkre/ebusd-configuration/tree/feature/support-ctlv2)

and added the local configpath the config

EBUSD_OPTS="--configpath=/etc/ebusd/ebusd-configuration/ebusd-2.1.x/en/ -d /dev/ttyUSB0 -p 8888 -l /var/log/ebusd.log --scanconfig=full --httpport=

then i got:

root@raspberrypi:/etc/ebusd# ebusctl info
version: ebusd 22.3.p20220508
update check: revision v22.3 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.hmu.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/ec.sol.sc.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available, vaillant/iotestbsol.inc: different version available, vaillant/yield3f40.inc: different version available
device: /dev/ttyUSB0
signal: acquired
symbol rate: 43
max symbol rate: 184
min arbitration micros: 610
max arbitration micros: 5916
min symbol latency: 2
max symbol latency: 13
reconnects: 0
masters: 4
messages: 562
conditional: 0
poll: 1
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0514;HW=1104", loaded "vaillant/ec.sol.sc.csv"

mf76130, 2022-08-30:
thanks for the quick answer, i loaded files now local as well, got a lot errors from broken files, removed them from the folder and now i run it with this config
EBUSD_OPTS="--configpath=/etc/ebusd/ -d /dev/ttyUSB0 -p 8888 --scanconfig=full  --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtthost=192.168.0.17 --mqttuser=user --mqttpass=pass --mqttjson"

but still it is not finding address 15

It anyhow only can find any address when i restart my heatpump. when the heatpump is running and i start ebusd or restart it, its not finding any address

version: ebusd 22.3.p20220508
device: /dev/ttyUSB0
signal: acquired
symbol rate: 61
max symbol rate: 126
reconnects: 0
masters: 4
messages: 15
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9
address e8: slave

mf76130, 2022-08-30:
> Did you already adjust the potentiometer?
>
> Heizungsdaten für den ioBroker auslesen - ebus Koppler | verdrahtet.info
> [4K] - YouTube <https://www.youtube.com/watch?v=5X_OUs1aTls&t=375s> @9m45?

thats what i am testing at the moment, but wow, thats very sensitive :-)

mf76130, 2022-08-30:
thats what i get when i am on the felt best adjustment: (latency set to 15)

2022-08-30 14:46:47.839 [bus info] scan 15 cmd: 3115070400
2022-08-30 14:46:47.852 [bus debug] start request 31
2022-08-30 14:46:47.852 [bus debug] arbitration start with 31
2022-08-30 14:46:47.900 [bus debug] arbitration won
2022-08-30 14:46:47.900 [bus debug] arbitration delay 403 micros
2022-08-30 14:46:47.900 [bus debug] switching from ready to send command
2022-08-30 14:46:47.907 [bus debug] send/receive symbol latency 6 ms
2022-08-30 14:46:47.914 [bus debug] send/receive symbol latency 6 ms
2022-08-30 14:46:47.921 [bus debug] send/receive symbol latency 6 ms
2022-08-30 14:46:47.928 [bus debug] send/receive symbol latency 6 ms
2022-08-30 14:46:47.928 [bus debug] switching from send command to send command CRC
2022-08-30 14:46:47.935 [bus debug] notify request: ERR: wrong symbol received
2022-08-30 14:46:47.935 [bus debug] ERR: wrong symbol received during send command CRC, switching to skip
2022-08-30 14:46:47.935 [main error] scan config 15: ERR: wrong symbol received

i measuered 15v to 21v on the bus

jkrauss82, 2022-08-31:
If I understand the conversation correctly, the observed issues might come from a suboptimal ebus adapter.

However, I would like to point to this related issue on the ebusd main repo: https://github.com/john30/ebusd/issues/437#issuecomment-947070000

The current CSV for Vaillant heatpumps in the ebusd-configuration repo does not deliver many values present in the CSV of the link above. I have tried that locally and got it to work with some adaptions (CSV contains template values such as "onoff", these have to be replaced by built-in types).

Anyone looking for more data from their VWL */6 I recommend to use this CSV.

mf76130, 2022-08-31:
i am afraid that my 114€ esera adapter is not working as it needs for that system.
i even tried placing to the VRC 700, to overcame the cable distance, but still not able to always find it.
Lets wait on an answer from Esera Support

mf76130, 2022-08-31:
stupid question, could the usb cable between the esera and the pi cause these issues as well?
are there any special recommended cables

jkrauss82, 2022-08-31:
As you already get some readings through the USB cable I would say it is highly unlikely that this is the issue here. But surely it does not hurt to try another cable either.

Have you checked your ebus wiring? I can imagine there could be problems if the connection is not done parallel for instance. Also check the voltage of the connection to your ebus adapter and at other devices and see if there is a difference.

mf76130, 2022-08-31:
I am using 1,5mm² installation wire to connect the ebus adapter, not drilled, could that cause it? But thats only like 30cm long.
I tried placing it parallel to the VRC 700, no change.
I tried 3 different usb cables already (ordered possible better ones today) 2 of the 3 worked, one caused errors.
Voltage is between 15V and 21V on the adapter, on the heatpump itself it is the same.
I placed the VR920 i have behind the cabeling of the ebus adapter and that still works fine.
I don't get it why that esera can be too slow, is the vaillant system the fastest on the market? :-)

mf76130, 2022-09-15:
i got it working now, i am using local files and have only the needed files local available and did remove scanconfig from the configuration.
best, you don't need to restart the heatpump anymore, just the ebusd service.

rivertechnology, 2022-09-17:
> If I understand the conversation correctly, the observed issues might come from a suboptimal ebus adapter setup.
> 
> However, I would like to point to this related issue on the ebusd main repo: [john30/ebusd#437 (comment)](https://github.com/john30/ebusd/issues/437#issuecomment-947070000)
> 
> The current CSV for Vaillant heatpumps in the ebusd-configuration repo does not deliver many values present in the CSV of the link above. I have tried that locally and got it to work with some adaptions (CSV contains template values such as "onoff", these have to be replaced by built-in types).
> 
> Anyone looking for more data from their VWL */6 I recommend to use this CSV.

Thanks for sharing this file. Fantastic work to make it possible to read all the info from the heatpump. 
Can you explain how I can tell ebusd to load this file? I've tried to replace my current 08.hmu.csv file with the files of yours, but I don't get it to work.

mf76130, 2022-09-17:
starting ebusctl yourconfig --checkconfig
will show you the errors, then you can work on it

rivertechnology, 2022-09-17:
Thanks for the quick answer. For me it's hard to understand and to use the ebusd commands. After trying to read the wiki as good as possible I'm still not understand the command you suggest.

This is my ebusd config: EBUSD_OPTS="--scanconfig --configpath=/home/pi/config/ebusd-configuration/ebusd-2.1.x/en -d /dev/ttyUSB0 --latency=20000 -p 8888 -l /var/log/ebusd.log --httpport=8889 --htmlpath=/var/ebusd/html"

I've uploaded the 08.hmu00.csv file at the /home/.../en config folder

After a ebusd service restart 'ebusctl info' shows

```
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

When i delete the 08.hmu00.csv file and restart ebusd I get:

```
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

My questions is that I don't understand ebusd don't want to load the HMU00 configfile and how can I ebusd forces to load the this file?

mf76130, 2022-09-17:
try it without --scanconfig, then it loads all files automatic, but you won't see it next to the devices, but when the device appears later it still uses the csv files.

rivertechnology, 2022-09-17:
Result at ebusctl info:
```
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 15: slave #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9
```
Also ebusctl find doesn't give readable parameters:
```
root@raspberrypi:~# ebusctl find
350 ActualPumpPower =  (ERR: invalid position for 3115b509030d0700 / 00)
350 ActualRoomTempDesired = no data stored
350 ActualTempDesired = no data stored
350 ActualWeekday = no data stored
350 BypassValve = no data stored
```

john30, 2022-09-18:
[ebusd 22.4](https://github.com/john30/ebusd/releases/latest) is out and it continues to try scanning devices until successful, maybe that helps

jkrauss82, 2022-09-23:
I just found that for the VWL 75/6 the definitions for item CurrentCompressorUtil in the shipped [08.hmu.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/08.hmu.csv) have to be adjusted slightly for datatype: instead of D1B it needs to be D2C.

I am still learning how to figure out the correct definitions but made some progress today. I hope to eventually be able to produce a combined version for the hmu00 from Yunkie's file and the current shipped csv, would create a pull request then.

TheJokerHA, 2022-11-28:
Hi,

I have also an VWL 75/6 from Vaillant and looking for more useful data that I can use (consumption and so on).
I just get this read out via MQTT and no consumption is received!
[image]

Ebusd is running on my Qnap NAS as an container! 

-scanconfig -d tcp:192.168.1.97:5000 --mqtthost=192.168.1.47 --mqttport=1883

Will consumption be integrated in a new john30/ebusd version, I have the latest at the moment v22.4.

BR

bknoors, 2022-12-31:
Hi,
I have the following setup:
VWL75/6 A 230V [scan.08  = Vaillant;HMU00;0607;5103]
VWZ MEH 97/6 [scan.76  = Vaillant;VWZIO;0607;5103]
VRC720 [scan.15  = Vaillant;CTLV2;0514;1104]

To get this to work I adjusted some data types (mainly from EXP to UIN) for the HMU00, so this file might be useful: 
[08.hmu00.csv](https://github.com/john30/ebusd-configuration/files/10327344/08.hmu00.csv)

Also minor changes to the 15.700.csv (and change of name to work with VRC720)
[15.ctlv2.csv](https://github.com/john30/ebusd-configuration/files/10327349/15.ctlv2.csv)

Thanks for the great work, very happy with all the data! 

BK

karusell, 2023-01-03:
Hello,
does anyone also have csv file for VWZ MEH 97/6? 
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

Whould be great to have transformation file csv for that, because there are some parameters which I need for further use.

Thank you.

JongsmaSimon, 2023-04-10:
a new 08.hmu.csv (English version) has been finalized by me. It works for Vaillant Arotherm Plus 125/6 and similar heat pumps. It makes available **all** info visible in the VWZ AI+ menu. If you are using a thermostat (besides VWZ AI+) not all entries in this csv will return a valid value (these will be accessible via the thermostat csv file).
This csv also contains three '**write**' entries that will (when **uncommented**) enable **changes** to be made directly from the Home Assistant UI.
Being: Heatcurve, TargetTempHwc & TargetTempHc
[08.hmu.csv english version ](https://github.com/john30/ebusd-configuration/files/11194651/08.hmu.csv)
[08.hmu.csv dutch version](https://github.com/john30/ebusd-configuration/files/11194663/08.hmu.csv)

tomvandervliet, 2023-08-26:
I want to thank you all for these files and how to use them. I just got my adapter on thursday, thanks to you I now get all the data from my VR720.

ErikCoen, 2023-09-15:
I have applied two files to my system:
08.hmu.csv (en)
15.ctlv2.csv
Both are working. However I do not see the any good readouts or nothing from my VWZ MEH 97/6 (address 76). Is the ec.sol.sc.csv correct (original file from repository)?

ebusctl info:
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0901;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0901;HW=5103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0514;HW=1104", loaded "vaillant/ec.sol.sc.csv"

My hardware setup:
  VWL125/6 A 400V
  VWZ MEH 97/6
  VRC720

AlexandreA74, 2024-01-23:
> I have applied two files to my system: 08.hmu.csv (en) 15.ctlv2.csv Both are working. However I do not see the any good readouts or nothing from my VWZ MEH 97/6 (address 76). What is the proper file? Any to share it to me?
> 
> Which device is the ec.sol.sc.csv? Is this file correct (original file from repository)?
> 
> ebusctl info: address 03: master #11 address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0901;HW=5103", loaded "vaillant/08.hmu.csv" address 10: master #2 address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv" address 31: master #8, ebusd address 36: slave #8, ebusd, scanning address 71: master #9 address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0901;HW=5103" address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0514;HW=1104", loaded "vaillant/ec.sol.sc.csv"
> 
> My hardware setup: VWL125/6 A 400V VWZ MEH 97/6 VRC720

I have a similar setup.  Did you get it working ?

ErikCoen, 2024-01-24:
Yes it is working perfectly. Of course I need time to get it running after
reading such posts (Tweakers.net - Verwarming uitlezen met ebus -
https://gathering.tweakers.net/forum/list_messages/2174564) / update xls
files several times.

Hereby my backup of current files/folders. There is also a readme.txt. I am
not sure if this readme is up to date. I think so.

That is for the configuration: VWL125/6 A 400V / VWZ MEH 97/6 / VRC720
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0901;HW=5103",
loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104",
loaded "vaillant/15.ctlv2.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103",
loaded "vaillant/76.vwzio.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0514;HW=1104", loaded
"vaillant/ec.sol.sc.csv"

ErikCoen, 2024-01-24:
[ebusd_240124.zip](https://github.com/john30/ebusd-configuration/files/14040315/ebusd_240124.zip)