john30/ebusd-configuration #268: Solar pump state

swenrengers, 2022-08-26:
Is it possible, to read the actual state of the solar pump?

Vaillant autoMatic VRS 620/3
I can't finde the corresponding value...

version: ebusd 22.3.p20220508
update check: revision v22.3 available
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 51
max symbol rate: 163
min arbitration micros: 300
max arbitration micros: 5642
min symbol latency: 0
max symbol latency: 23
reconnects: 0
masters: 3
messages: 737
conditional: 12
poll: 1
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0706;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0501;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/ec.solsy.sc.csv"

janphkre, 2022-08-26:
By looking at the configuration file (https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/ec.solsy.sc.csv) of the ec slave:
You could either read out the overall runtime of the pump by looking at the message _CollPumpHRuntime1_ / _CollPumpHRuntime2_, but that will only provide hourly granularity.
Otherwise the message _SolCollPumpED1_ / _SolCollPumpED2_ could be what you are looking for as it gives out a percentage at which the pump is actually running. However this might be not useful as it may not work on non-PWM based pumps or stay at a percentage even though the pump is off.
If none of the above message cover the use case you need, you may have to look at messages, that have not been decoded yet (https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#grab) and decode them yourself.

Nico184, 2022-08-28:
Maybe I can help:
I found the follwing document: [https://ebus-wiki.org/lib/exe/fetch.php/ebus/vaillant_ebus_v0.6.0.pdf](url)
Whith this I could decode som unkown massages und startet a file "scmode.inc" with a massage Status02
`r,,Status02,Statusabfrage,,,,21,CollectorTemp,,temp0,,,Kollektortemperatur,SolProtection,,temp0,,,Solarkreisschutz,HeaterSorce,,scmode,,,Status,StorageLevel,,percent0,,,Speicherfüllung,power,,percent0,,,ErtragsgÃ¼te
` THe Byte HeaterSource can appeat in 3 different values `scmode,UCH,0=off;1=solar;2=heater,,Solar Modus`. If It is 1, solarpump is working.