john30/ebusd-configuration #269: 15.700 PrEnergyXX wrong symbol - Energy Values missing

mf76130, 2022-09-15:
Hi,
I have a vrc 700 and am using the 15.700.csv file.
On all Energy values i get wrong symbol error.

Can someone give me hint how to get that fixed?

thanks a lot in advance
michael

Hardware:
Esera Ebus USB Adapter
Raspberry 3
[ebusd.log](https://github.com/john30/ebusd-configuration/files/9571718/ebusd.log)

mf76130, 2022-09-15:
i blanked out the energy paramters in the csv's and grabbed the new result

ebusctl grab result
1008b507010a / 037e8207 = 2
1008b5110100 / 09c502ff000000008000 = 76
1076b5040100 / 0a0310250715090422600f = 76
1076b5110101 / 09fffff600ff62000058 = 459
10feb505025c00 = 75
0015b503020002 / 0affffffffffffffffffff = 5
0076b503020001 / 0affffffffffffffffffff = 5
00e8b503020001 / 0affffffffffffffffffff = 5
1008b507020900 / 02bc03 = 76
1008b508020900 / 00 = 8
1008b5120204ff / 0101 = 76
1008b513020528 / 0101 = 8
1008b5100305ff03 / 0101 = 76
1076b512030f0201 / 07100300c5020b03 = 239
1076b512030f0202 / 0711030001030d03 = 221
1076b51303040d00 / 0a0d000300000003000300 = 76
7108b507030bffff / 0100 = 449
7108b51a03040535 / 0e0500000000000000000000000000 = 1
7108b51a0304c832 / 0ec807000080790000001b00000000 = 1
7108b51a0304c933 / 0ec900000000000000000000000000 = 1
7108b51a0304d834 / 0ed8cf030080003000000000001800 = 1
7108b51a0304e635 / 0ee600000000000000000000000000 = 1
7108b51a0304e832 / 0ee807000080790000001b00000000 = 1
7108b51a0304f633 / 0ef600000000000000000000000000 = 1
7108b51a0304f734 / 0ef7cf030080003000000000001800 = 1
0008b51608100000ff83050000 / 0b00000483052f2d80798849 = 12
0015b52406020000001600 / 06020016000100 = 247
0015b52406020002000600 / 050302060001 = 38
0015b52406020003000100 / 06030301000000 = 185
1076b51009000000ffffff010000 / 0101 = 460
03e8b512060b14010b0100 / 0be100000001ffffffffff00 = 1
03e8b512060b1f020e0200 / 0bfc01000001ffffffffff00 = 1
03e8b512060b1102010200 / 0be101000001ffffffffff00 = 1
03e8b512060b0000000000 / 0b0000000001ffffffffff00 = 267
03e8b5120a0c010000000028006401 / 0101 = 7
03e8b512060b1802070200 / 0bef01000001ffffffffff00 = 1
03e8b512060ba902050000 / 0b5d01000001ffffffffff00 = 1
03e8b512060b2b021a0200 / 0b2a02000001ffffffffff00 = 32
03e8b512060b2f021e0200 / 0b2902000001ffffffffff00 = 1
03e8b512060b2302120200 / 0b0a02000001ffffffffff00 = 1
03e8b512060b2802170200 / 0b1702000001ffffffffff00 = 1
03e8b512060b3d022b0200 / 0b3502000001ffffffffff00 = 1
03e8b512060b41022f0200 / 0b3a02000001ffffffffff00 = 1
03e8b512060b2c01230100 / 0bf200000001ffffffffff00 = 1
03e8b512060b3202210200 / 0b2d02000001ffffffffff00 = 1
03e8b512060b3902270200 / 0b3402000001ffffffffff00 = 1
03e8b512060b3602250200 / 0b3002000001ffffffffff00 = 1
03e8b512060b4d023b0200 / 0b4602000001ffffffffff00 = 1
03e8b512060b44013a0100 / 0b0301000001ffffffffff00 = 1
03e8b512060b51023f0200 / 0b4b02000001ffffffffff00 = 1
03e8b512060b4502330200 / 0b3f02000001ffffffffff00 = 1
03e8b512060b4902370200 / 0b4202000001ffffffffff00 = 1
03e8b512060b5d024a0200 / 0b5702000001ffffffffff00 = 1
03e8b512060b60024d0200 / 0b5a02000001ffffffffff00 = 1
03e8b512060b5502430200 / 0b4c02000001ffffffffff00 = 1
03e8b512060b5902460200 / 0b5302000001ffffffffff00 = 1
03e8b512060b5e01530100 / 0b5c01000001ffffffffff00 = 4
03e8b512060b6402510200 / 0b6a02000001ffffffffff00 = 2
03e8b512060b5b01500100 / 0b2001000001ffffffffff00 = 1
03e8b512060b6a02570200 / 0b6d02000001ffffffffff00 = 115
03e8b512060b6702540200 / 0b6d02000001ffffffffff00 = 2
03e8b512060b6400610000 / 0b6400000001ffffffffff00 = 11
03e8b512060b7301670100 / 0b3101000001ffffffffff00 = 1
03e8b512060b6900650000 / 0b6400000001ffffffffff00 = 1
03e8b512060b8a017e0100 / 0b4d01000001ffffffffff00 = 1
03e8b512060b7b00770000 / 0b6800000001ffffffffff00 = 1
03e8b512060b8f008a0000 / 0b7300000001ffffffffff00 = 1
03e8b512060ba3009e0000 / 0b8700000001ffffffffff00 = 1
03e8b512060b9f01920100 / 0b5901000001ffffffffff00 = 1
03e8b512060bb301a50100 / 0b7801000001ffffffffff00 = 1
03e8b512060bb900b30000 / 0b9a00000001ffffffffff00 = 1
03e8b512060bc501b70100 / 0b8701000001ffffffffff00 = 1
03e8b512060bce00c70000 / 0bac00000001ffffffffff00 = 1
03e8b512060bd601c70100 / 0b9a01000001ffffffffff00 = 1
03e8b512060be400dd0000 / 0bbe00000001ffffffffff00 = 1
03e8b512060be601d70100 / 0ba701000001ffffffffff00 = 1
03e8b512060bff01ef0100 / 0bc701000001ffffffffff00 = 1
03e8b512060bf301e40100 / 0bb801000001ffffffffff00 = 1
03e8b512060b0802f80100 / 0bd401000001ffffffffff00 = 1
03e8b512060bfc00f40000 / 0bd000000001ffffffffff00 = 1

edudele, 2022-10-10:
Hello @mf76130,
I follow you on several issues since we have a more or less equal environment and we both use essera USB.

> 2022-10-10 13:46:13.123 [bus error] poll hmu00 YieldThisYear11 failed: ERR: wrong symbol received
2022-10-10 13:46:19.068 [bus error] poll hmu00 YieldThisYear12 failed: ERR: wrong symbol received
2022-10-10 13:46:25.092 [bus error] poll hmu00 YieldThisYear1 failed: ERR: wrong symbol received
2022-10-10 13:46:31.099 [bus error] poll hmu00 YieldThisYear2 failed: ERR: wrong symbol received
2022-10-10 13:46:37.096 [bus error] poll hmu00 YieldThisYear3 failed: ERR: wrong symbol received
2022-10-10 13:46:43.130 [bus error] poll hmu00 YieldThisYear4 failed: ERR: wrong symbol received
2022-10-10 13:46:49.133 [bus error] poll hmu00 YieldThisYear5 failed: ERR: wrong symbol received
2022-10-10 13:46:55.102 [bus error] poll hmu00 YieldThisYear6 failed: ERR: wrong symbol received
2022-10-10 13:47:01.140 [bus error] poll hmu00 YieldThisYear7 failed: ERR: wrong symbol received
2022-10-10 13:47:07.115 [bus error] poll hmu00 YieldThisYear8 failed: ERR: wrong symbol received
2022-10-10 13:47:13.109 [bus error] poll hmu00 YieldThisYear9 failed: ERR: wrong symbol received
2022-10-10 13:47:19.058 [bus error] poll hmu00 YieldTotal failed: ERR: wrong symbol received
2022-10-10 13:53:25.121 [bus error] poll basv0 SolarYieldTotal failed: ERR: wrong symbol received
2022-10-10 13:53:37.119 [bus error] poll basv0 YieldTotal failed: ERR: wrong symbol received
2022-10-10 13:56:43.106 [bus error] poll hmu00 CurrentYieldPower failed: ERR: wrong symbol received
2022-10-10 13:57:37.133 [bus error] poll hmu00 YieldThisYear10 failed: ERR: wrong symbol received
2022-10-10 13:57:43.097 [bus error] poll hmu00 YieldThisYear11 failed: ERR: wrong symbol received
2022-10-10 13:57:49.120 [bus error] poll hmu00 YieldThisYear12 failed: ERR: wrong symbol received
2022-10-10 13:57:55.125 [bus error] poll hmu00 YieldThisYear1 failed: ERR: wrong symbol received
2022-10-10 13:58:01.091 [bus error] poll hmu00 YieldThisYear2 failed: ERR: wrong symbol received
2022-10-10 13:58:07.055 [bus error] poll hmu00 YieldThisYear3 failed: ERR: wrong symbol received
2022-10-10 13:58:13.096 [bus error] poll hmu00 YieldThisYear4 failed: ERR: wrong symbol received
2022-10-10 13:58:19.127 [bus error] poll hmu00 YieldThisYear5 failed: ERR: wrong symbol received
2022-10-10 13:58:25.119 [bus error] poll hmu00 YieldThisYear6 failed: ERR: wrong symbol received
2022-10-10 13:58:31.261 [bus error] poll hmu00 YieldThisYear7 failed: ERR: read timeout
2022-10-10 13:58:37.107 [bus error] poll hmu00 YieldThisYear8 failed: ERR: wrong symbol received
2022-10-10 13:58:43.095 [bus error] poll hmu00 YieldThisYear9 failed: ERR: wrong symbol received
2022-10-10 13:58:49.054 [bus error] poll hmu00 YieldTotal failed: ERR: wrong symbol received

have you made any progress?
Everything indicates that in the CSV files, there is something wrong.

mf76130, 2022-10-11:
VR900 which is registered at vaillant
https://www.wolf-online-shop.de/Vaillant-Internetkommunikationsmodul-VR-900-E-BUS::57479.html

edudele, 2022-10-11:
> VR900 which is registered at vaillant https://www.wolf-online-shop.de/Vaillant-Internetkommunikationsmodul-VR-900-E-BUS::57479.html

I am in your same situation, my VR-921 also asks a lot and I see a lot of data.