john30/ebusd-configuration #270: vaillant arotherm plus vwl 85/6 only load one template file

edudele, 2022-09-15:
Good,
After days of testing, ebusd identifies all the components.

**ebusctl find**
>root@debian-ha:/# ebusctl find | grep scan
scan id = no data stored
scan.08  = Vaillant;HMU00;0607;5103
scan.15  = Vaillant;BASV0;0217;8803
scan.36  = no data stored
scan.76  = Vaillant;VWZIO;0202;0103

I have the **local folder** configured and I have generated the files with the names of my components:

> root@debian-ha:/docker/ebusd/ebusd-configuration/ebusd-2.1.x/en/vaillant# ls -la
lrwxrwxrwx 1 root root    10 sep 15 21:08 08.hmu00.csv -> 08.hmu.csv
lrwxrwxrwx 1 root root    10 sep 15 21:07 15.basv0.csv -> 15.700.csv

The problem is that if I load the 08.hmu00.csv file it does not load the 15.basv0.csv I show two screenshots explaining said behavior:

**Load 08.hmu00.csv:**

> version: ebusd 22.3.p20220508
update check: revision v22.3 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 93
max symbol rate: 180
min arbitration micros: 83
max arbitration micros: 287
min symbol latency: 4
max symbol latency: 9
reconnects: 0
masters: 4
messages: 68
conditional: 0
poll: 23
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu00.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=BASV0;SW=0217;HW=8803"
address 31: master #8, ebusd, conflict
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

I delete the file **08.hmu00.csv and 08.hmu.csv** and it loads me the **15.basv0.csv**:

> version: ebusd 22.3.p20220508
update check: revision v22.3 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 93
max symbol rate: 180
min arbitration micros: 83
max arbitration micros: 287
min symbol latency: 4
max symbol latency: 9
reconnects: 0
masters: 4
messages: 68
conditional: 0
poll: 23
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=BASV0;SW=0217;HW=8803", loaded "vaillant/15.basv0.csv"
address 31: master #8, ebusd, conflict
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

Here the docker compose configuration of the container:
>version: '3'
services:
  ebusd:
    image: john30/ebusd 
    container_name: ebusd
    restart: unless-stopped
    privileged: true
    network_mode: host
    environment:
      - EBUSD_DEVICE=/dev/ttyUSB0
      - EBUSD_CONFIGPATH=/ebusd/ebusd-configuration/ebusd-2.1.x/en
      - EBUSD_LOGAREAS=all
      - EBUSD_LOGLEVEL=info
      - EBUSD_ACCESSLEVEL=*
      - EBUSD_MQTTHOST=192.168.1.251
      - EBUSD_MQTTPORT=1883
      - EBUSD_MQTTUSER=xxxxx
      - EBUSD_MQTTPASS=xxxx
      - EBUSD_MQTTJSON=
      - EBUSD_MQTTINT=/etc/ebusd/mqtt-hassio.cfg
      - EBUSD_MQTTCHANGES=            
    volumes:
      - /docker/ebusd/config:/ebusd/config
      - /etc/localtime:/etc/localtime:ro
    ports:
      - 8888:8888

**Why does the daemon have this behavior, is there a conflict between files?**

Now I have to choose which side to have visibility ¬¬

john30, 2022-09-17:
it is not a matter of either or but rather of whether the scan works, which is not the case for both devices according to the logs

edudele, 2022-09-18:
Good Morning,

so what could be happening?

john30, 2022-09-18:
probably the interface latency is an issue. try with increasing it e.g. use `--latency=50`

edudele, 2022-09-18:
After increasing the latency 50 in my environment variable:

> version: '3'
services:
  ebusd:
    image: john30/ebusd 
    container_name: ebusd
    restart: unless-stopped
    privileged: true
    network_mode: host
    environment:
      #- EBUSD_SCANCONFIG=
      - EBUSD_ADDRESS=02
      - EBUSD_DEVICE=/dev/ttyUSB0
      - **EBUSD_LATENCY=50**
      - EBUSD_CONFIGPATH=/ebusd/ebusd-configuration/ebusd-2.1.x/en
      - EBUSD_LOGAREAS=all
      - EBUSD_LOGLEVEL=info
      - EBUSD_ACCESSLEVEL=*
      - EBUSD_MQTTHOST=192.168.1.251
      - EBUSD_MQTTPORT=1883
      - EBUSD_MQTTUSER=xxxx
      - EBUSD_MQTTPASS=xxx
      - EBUSD_MQTTJSON=
      - EBUSD_MQTTINT=/etc/ebusd/mqtt-hassio.cfg
      - EBUSD_MQTTCHANGES=
    volumes:
      - /docker/ebusd:/ebusd
      - /etc/localtime:/etc/localtime:ro
    ports:
      - 8888:8888

The problem persists, it only loads the 08.hmu file:

> root@debian-ha:/# ebusctl i
version: ebusd 22.3.p20220508
update check: revision v22.3 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 42
max symbol rate: 169
min arbitration micros: 83
max arbitration micros: 228
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 5
messages: 68
conditional: 0
poll: 24
update: 10
address 02: slave, ebusd
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "**vaillant/08.hmu00.csv**"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=**BASV0**;SW=0217;HW=8803"
address 31: master #8
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

The logs and configuration remains the same.

Does anyone have anything else to check?

john30, 2022-09-18:
you could check with the john30/ebusd:devel version

edudele, 2022-09-18:
Hello,

chage to "john30/ebusd:devel":

> version: '3'
services:
  ebusd:
    image: **john30/ebusd:devel**
    container_name: ebusd
    restart: unless-stopped
    privileged: true
    network_mode: host
    environment:
      #- EBUSD_SCANCONFIG=
      - EBUSD_ADDRESS=02
      - EBUSD_DEVICE=/dev/ttyUSB0
      - EBUSD_LATENCY=50
      - EBUSD_CONFIGPATH=/ebusd/ebusd-configuration/ebusd-2.1.x/en
      - EBUSD_LOGAREAS=all
      - EBUSD_LOGLEVEL=info
      - EBUSD_ACCESSLEVEL=*
      - EBUSD_MQTTHOST=192.168.1.251
      - EBUSD_MQTTPORT=1883
      - EBUSD_MQTTUSER=xxx
      - EBUSD_MQTTPASS=xxx
      - EBUSD_MQTTJSON=
      - EBUSD_MQTTINT=/etc/ebusd/mqtt-hassio.cfg
      - EBUSD_MQTTCHANGES=
    volumes:
      - /docker/ebusd:/ebusd
      - /etc/localtime:/etc/localtime:ro
    ports:
      - 8888:8888

And recreate my container.

> root@debian-ha:/# ebusctl i
version: ebusd 22.4.v22.4-1-g7d8a871
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 43
max symbol rate: 168
min arbitration micros: 87
max arbitration micros: 273
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 5
messages: 15
conditional: 0
poll: 0
update: 4
address 02: slave, ebusd
address 03: master #11
address 08: slave #11, **scanning**
address 10: master #2
address 15: slave #2, **scanning**
address 31: master #8
address 71: master #9
address 76: slave #9, **scanning**

I get this log:
> 2022-09-18 21:17:04.603 [main error] invalid/unknown argument in env: address
2022-09-18 21:17:04.603 [main notice] ebusd 22.4.v22.4-1-g7d8a871 started with auto scan on device /dev/ttyUSB0
2022-09-18 21:17:04.603 [main info] loading configuration files from /ebusd/ebusd-configuration/ebusd-2.1.x/en/
2022-09-18 21:17:04.603 [main info] reading templates /
2022-09-18 21:17:04.604 [main info] read templates in /
2022-09-18 21:17:04.604 [main info] reading file broadcast.csv
2022-09-18 21:17:04.605 [main info] successfully read file broadcast.csv
[... 79 lines cut ...]
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 42
max symbol rate: 169
min arbitration micros: 83
max arbitration micros: 228
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 5
messages: 68
conditional: 0
poll: 24
update: 10
address 02: slave, ebusd
address 03: master https://github.com/john30/ebusd-configuration/pull/11
address 08: slave https://github.com/john30/ebusd-configuration/pull/11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu00.csv"
address 10: master https://github.com/john30/ebusd-configuration/pull/2
address 15: slave https://github.com/john30/ebusd-configuration/pull/2, scanned "MF=Vaillant;ID=BASV0;SW=0217;HW=8803"
address 31: master https://github.com/john30/ebusd-configuration/issues/8
address 71: master https://github.com/john30/ebusd-configuration/issues/9
address 76: slave https://github.com/john30/ebusd-configuration/issues/9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

My appreciation is that the identification data of each address is only received if the element is completely reset and advertised on the ebus.

john30, 2022-09-18:
why did you use 02 as address? that is not a valid master address and will certainly fail during arbitration. better use ff

edudele, 2022-09-18:
Use the address because it gave me conflict.

I just changed it to ff:

> version: '3'
services:
  ebusd:
    image: john30/ebusd:devel
    container_name: ebusd
    restart: unless-stopped
    privileged: true
    network_mode: host
    environment:
      #- EBUSD_SCANCONFIG=
      - EBUSD_ADDRESS=**ff**
      - EBUSD_DEVICE=/dev/ttyUSB0
      - EBUSD_LATENCY=50
      - EBUSD_CONFIGPATH=/ebusd/ebusd-configuration/ebusd-2.1.x/en
      - EBUSD_LOGAREAS=all
      - EBUSD_LOGLEVEL=info
      - EBUSD_ACCESSLEVEL=*
      - EBUSD_MQTTHOST=192.168.1.251
      - EBUSD_MQTTPORT=1883
      - EBUSD_MQTTUSER=edudele
      - EBUSD_MQTTPASS=villamanin
      - EBUSD_MQTTJSON=
      - EBUSD_MQTTINT=/etc/ebusd/mqtt-hassio.cfg
      - EBUSD_MQTTCHANGES=
    volumes:
      - /docker/ebusd:/ebusd
      - /etc/localtime:/etc/localtime:ro
    ports:
      - 8888:8888

But we continue with the same result recreate container:

> root@debian-ha:/# ebusctl i 
version: ebusd 22.4.v22.4-1-g7d8a871
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 99
max symbol rate: 152
min arbitration micros: 157
max arbitration micros: 166
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 5
messages: 15
conditional: 0
poll: 0
update: 4
address 03: master #11
address 04: slave #25, ebusd
address 08: slave #11, scanning
address 10: master #2
address 15: slave #2, scanning
address 31: master #8
address 71: master #9
address 76: slave #9, scanning
address ff: master #25, ebusd

Logs short:
> 2022-09-18 21:57:37.485 [main notice] ebusd 22.4.v22.4-1-g7d8a871 started with auto scan on device /dev/ttyUSB0
2022-09-18 21:57:47.492 [bus info] scan 08 cmd: ff08070400
2022-09-18 21:57:47.616 [main error] scan config 08: ERR: read timeout
2022-09-18 21:57:49.617 [bus info] scan 15 cmd: ff15070400
2022-09-18 21:57:49.722 [main error] scan config 15: ERR: read timeout
2022-09-18 21:57:51.722 [bus info] scan 36 cmd: ff36070400
2022-09-18 21:57:51.836 [main error] scan config 36: ERR: read timeout
2022-09-18 21:57:53.836 [bus info] scan 76 cmd: ff76070400
2022-09-18 21:57:53.944 [main error] scan config 76: ERR: read timeout
2022-09-18 21:58:55.946 [bus info] scan 08 cmd: ff08070400
2022-09-18 21:58:56.036 [main error] scan config 08: ERR: read timeout
2022-09-18 21:58:58.037 [bus info] scan 15 cmd: ff15070400
2022-09-18 21:58:58.144 [main error] scan config 15: ERR: read timeout
2022-09-18 21:59:00.145 [bus info] scan 36 cmd: ff36070400
2022-09-18 21:59:00.259 [main error] scan config 36: ERR: read timeout
2022-09-18 21:59:02.259 [bus info] scan 76 cmd: ff76070400
2022-09-18 21:59:02.379 [main error] scan config 76: ERR: read timeout
2022-09-18 22:00:14.555 [bus info] scan 08 cmd: ff08070400
2022-09-18 22:00:14.671 [main error] scan config 08: ERR: read timeout

After restarting the container and the machine differencial:

> root@debian-ha:/# ebusctl i
version: ebusd 22.4.v22.4-1-g7d8a871
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 166
max symbol rate: 166
min arbitration micros: 98
max arbitration micros: 246
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 5
messages: 68
conditional: 0
poll: 24
update: 10
address 03: master #11
address 04: slave #25, ebusd
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu00.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=BASV0;SW=0217;HW=8803"
address 31: master #8
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"
address ff: master #25, ebusd

log part when detection basv0 and not load file (15.basv0.csv -> 15.700.csv)

> 2022-09-18 22:05:19.221 [update info] received MS cmd: 3115070400 / 0ab5424153563002178803
2022-09-18 22:05:19.221 [bus notice] scan 15: ;Vaillant;BASV0;0217;8803
2022-09-18 22:05:19.221 [update notice] store 15 ident: done
2022-09-18 22:05:19.221 [update notice] received scan-read scan.15  QQ=31: Vaillant;BASV0;0217;8803
2022-09-18 22:05:19.398 [update info] received MS cmd: 1008b51009000000ffffff070000 / 0101
2022-09-18 22:05:19.398 [update notice] received update-write hmu00 SetMode QQ=10: auto;0.0;-;-;1;1;1;0;0;0
2022-09-18 22:05:19.530 [update info] received MS cmd: 3108070400 / 0ab5484d55303006075103
2022-09-18 22:05:19.530 [update notice] received scan-read scan.08  QQ=31: Vaillant;HMU00;0607;5103
2022-09-18 22:05:19.707 [update info] received MS cmd: 1076b51009000000ffffff050000 / 0101
2022-09-18 22:05:19.707 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2022-09-18 22:05:19.844 [update info] received MS cmd: 3115b52409020100003500160513 / 020000
2022-09-18 22:05:19.844 [update notice] received unknown MS cmd: 3115b52409020100003500160513 / 020000
2022-09-18 22:05:20.025 [update info] received MS cmd: 3115b52409020100003400120916 / 020000
2022-09-18 22:05:20.025 [update notice] received unknown MS cmd: 3115b52409020100003400120916 / 020000
2022-09-18 22:05:20.025 [bus info] poll cmd: ff08b516041a010700
2022-09-18 22:05:20.073 [bus error] poll hmu00 YieldThisYear7 failed: ERR: wrong symbol received

john30, 2022-09-18:
your interface can't write to the bus, that's the problem. can be proven rather simple: you'll not be able to actively query any message

edudele, 2022-09-18:
Then I have no choice but to wait... ¬¬

Adapter Reservierungsstatus

Dein Reservierungsstatus von 1 WIFI Adapter:
Bestätigt

Verfügbarkeit:
Warten auf Bauteile/Chips, voraussichtlich Ende 10/2022

edudele, 2022-09-19:
Essera device not write in bus.

closed 2022-09-19 by edudele