john30/ebusd-configuration #275: WOLF heater, looking for the correct config

mzsolnay, 2022-10-12:
Hi !

Need help/advice, where to find and how to use the right config file.
ebusctl info gives me this error:  
[bus notice] scan 08: ;65; &;0023;0001
[main error] scan config 08: ERR: element not found

Thank you for heping me !

stephan1827, 2022-10-12:
Did you follow this guide

https://github.com/john30/ebusd-configuration/issues/216

mzsolnay, 2022-10-13:
Hi Stephan,

Thank you for taking the time for my problem!

I did follow the guide and got stucked at this step 28, see below:

pi@raspberrypi4:/etc/ebusd $ ebusctl i
version: ebusd 22.4.v22.4
device: 192.168.0.200:3333
signal: acquired
symbol rate: 39
max symbol rate: 54
min arbitration micros: 5
max arbitration micros: 52
min symbol latency: 5
max symbol latency: 56
reconnects: 0
masters: 2
messages: 14
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=65;ID= &;SW=0023;HW=0001"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 50: slave, scanned "MF=65;ID= &;SW=0023;HW=0001"

pi@raspberrypi4:/etc/ebusd $ ebusctl scan full
done

pi@raspberrypi4:/etc/ebusd $ ebusctl scan results
ERR: invalid numeric argument

mzsolnay, 2022-10-14:
Gave it another try, this time I get this:

pi@raspberrypi4:~ $ sudo service ebusd start
pi@raspberrypi4:~ $ ebusctl scan full
done

pi@raspberrypi4:~ $ ebusctl scan results
ERR: invalid numeric argument

pi@raspberrypi4:~ $ ebusctl info
version: ebusd 22.4.v22.4
update check: OK, broadcast.csv: different version available, vaillant/23.ehp.cc.csv: different version available
device: 192.168.0.200:3333
signal: acquired
symbol rate: 40
max symbol rate: 61
min arbitration micros: 5
max arbitration micros: 14
min symbol latency: 5
max symbol latency: 74
reconnects: 0
masters: 2
messages: 79
conditional: 0
poll: 0
update: 6
address 03: master #11
address 08: slave #11, scanned "MF=65;ID= &;SW=0023;HW=0001"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 50: slave, scanned "MF=65;ID= &;SW=0023;HW=0001"

And get this in browser:
{
"broadcast": {
"messages": {
"betrd": {
"name": "betrd",
"passive": true,
"write": false,
"lastup": 1665715478,
"zz": 254,
"fields": {
"status": {
"value": 0
},
"zustand": {
"value": "00"
},
"stellgrad": {
"value": 0
},
"kesseltemp": {
"value": 24.5
},
"ruecklauftemp": {
"value": 24
},
"boilertemp": {
"value": 217
},
"aussentemp": {
"value": -59
}
}
},
"datetime": {
"name": "datetime",
"passive": true,
"write": false,
"lastup": 0
},
"error": {
"name": "error",
"passive": true,
"write": false,
"lastup": 0
},
"ident-u": {
"name": "ident",
"passive": true,
"write": false,
"lastup": 0
},
"signoflife": {
"name": "signoflife",
"passive": true,
"write": false,
"lastup": 0
},
"sollw": {
"name": "sollw",
"passive": true,
"write": false,
"lastup": 0
}
}
},
"cc": {
"messages": {
"CirPumpED": {
"name": "CirPumpED",
"passive": false,
"write": false,
"lastup": 0
},
"currenterror": {
"name": "currenterror",
"passive": false,
"write": false,
"lastup": 0
},
"errorhistory": {
"name": "errorhistory",
"passive": false,
"write": false,
"lastup": 0
},
"Mode": {
"name": "Mode",
"passive": false,
"write": false,
"lastup": 0
},
"Status": {
"name": "Status",
"passive": false,
"write": false,
"lastup": 0
},
"Timer.Friday": {
"name": "Timer.Friday",
"passive": false,
"write": false,
"lastup": 0
},
"Timer.Monday": {
"name": "Timer.Monday",
"passive": false,
"write": false,
"lastup": 0
},
"Timer.Saturday": {
"name": "Timer.Saturday",
"passive": false,
"write": false,
"lastup": 0
},
"Timer.Sunday": {
"name": "Timer.Sunday",
"passive": false,
"write": false,
"lastup": 0
},
"Timer.Thursday": {
"name": "Timer.Thursday",
"passive": false,
"write": false,
"lastup": 0
},
"Timer.Tuesday": {
"name": "Timer.Tuesday",
"passive": false,
"write": false,
"lastup": 0
},
"Timer.Wednesday": {
"name": "Timer.Wednesday",
"passive": false,
"write": false,
"lastup": 0
}
}
},
"feuerung": {
"messages": {
"hg01": {
"name": "hg01",
"passive": false,
"write": false,
"lastup": 0
},
"hg02": {
"name": "hg02",
"passive": false,
"write": false,
"lastup": 0
},
"hg03": {
"name": "hg03",
"passive": false,
"write": false,
"lastup": 0
},
"hg04": {
"name": "hg04",
"passive": false,
"write": false,
"lastup": 0
},
"hg06": {
"name": "hg06",
"passive": false,
"write": false,
"lastup": 0
},
"hg07": {
"name": "hg07",
"passive": false,
"write": false,
"lastup": 0
},
"hg08": {
"name": "hg08",
"passive": false,
"write": false,
"lastup": 0
},
"hg09": {
"name": "hg09",
"passive": false,
"write": false,
"lastup": 0
},
"hg10": {
"name": "hg10",
"passive": false,
"write": false,
"lastup": 0
},
"hg11": {
"name": "hg11",
"passive": false,
"write": false,
"lastup": 0
},
"hg12": {
"name": "hg12",
"passive": false,
"write": false,
"lastup": 0
},
"hg15": {
"name": "hg15",
"passive": false,
"write": false,
"lastup": 0
},
"hg16": {
"name": "hg16",
"passive": false,
"write": false,
"lastup": 0
},
"hg17": {
"name": "hg17",
"passive": false,
"write": false,
"lastup": 0
},
"hg21": {
"name": "hg21",
"passive": false,
"write": false,
"lastup": 0
},
"hg22": {
"name": "hg22",
"passive": false,
"write": false,
"lastup": 0
},
"hg73": {
"name": "hg73",
"passive": false,
"write": false,
"lastup": 0
},
"hg74": {
"name": "hg74",
"passive": false,
"write": false,
"lastup": 0
},
"hg75": {
"name": "hg75",
"passive": false,
"write": false,
"lastup": 0
},
"hg90": {
"name": "hg90",
"passive": false,
"write": false,
"lastup": 0
},
"hg91": {
"name": "hg91",
"passive": false,
"write": false,
"lastup": 0
},
"hg9x": {
"name": "hg9x",
"passive": false,
"write": false,
"lastup": 0
},
"pwm_pumpe": {
"name": "pwm_pumpe",
"passive": false,
"write": false,
"lastup": 0
},
"ruecklauf_ist": {
"name": "ruecklauf_ist",
"passive": false,
"write": false,
"lastup": 0
},
"vorlauf_ist": {
"name": "vorlauf_ist",
"passive": false,
"write": false,
"lastup": 0
},
"vorlauf_soll": {
"name": "vorlauf_soll",
"passive": false,
"write": false,
"lastup": 0
},
"warmwasser_ist": {
"name": "warmwasser_ist",
"passive": false,
"write": false,
"lastup": 0
},
"warmwasser_soll": {
"name": "warmwasser_soll",
"passive": false,
"write": false,
"lastup": 0
}
}
},
"global": {
"version": "22.4.v22.4",
"signal": true,
"symbolrate": 12,
"maxsymbolrate": 19,
"qq": 49,
"reconnects": 0,
"masters": 2,
"messages": 77,
"lastup": 1665715478
}
}