john30/ebusd-configuration #276: Saunier Duval Thema Condens + exaControl E7R C

dariuszszwajcok, 2022-10-12:
Witam. Posiadam kocioł saunier thema condens i sterownik exacontrol e7r c (radio). Moge odczytać parametry z kotła, jednak nie mogę załadować pliku dla sterownika exacontrol e7r c. 

fragment loga
2022-10-12 23:21:43.678 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "e7rc0", SW0208, HW7502

bash-5.1# ebusctl scan 15
ERR: wrong symbol received

bash-5.1# ebusctl info
version: ebusd 22.4.v3.16.2-200-ga8ca21f541
update check: revision v22.4 available
device: 192.168.0.171:9999
signal: acquired
symbol rate: 22
max symbol rate: 128
min arbitration micros: 2
max arbitration micros: 137
min symbol latency: 0
max symbol latency: 55
reconnects: 0
masters: 3
messages: 208
conditional: 0
poll: 73
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=1204;HW=6701", loaded "vaillant/bai.0010003886.inc" ([HW=6701]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7RC0;SW=0208;HW=7502"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning

bash-5.1# ebusctl scan result                                                                                                                                                                                                            
1 scan(s) still running                                                                                                                                                                                                                  
08;Vaillant;BAI00;1204;6701                                                                                                                                                                                                              
15;Vaillant;E7RC0;0208;7502

Czy mogę prosić o pomoc?

dariuszszwajcok, 2022-10-13:
bash-5.1# ebusctl grab result
1008b5070108 / 00 = 54
1008b51302050a / 0101 = 5

nyxnyx, 2023-01-02:
e7rc jest wspierany przez konfigurację
pokaż info