john30/ebusd-configuration #280: wrong configuration loaded?

michas79, 2022-10-23:
I have an ecoTEC plus VC DE 196/3-5 gas boiler. Unfortunately the wrong configuration file is loaded (bai.0010006101.inc instead of .bai.308523.inc). 
Is this an error or am I doing something wrong?

Unfortunately, in bai.0010006101.inc the flag for writing PrEnergyCountHc1 or PrEnergyCountHwc1 is not set. Is there a reason for this?

```
version: ebusd 22.4.p20221022
update check: revision v22.4 available
device: /dev/ttyebus
signal: acquired
symbol rate: 40
max symbol rate: 160
min arbitration micros: 8
max arbitration micros: 1085
min symbol latency: 2
max symbol latency: 11
reconnects: 0
masters: 3
messages: 451
conditional: 16
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=43000;SW=0215;HW=2002", loaded "vaillant/15.430.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

Thanks for your help.

michas79, 2022-11-29:
Is this the wrong place to ask for a solution to my problem? Or do I need to describe the facts differently? I am a bit confused, since no one has responded to my entry yet.

michas79, 2023-01-05:
Can I possibly get help somewhere else? I am really at a loss.

Silverstar, 2023-01-31:
I don't know which config is correct for your Bai, ebusd tries to detect the correct file for your device and afaik the 308523.inc is just a fallback if nothing else matches.
But what you could try is downloading the configuration from here, point your ebusd to that local config and put the lines regarding PrEnergyCountHc1 in the loaded .Inc, save, reload, check if you can read that value, profit.
If it is about writing that value, I don't know if that is possible. If so, edit the line if it is only "r,..." into "r;w,..." (semicolon) and test. Report back.