john30/ebusd-configuration #281: Vailant: Kühlung über ebusd einschalten

m-ask, 2022-11-01:
Hallo!
Ich habe folgendes Problem. 
Mittlerweile kann ich perfekt die Daten meine Vailant Heizung auslesen und Heizung/Warmwasser steuern.
Was ich leider nicht raus finde, ist wie ich die passive Kühlung über ebusd aktivieren kann.
Manuell über das Bedienteil wird doch die Anforderung auch über den EBUS übertragen?
Kann mir jemand dabei helfen?
Danke
m-ask

edsub, 2023-02-05:
Would like to know this as well (once I get my setup working ;-( )

hmax-72, 2023-02-22:
Hallo!
Wie steuerst du die Warmwasserbereitung genau? Welche Werte sind da zu setzen?
Ich möchte die Warmwasserbereitung gem. dem günstigsten stündlichen Strompreis steuern (ich kaufe zum stündlichen Börsenpreis).

m-ask, 2023-03-04:
Hallo hmax-72!
Beim Warmwasser kann ich  dzt. nur die Einschaltschwelle umstellen. Damit kannst du schon eine erste Steuerung machen.
Es gibt auch noch die einmalige Speicherladung nur funktioniert die Ansteuerung bei mir nicht. Da konnte mir noch keiner weiterhelfen. Die wäre aber für deinen Fall die beste Lösung, dann bräuchtest du nur beim billigen Tarif die Speicherladung starten.

Eraser-1, 2023-03-05:
Ich nutze die Boilerladung erfolgreich.
Dazu sende ich einen MQTT-Befehl mit dem Wert 6 auf ebusd/700/SFMode/set.

m-ask, 2023-03-11:
Hallo Erase-1!
SFMode hab ich auch schon probiert, das funktioniert leider bei meiner Steuerung nicht :-(