john30/ebusd-configuration #282: Calormatic 470/3 on ESERA USB at Raspberry Pi

physman38, 2022-11-03:
Hi,

just as summer is over, I start to come back on the ebusd.
I'm running a Vaillant system with

aurocompact VSC S 196/3-5 200 R3
Calormatic 470/3 installed in the heating system
ebusd on version 22.4
Bullyseye on Raspberry
Esera ebusd USB adapter (approx 3m apart from the Vaillant) and with this Poit adjustments

Poti adjustments are done several times, but the Esera finds only the BAI regulation (see below).
On the lograwdata it shows no signal lost and several "aa" and than some HEX codes.

root@raspberrypi:/home/raspberry.dirk# sudo systemctl restart ebusd
root@raspberrypi:/home/raspberry.dirk# ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyHZG
access: *
signal: acquired
symbol rate: 22
max symbol rate: 112
min arbitration micros: 581
max arbitration micros: 5998
min symbol latency: 4
max symbol latency: 11
reconnects: 0
masters: 3
messages: 225
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning

The logfile looks as:

2022-11-03 08:40:49.869 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:40:49.869 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:40:50.419 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.250;54.0;55.0;off
2022-11-03 08:40:51.869 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:40:52.409 [update notice] received unknown MS cmd: 1008b5110100 / 08e001140000080081
2022-11-03 08:40:52.635 [update notice] received update-read broadcast outsidetemp QQ=10: 6.250
2022-11-03 08:40:56.405 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:41:00.443 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.250;54.0;55.0;off
2022-11-03 08:41:01.981 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:41:01.981 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:41:02.430 [update notice] received read bai Status02 QQ=10: auto;60;50.0;80;55.0
2022-11-03 08:41:03.981 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:41:06.470 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:41:06.761 [mqtt notice] connection re-established
2022-11-03 08:41:10.449 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.250;53.0;55.0;off
2022-11-03 08:41:11.858 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:41:11.859 [mqtt error] communication error: A network protocol error occurred when communicating with the broker.
2022-11-03 08:41:12.457 [update notice] received read bai DateTime QQ=10: valid;08:41:12;03.11.2022;6.250
2022-11-03 08:41:12.859 [mqtt error] communication error: not connected
2022-11-03 08:41:13.859 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:41:16.518 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:41:20.492 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.438;53.0;55.0;off
2022-11-03 08:41:22.957 [update notice] received unknown MS cmd: 1008b5100305ff01 / 00
2022-11-03 08:41:23.859 [mqtt notice] connection re-established
2022-11-03 08:41:23.962 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:41:26.542 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:41:30.572 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.438;53.0;55.0;off
2022-11-03 08:41:36.082 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:41:41.421 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:41:41.669 [update notice] received update-read broadcast vdatetime QQ=10: 08:41:42;03.11.2022
2022-11-03 08:41:41.954 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.438;53.0;55.0;off
2022-11-03 08:41:42.210 [update notice] received update-write bai StatusCirPump QQ=10: on
2022-11-03 08:41:43.089 [mqtt error] communication error: A network protocol error occurred when communicating with the broker.
2022-11-03 08:41:43.540 [update notice] received read bai DateTime QQ=10: valid;08:41:44;03.11.2022;6.438
2022-11-03 08:41:43.810 [update notice] received read bai Status02 QQ=10: auto;60;50.0;80;55.0
2022-11-03 08:41:44.089 [mqtt error] communication error: not connected
2022-11-03 08:41:45.402 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:41:48.204 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:41:48.204 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:41:50.204 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:41:51.550 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.438;53.0;55.0;off
2022-11-03 08:41:55.086 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:41:55.090 [mqtt notice] connection re-established
2022-11-03 08:41:55.370 [update notice] received unknown MS cmd: 1008b5110100 / 08e001140000080081
2022-11-03 08:41:55.596 [update notice] received update-read broadcast outsidetemp QQ=10: 6.438
2022-11-03 08:42:00.300 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:42:01.123 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.438;53.0;55.0;off
2022-11-03 08:42:03.107 [update notice] received read bai Status02 QQ=10: auto;60;50.0;80;55.0
2022-11-03 08:42:05.147 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:42:07.307 [mqtt error] communication error: A network protocol error occurred when communicating with the broker.
2022-11-03 08:42:08.307 [mqtt error] communication error: not connected
2022-11-03 08:42:11.161 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.438;53.0;55.0;off
2022-11-03 08:42:12.430 [main error] scan config 15: ERR: wrong symbol received
2022-11-03 08:42:12.430 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:42:13.210 [update notice] received read bai DateTime QQ=10: valid;08:42:13;03.11.2022;6.438
2022-11-03 08:42:14.430 [mqtt error] publish: The client is not currently connected.
2022-11-03 08:42:15.208 [update notice] received update-write bai SetMode QQ=10: auto;34.0;60.0;-;0;0;0;0;0;0
2022-11-03 08:42:21.223 [update notice] received read bai Status01 QQ=10: 30.0;29.0;6.438;52.0;55.0;off
2022-11-03 08:42:23.200 [update notice] received unknown MS cmd: 1008b5100305ff01 / 00

Seems there are some wrong symbols - how can I fix that?
Furthermore there seems to be an mqtt error - how can I fix that?

physman38, 2022-11-03:
Is the Calormatic correctly conneced?
The Calormatic has 3 pins, which connect to the mainboard, which on the backside has 4 white plugs.
I guess / hope that those are the ebus pins, and they interconnect through the mainboard itself. I have connected to the ebus +/- pins called BUS in red only and nothing to the 4 white pins.
Is that correct?

[image]
[image]
[image]

john30, 2022-11-06:
no idea what these 3 wires are for. ask the manual in doubt and stick to eBUS named connectors for eBUS connections

physman38, 2022-11-06:
you mean the 3 pins?
Those are the connections for the 470/3 integrated in my heater.

As my 470/3 is not recognized in the scan, I wonder if I need to wire it up differently?
Or do I have some wrong settings?

physman38, 2022-11-06:
Sorry for double postings, but was unsure in FHEM forum that the message went out...

Here is a strange behaviour.
The ebusctl info delivered mostly:
 root@raspberrypi:/home/raspberry.dirk# ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyHZG
signal: acquired
symbol rate: 87
max symbol rate: 106
min arbitration micros: 574
max arbitration micros: 2695
min symbol latency: 2
max symbol latency: 8
reconnects: 0
masters: 3
messages: 225
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning

But suddenly (I believe I played a bit with the ebusd but unsure what I did to achive this):
root@raspberrypi:/home/raspberry.dirk# ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyHZG
access: *
signal: acquired
symbol rate: 41
max symbol rate: 98
min arbitration micros: 583
max arbitration micros: 2648
min symbol latency: 5
max symbol latency: 6
reconnects: 1
masters: 3
messages: 225
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanning
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning

You see that the slave 2 at adress 15 suddenly showed scanning, wheras before it was not listed at all and the ebusd.log delivered errors like:

physman38, 2022-11-07:
Interessant:

Habe den ebusd am laufen mit der BAI gefunden.
Dann die Therme einmal aus (ca 10-20sec gewartet) dann wieder an --> siehe da, der slave #2 auf adresse 15 ist wieder da und steht auf scanning...