john30/ebusd-configuration #285: Vaillant eloBlock set FlowTempDesired to higher value

BalintPogatsa, 2022-11-29:
Dear All,

I have a Vaillant eloBlock which I'm able to control as described in #242 (thanks for this tool and infos to everybody)
MF=Vaillant;ID=BAI00;SW=0109;HW=7503", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv

I'm able to change the FlowTempDesired to a lower value than set on the eloBlock device, by writing the SetModeOverride.
Do you know if there is a way that I set my device (directly with the knob on it) at 25 degrees or frost protection and to use SetModeOverride with a higher FlowTempDesired value?

I was experimenting with mode 2=heat option (via SetModeOverride first number) and FlowTempDesired was taken over the circulation pump switched on, but the flame was never turned on.

I would appreciate any suggestion. My goal is that if my automation fails and I'm away not to overheat my home...
Thanks in advance

BalintPogatsa, 2022-12-20:
I think your question is more related to other topics. Like the one I've linked in my first comment.
Just to summarize: you cannot write FlowTempDesired directly as it is read only. The solution is to edit your configuration and add the SetModeOverride writable value which can control indirectly the FlowTempDesired value.

pawelmuszynski, 2023-04-13:
Hello @BalintPogatsa 
Once you set SetModeOverride periodically each 10 mins your FlowTempDesired is taken from SetModeOverride, not from device settings. That's my observation. Once you stop setting SetModeOverride, the value from your device is taken after 20 mins. Does it work differently for you?

BalintPogatsa, 2023-04-25:
@pawelmuszynski 
The difference what I observe is that if I set the desired flow temp in SetModeOverride with a value higher than the value in my electric heater, my value is not taken over.

If I set a lower value in SetModeOverride compared to the desired flow temp in my electric heater device, it is accepted by the device.

If anybody has a hint, please let me know.

moskovskiy82, 2023-05-15:
> If anybody has a hint, please let me know.

Same as here https://github.com/john30/ebusd-configuration/issues/250#issuecomment-1354701376