john30/ebusd-configuration #286: ctlv0 support config file

1achy, 2022-12-07:
720 and 380 same family

https://user-images.githubusercontent.com/6684654/206010740-e4338efc-94cf-446e-9d57-ef2695e84454.jpg

ebusd find controller vrt380 : [update notice] sent scan-read scan.15 QQ=31: Vaillant;CTLV0;0313;9103

dont recognize csv automatic way

Copy all file:

https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/en/vaillant/

broadcast.csv
_templates.csv
scan.csv
08.bai.csv
... etc....

IN TO

home/pi/config

Rename https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.700.csv
in to 15.ctlv0.csv

I dont know i need change some info in file
disable auto config scan and set --configpath=home/pi/config/valliant
Restart service

IS IT CORRECT ????????

And can update reposity with supporto  15.ctlv0.csv      --->>   15.700.csv

1achy, 2022-12-08:
rename and work

BertVano, 2023-10-12:
> rename and work

Can you provide your config file for the VRT380?