john30/ebusd-configuration #287: --scanconfig option not working

juanjoqg, 2022-12-13:
Hi,

I've been ussing ebusd for months whitout any issue, but after my last reboot it can't get the csv files with the --scanconfig option.

All I get is this error:
2022-12-13 11:36:13.366 [main error] error reading config files from http://ebusd.eu/config/: ERR: element not found, last error:

Is there some change than I should do?.

Thanks.

Matt47989, 2022-12-17:
Hi - me too! I get the same error after months (years!) of no issues. I get the exact same error in ebusd.log. Can anyone out there help? My Rasp PI is internet connected (I can ping no issues).

Matt47989, 2022-12-17:
I think the config location has changed. See here: https://www.npmjs.com/package/iobroker.ebus

Matt47989, 2022-12-17:
Juan - I think I fixed mine.. I was on an old version and I think (as above) the config file location has changed. Without the config files, ebusd cannot interpret the ebus data so it's useless. I fixed my system by going to RELEASES (https://github.com/john30/ebusd/releases) and finding the appropriate latest version for my setup. I then went here (https://chipwired.com/install-deb-file-raspberry-pi/) to understand how to install it on my Rasp Pi. After installation, I checked it all and it is working just fine now :). Hope you get yours working too.

juanjoqg, 2022-12-17:
Thanks. I will try it.