john30/ebusd-configuration #289: Neue Vailant Platine in Gastherme - ebusctl info findet Therme nicht.

kratkale, 2022-12-14:
Hallo, 

ich habe leider gröbere Probleme mit meiner Vaillant Therme. atmoTEC plus VCW AT 174/4-5-H. Es ist am Ebus noch ein Thermostat Calormatic 370. 

Die Reparaturkosten übersteigen nun schon den Preis einer Neuen. Als letztes wurde nun auch noch die Platine in der Therme getauscht. Die Therme funktioniert aber immer noch nicht zuverlässig. 

Ich habe zuvor mit einem ESERA Ebus 2 USB und einem RPi die Daten auslesen können und mir auch eine Webseite gebastelt um die Temperatur der Therme einzustellen. Das funktioniert mit der neuen Platine nicht mehr. 

Versuche mit dem Ebus Adapter 3 liefern nun: 
===================================
ebusd -f -c /tmp --logareas bus --loglevel info -d /dev/ttyUSB0
ebusctl raw
raw logging enabled

2022-12-14 10:14:00.726 [bus notice] ...<fefefefefefefefefefffffcfefefefefefffefefefefefefefefefefefefe...
2022-12-14 10:14:02.357 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefffc7ffffefefefefefe...
2022-12-14 10:14:03.759 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2022-12-14 10:14:05.179 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...
2022-12-14 10:14:06.589 [bus notice] ...<fefefefefefefefefefefefefefefefefefefefefefefefefefefefefefefe...

root@CF19U2400:/home/kratkale# ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyUSB0
signal: acquired
symbol rate: 23
max symbol rate: 23
reconnects: 0
masters: 1
messages: 0
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd

mit dem ESERA ebus2USB Adapter
==============================
root@CF19U2400:/home/kratkale# ebusd -f -c /tmp --logareas bus --loglevel info -d /dev/ttyUSB0 

ebusctl raw

2022-12-14 10:26:54.966 [bus notice] bus started with own address 31/36
2022-12-14 10:26:54.971 [bus notice] signal acquired
2022-12-14 10:26:55.138 [bus notice] new master 10, master count 2
2022-12-14 10:26:55.207 [bus notice] new master 03, master count 3
2022-12-14 10:27:03.601 [bus notice] <1008b511010189000930320080ffff0000ff1d00
2022-12-14 10:27:05.712 [bus notice] <1008b50401003d000a00152711ffffffff00805a00
2022-12-14 10:27:05.986 [bus notice] <1008b51101028a0005033c7846827c00
2022-12-14 10:27:09.917 [bus notice] <1008b5100900000078ffff01ff00f80001019a00
2022-12-14 10:27:14.114 [bus notice] <1008b511010189000930320080ffff0000ff1d00
2022-12-14 10:27:18.316 [bus notice] <1008b5100900000078ffff01ff00f80001019a00
2022-12-14 10:27:24.587 [bus notice] <1008b511010189000930320080ffff0000ff1d00
2022-12-14 10:27:26.679 [bus notice] <1008b51101028a0005033c7846827c00

root@CF19U2400:/home/kratkale# ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyUSB0
signal: acquired
symbol rate: 22
max symbol rate: 41
reconnects: 0
masters: 3
messages: 0
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd

Wie soll ich da nun weiter machen???

DANKE
Alexander

john30, 2022-12-14:
`-d /dev/ttyUSB0` stimmt sicher nicht, denn der Adapter 3 verwendet ja enhanced protocol, also entweder muss es auf `-d enh:/dev/ttyUSB0` lauten oder auf `-d ens:/dev/ttyUSB0` je nach Einstellung am Adapter und Version (siehr [3.0 hier](https://adapter.ebusd.eu/v30/#usb) und [3.1 hier](https://adapter.ebusd.eu/#usb))

kratkale, 2022-12-14:
Hallo John, 

Danke für Deine Antwort. hier sind die Ausgaben mit enh und ens. Ich denke ich habe den 3.0. 

root@CF19U2400:/home/kratkale# ebusd -f -c /tmp --logareas bus --loglevel info -d enh:/dev/ttyUSB0 
2022-12-14 19:02:11.138 [bus notice] device status: resetting
2022-12-14 19:02:11.139 [bus notice] bus started with own address 31/36
2022-12-14 19:02:11.143 [bus notice] signal acquired
2022-12-14 19:02:11.153 [bus error] device status: eBUS comm error: framing
2022-12-14 19:02:11.666 [bus notice] device status: reset
2022-12-14 19:02:15.298 [bus notice] new master 10, master count 2
2022-12-14 19:02:15.366 [bus notice] new master 03, master count 3

==============================================

kratkale@CF19U2400:~$ ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyUSB0, enhanced
signal: acquired
symbol rate: 22
max symbol rate: 47
reconnects: 0
masters: 3
messages: 0
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd

=====================================================

kratkale@CF19U2400:~$ ebusctl raw
raw logging enabled

2022-12-14 19:05:16.123 [bus notice] <1008b511010189000954540080ffff0000ff2000
2022-12-14 19:05:18.208 [bus notice] <1008b51101028a0005033c7846827c00
2022-12-14 19:05:20.319 [bus notice] <1008b5100900000078ffff01ff00f80001019a00
2022-12-14 19:05:26.634 [bus notice] <1008b511010189000954540080ffff0000ff2000
2022-12-14 19:05:28.743 [bus notice] <1008b50401003d000a00062020ffffffff00800400
2022-12-14 19:05:30.814 [bus notice] <1008b5100900000078ffff01ff00f80001019a00
2022-12-14 19:05:35.009 [bus notice] <1008b511010189000954540080ffff0000ff2000
2022-12-14 19:05:41.337 [bus notice] <1008b5100900000078ffff01ff00f80001019a00
2022-12-14 19:05:45.527 [bus notice] <1008b511010189000954540080ffff0000ff2000
2022-12-14 19:05:47.601 [bus notice] <10feb516080025051914120322fd
2022-12-14 19:05:47.867 [bus notice] <1008b512020064ae00000000
2022-12-14 19:05:49.749 [bus notice] <1008b5100900000078ffff01ff00f80001019a00
2022-12-14 19:05:56.026 [bus notice] <1008b511010189000954540080ffff0000ff2000
2022-12-14 19:05:57.833 [bus notice] <0364b5120202fe98
2022-12-14 19:05:58.127 [bus notice] <1008b50401003d000a00372020ffffffff00806f00
2022-12-14 19:05:58.394 [bus notice] <1008b51101028a0005033c7846827c00
2022-12-14 19:06:00.234 [bus notice] <1008b5100900000078ffff01ff00f80001019a00

==========================================================

oot@CF19U2400:/home/kratkale# ebusd -f -c /tmp --logareas bus --loglevel info -d ens:/dev/ttyUSB0 
2022-12-14 19:07:08.670 [bus notice] device status: resetting
2022-12-14 19:07:08.672 [bus notice] bus started with own address 31/36
2022-12-14 19:07:08.705 [bus notice] signal acquired
2022-12-14 19:07:11.033 [bus notice] max. symbols per second: 133
2022-12-14 19:07:12.006 [bus notice] max. symbols per second: 173
2022-12-14 19:07:16.010 [bus notice] max. symbols per second: 179
2022-12-14 19:07:27.033 [bus notice] max. symbols per second: 181

=====================================================================

kratkale@CF19U2400:~$ ebusctl info
version: ebusd 22.4.v22.4
device: /dev/ttyUSB0, enhanced
signal: acquired
symbol rate: 132
max symbol rate: 197
reconnects: 0
masters: 1
messages: 0
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd

kratkale@CF19U2400:~$ ebusctl raw
raw logging enabled

2022-12-14 19:09:11.722 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:11.953 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:12.181 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:12.405 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:12.639 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:12.870 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:13.128 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:13.359 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...
2022-12-14 19:09:13.583 [bus notice] ...<00000000000000000000000000000000000000000000000000000000000000...

kratkale, 2022-12-21:
Wie kann ich da nun weitermachen? Mit ens kommen nun gleiche Ausgaben wie mit ESERA.

Silverstar, 2023-02-20:
> root@CF19U2400:/home/kratkale# ebusd -f -c /tmp --logareas bus --loglevel info -d enh:/dev/ttyUSB0 2022-12-14 19:02:11.138 

In /tmp liegen auch die csv für die am Bus angeschlossenen Geräte?

Was ergibt denn ein Start mit -s (scanconfig)?
Eventuell ohne -c, falls dort lokal gar keine csv liegen...