john30/ebusd-configuration #291: Calormatic 470 - SingleHwcLoading/PartyMode

nisi80, 2022-12-19:
Hallo,

ich habe eine Calormatic 470 und möchte den Heilßwasser-Modus per ebusd ansteuern. 
So mache ich es auch mit meinem Solarregler, der Automatic (sdr_p), der dann die Heißwasser-Anfrage an die Calormatic per Draht übergibt und den Speicher überwacht.

Leider kann ich den HwcOPMode im 470er-Kreis nur lesen und nicht schreiben.
Gibt es nicht die Möglichkeit, dieses Register zu schreiben?

Ich dachte ich hätte einmal ein externes Bedienteil gesehen, bei der diese Modi auch steuerbar waren.

Vielen Dank! 

VG Denis

Silverstar, 2023-02-20:
Moin,

ich weiß nicht genau, was du erreichen möchtest. Einmalige Speicherladung findest du beim 470 unter hc1sfmode = 8

https://github.com/john30/ebusd-configuration/blob/fdf1266cd32e238cda597273ccc4823a5ba3d992/ebusd-2.1.x/en/vaillant/15.470.csv#L20

Ansonsten kannst du versuchen, die csv lokal zu speichern und zu editieren, und machst in der 15.470.csv aus r,... ein r;w,.... in der Zeile zum HwcOPMode. Dann kannst du versuchen, diesen Punkt zu beschreiben. Ich weiß aber nicht, ob der 470 diesen Punkt beschreiben lässt und mit welchem Wert.

nisi80, 2023-02-22:
Ich mache nur einmal am Tag heißes Wasser und ansonsten lebe ich mit der Temperatur des Speichers. 
Es kommt jedoch vor, dass jemand mal heißes Wasser zum Duschen braucht :-(. 
Dann drückt er auf dem Tablet die Speicherladung und ich muss einmal dem Solarregler und andererseits der Calormatic die Speicherladung kommandieren.
Jedoch wollte dies nicht funktionieren, da es ja solch einen Wert in der Calormatic nicht gibt. 

Ich habe zwischenzeitlich einen Workaround gefunden: Indem ich die Heißwasserzeit kurz anpasse, bis das Flag der Speicherladung im Solarregler wieder weg ist (=Temperatur erreicht) und schreibe dann wieder die "normalen" Heißwasserzeiten rein. 

Eleganter wäre es jedoch, wenn der HwcOPMode schreibbar wäre, auch nach einem Update.

Silverstar, 2023-02-23:
Müsste dann nicht das Setzen des hc1sfmode = 8 genau das sein, was du brauchst?

nisi80, 2023-02-28:
Vielen Dank für den Hinweis, den Schreibschutz in der Definitionsdatei zu ermöglichen.
Das hat soweit geklappt und der Hc1SFMode lässt sich per ebusd setzen.

Leider gehen alle Werte von 0 bis 6 (0=none;1=party;2=onedaybankholiday;3=onedayholiday;4=holiday;5=bankholiday;6=quickveto)

7 und 8 werden leider nicht akzeptiert :-(
Somit muss ich bei meiner alten Strategie bleiben und die Zeiten ändern.

Aber das hat mir schon mal für den Partymodus geholfen, vielen Dank dafür!!!