john30/ebusd-configuration #294: Weird behavior of Vaillant VRC 700 outside temperature after connecting to ebusd

brozikcz, 2023-01-05:
Hi guys,

did some meet the similar issues with VRC 700? The setup with ebusd worked fine almost the ~1/2 year but now the VCR 700 unit behaves very weird.
After launch the `ebusd` the VCR 700 starts "losting" the outside temperature from the external temperature. The outside temperature disappear completely from the display of VCR 700 (top bar at the attached picture) and start reporting the `outsidetemp` as `-40`. The outside temperature cycling between `-40` and right value in ~4min intervals.

If the `ebusd` is not launched, the outside temperature works fine and doesn't disappear from the the display of unit. 

```
docker run -d --restart=unless-stopped --name=ebusd -v /var/opt/ebusd/ebusd-configuration/ebusd-2.1.x/en:/ebusd-configuration -it -p 8888 john30/ebusd:v22.4 --configlang=en --configpath=ebusd-configuration --scanconfig -d 192.168.100.223:3333 --mqttport=1883 --mqtthost=192.168.100.10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson --mqtttopic=ebus --latency=20 '--mqttvar=filter-name=psycho' --pollinterval=10
```

```
2023-01-05 05:27:51.770 [main notice] ebusd 22.4.v22.4 started with auto scan on device 192.168.100.223:3333
2023-01-05 05:27:51.818 [bus notice] bus started with own address 31/36
2023-01-05 05:27:51.822 [mqtt notice] connection established
2023-01-05 05:27:51.859 [bus notice] signal acquired
2023-01-05 05:28:00.118 [bus notice] new master 10, master count 2
2023-01-05 05:28:00.214 [bus notice] new master 03, master count 3
2023-01-05 05:28:00.214 [update notice] received unknown MS cmd: 1008b5110101 / 09342f200aff580100ff
2023-01-05 05:28:00.514 [update notice] received unknown MS cmd: 1008b5100900002dffffff040000 / 0101
2023-01-05 05:28:02.064 [bus notice] scan 08: ;Vaillant;BAI00;0405;9602
2023-01-05 05:28:02.064 [update notice] store 08 ident: done
2023-01-05 05:28:02.064 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0405;9602
2023-01-05 05:28:02.064 [bus notice] scan 08: ;Vaillant;BAI00;0405;9602
2023-01-05 05:28:02.072 [main notice] read common config file vaillant/general.csv
2023-01-05 05:28:02.072 [main notice] read common config file vaillant/scan.csv
2023-01-05 05:28:02.073 [main notice] read common config file vaillant/broadcast.csv
2023-01-05 05:28:02.075 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0405, HW9602
2023-01-05 05:28:02.294 [update notice] sent scan-read scan.08 id QQ=31: 
2023-01-05 05:28:02.554 [update notice] sent scan-read scan.08 id QQ=31: 
2023-01-05 05:28:02.811 [update notice] sent scan-read scan.08 id QQ=31: 
2023-01-05 05:28:03.087 [update notice] sent scan-read scan.08 id QQ=31: ;12;10;6100012780;5000;15092N;6<
2023-01-05 05:28:03.105 [main notice] found messages: 202 (2 conditional on 27 conditions, 0 poll, 10 update)
2023-01-05 05:28:03.391 [update notice] sent scan-read scan.08 id QQ=31: ;12;10;6100012780;5000;15092N;6<
2023-01-05 05:28:03.628 [update notice] sent scan-read scan.08 id QQ=31: ;12;10;6100012780;5000;15092N;6<
2023-01-05 05:28:04.477 [update notice] sent scan-read scan.08 id QQ=31: ;12;10;6100012780;5000;15092N;6<
2023-01-05 05:28:04.768 [update notice] sent scan-read scan.08 id QQ=31: ;12;10;6100012780;5000;15092N;6<
2023-01-05 05:28:04.768 [bus notice] scan 08: ;;12;10;6100012780;5000;15092N;6<
2023-01-05 05:28:07.011 [bus notice] scan 15: ;Vaillant;70000;0614;6903
2023-01-05 05:28:07.011 [update notice] store 15 ident: done
2023-01-05 05:28:07.011 [update notice] sent scan-read scan.15  QQ=31: Vaillant;70000;0614;6903
2023-01-05 05:28:07.011 [bus notice] scan 15: ;Vaillant;70000;0614;6903
2023-01-05 05:28:07.258 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231323231323030
2023-01-05 05:28:07.515 [update notice] sent scan-read scan.15 id QQ=31: 
2023-01-05 05:28:07.766 [update notice] sent scan-read scan.15 id QQ=31: 
2023-01-05 05:28:08.784 [main notice] read scan config file vaillant/15.700.csv for ID "70000", SW0614, HW6903
2023-01-05 05:28:08.786 [main notice] found messages: 641 (2 conditional on 27 conditions, 0 poll, 10 update)
2023-01-05 05:28:10.291 [update notice] received read bai Status01 QQ=10: 26.0;23.5;10.125;-;44.5;on
2023-01-05 05:28:10.582 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
2023-01-05 05:28:12.042 [update notice] received read bai Status02 QQ=10: auto;60;40.0;70;70.0
2023-01-05 05:28:20.369 [update notice] received read bai Status01 QQ=10: 26.5;23.5;10.125;-;44.5;on
2023-01-05 05:28:20.522 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
2023-01-05 05:28:30.372 [update notice] received read bai Status01 QQ=10: 26.5;23.5;10.125;-;44.5;on
2023-01-05 05:28:30.696 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
2023-01-05 05:28:40.411 [update notice] received read bai Status01 QQ=10: 26.0;23.5;10.125;-;44.5;on
2023-01-05 05:28:40.587 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
2023-01-05 05:28:42.892 [update notice] received read bai DateTime QQ=10: valid;06:28:43;05.01.2023;10.125
2023-01-05 05:28:43.698 [update notice] received read bai Status02 QQ=10: auto;60;40.0;70;70.0
2023-01-05 05:28:43.905 [update notice] received update-read broadcast vdatetime QQ=10: 06:28:40;05.01.2023
2023-01-05 05:28:44.828 [update notice] received unknown MS cmd: 1008b5110100 / 08a5011226040f0001
2023-01-05 05:28:45.033 [update notice] received update-read broadcast outsidetemp QQ=10: 10.125
2023-01-05 05:28:50.391 [update notice] received read bai Status01 QQ=10: 26.0;23.5;10.125;-;44.5;on
2023-01-05 05:28:50.652 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
2023-01-05 05:29:00.448 [update notice] received read bai Status01 QQ=10: 26.0;23.5;10.125;-;44.5;on
2023-01-05 05:29:00.700 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
2023-01-05 05:29:10.466 [update notice] received read bai Status01 QQ=10: 26.5;24.0;10.125;-;44.5;on
2023-01-05 05:29:10.814 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
2023-01-05 05:29:11.014 [update notice] received read bai Status02 QQ=10: auto;60;40.0;70;70.0
2023-01-05 05:29:20.630 [update notice] received read bai Status01 QQ=10: 26.5;24.0;10.125;-;44.5;on
2023-01-05 05:29:20.946 [update notice] received update-write bai SetMode QQ=10: auto;22.5;-;-;0;0;1;0;0;0
[... 6 lines cut ...]
2023-01-05 05:29:48.079 [update notice] received unknown MS cmd: 1008b5110100 / 08a7011226040f0001
2023-01-05 05:29:48.827 [update notice] received update-read broadcast outsidetemp QQ=10: 10.125
2023-01-05 05:29:49.100 [update notice] received unknown MS cmd: 1008b5100305ff01 / 0101
2023-01-05 05:29:49.310 [update notice] received update-write bai StatusCirPump QQ=10: off
2023-01-05 05:29:49.541 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2023-01-05 05:29:49.783 [update notice] received unknown MS cmd: 1008b513020508 / 00
2023-01-05 05:29:50.039 [update notice] received unknown BC cmd: 10feb510020601
2023-01-05 05:29:51.152 [update notice] received read bai Status01 QQ=10: 26.0;23.5;10.125;-;44.5;on
2023-01-05 05:29:52.110 [update notice] received update-write bai SetMode QQ=10: auto;33.0;-;-;0;0;1;0;0;0
2023-01-05 05:30:42.216 [update notice] received read bai DateTime QQ=10: valid;06:30:42;05.01.2023;10.125
2023-01-05 05:30:42.938 [update notice] received read bai Status02 QQ=10: auto;60;40.0;70;70.0
2023-01-05 05:30:44.014 [update notice] received update-read broadcast vdatetime QQ=10: 06:30:40;05.01.2023
2023-01-05 05:30:46.258 [update notice] received unknown MS cmd: 1008b5110100 / 08e7011257040f0001
2023-01-05 05:30:46.980 [update notice] received update-read broadcast outsidetemp QQ=10: -40.000
....
2023-01-05 05:36:42.451 [update notice] received read bai Status01 QQ=10: 23.0;24.0;10.125;-;44.5;off
2023-01-05 05:36:42.676 [update notice] received update-write bai SetMode QQ=10: auto;22.0;-;-;0;0;1;0;0;0
2023-01-05 05:36:45.009 [update notice] received read bai DateTime QQ=10: valid;06:36:46;05.01.2023;10.125
2023-01-05 05:36:45.766 [update notice] received read bai Status02 QQ=10: auto;60;40.0;70;70.0
2023-01-05 05:36:46.177 [update notice] received update-read broadcast vdatetime QQ=10: 06:36:43;05.01.2023
2023-01-05 05:36:48.141 [update notice] received update-read broadcast outsidetemp QQ=10: 10.125
```

[image]
[image]

RotorMotor42, 2023-01-05:
I have the same problem with a VRC720/2.
But only every few days it receives -40 as broadcast outside temp.

RotorMotor42, 2023-01-17:
Can you provide information where to find that connection?

brozikcz, 2023-01-17:
Check your manual, something like that:

[image]

[image]

rossi67, 2023-02-02:
> I have the same problem with a VRC720/2.
> But only every few days it receives -40 as broadcast outside temp.
I had the same issue. It was caused by too many get requests from my own home automation. I solved it by syncronising my get request with the broadcast message ebus/broadcast/vdatetime on MQTT. The outsidetemp broadcast comes only but not always with the vdatetime broadcast. The vdatetime broadcast comes every minute. So you can send all your get requests right after recieving vdatetime, so you be sure not to create a conflict with outsidetemp broadcast.