john30/ebusd-configuration #296: Wie CSV Konfigurieren? Field 1 zum Abschneiden der Daten verwenden?

karusell, 2023-01-05:
Hallo zusammen,

ich möchte einen neuen parameter in der CSV erstellen und habe dazu auch die [Anleitung](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files) gelesen.

Das sind die 2 Nachrichten: Hex 32 ändert sich auf Hex 28.

7108b51a04052e3428 / 0a2e021b**32**001e003c0001
...
7108b51a0405473428 / 0a47021b**28**001e003c0001
...
7108b51a0405db3428 / 0adb021b**3c**001e003c0001

`"# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment # Change New Parameter,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,08,B51A,,,,IGN:2,,,,,,,,,,,,,,,,,,,,, r,,ParameterName,,,,,053b3428,3b021b,,D1B,,%,,,,,,,,,,,,,,,,,,, !include,hcmode.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,, !include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,"`

ZZ = 08
PBSB = B51A
NN= 04
DD = 052e3428
Datentyp D1B

Field 1 habe ich wie in der Beschreibung / Anleitung benutzt um hinten in der Antwort noch was abschneiden muss weil mein "Wert" steht ja mittel in den Daten drin. "32" im Startzustand. Und 28 nachdem ich es an der Steuerung verändert habe.

Ergebnis ist jetzt:
ebusctl read -f ParameterName 27

--> Als Dezimalwert sollte hier jetzt 

Das Ergebnis passt so nicht, ich muss beim Abschneiden/Filtern noch einen Fehler drin haben.

Hat jemand von euch eine Idee? 
Ich stehe gerade etwas auf dem Schlauch was ich noch machen kann.

Vor allem habe ich Field1 zum Abschneiden der Datennachricht von hinten richtig verwendet?
Kann jemand genauer erklären wie man das richtig anwendet?

Gruß karusell

karusell, 2023-01-06:
UPDATE: 

**Folgendes Ergebnis bekomme ich jetzt:**
ebusctl read -f FluessterbetriebKompr 6.78869e-21;15360

6.78869e-21 --> Wie kann ich dieses Ergebnis ignorieren?
15360 --> ist sehr knapp dran! 3C00 (HEX) umgerechnet ist  15360 (DEC)

**FRAGE:** Wie kann ich von dem HEX Ausdruck noch die 00 wegschneiden/Ignorieren?
Mit Datentyp D1B hatte ich es auch probiert.

**Die CSV Datei sieht so aus:**

`"# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment # Change Parameter,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,31,08,B51A,,,,IGN:2,,,,,,tempv,,,,,,,,,,,,,,, r,,Parametername,,,,,053b3428,3b021b,,UIN,,%,,,,,,,,,,,,,,,,,,, !include,hcmode.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
!include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,"`

Wenn ich die Daten mit `ebusctl grab result decode` analysiere, sieht es so aus als ob D1B genau der richtige Datentyp wäre. Ich habe es in der CSV probiert kommt aber dann decode error. Also war bisher besser über UIN zu gehen.

7108b51a0405b73428 / 0ab7021b3c001e003c0001 = 1
`"D1B   b7=-73, 02=2, 1b=27, 3c=60, 00=0, 1e=30, 00=0, 3c=60, 00=0, 01=1"`
`"UIN   b702=695, 021b=6914, 1b3c=15387, 3c00=60, 001e=7680, 1e00=30, 003c=15360, 3c00=60, 0001=256"`

Unter `ebusctl grab result all` sieht man auch dass er diese Nachricht auch dem Parameter schon richtig zuordnet...
3108b51a04053b3428 / 0a3b021b3c001e003c0001 = 1: vwzio Parametername

Hat jemand Hinweise wie ich damit weiterkommen könnte?

Danke euch.

karusell, 2023-01-06:
[](url)

karusell, 2023-01-06:
Hallo zusammen,

also ich habe es nun geschafft den Wert auszulesen (read). Es handelt es sich dabei um den "Fluessterbetrieb Kompr".
Leider schaffe ich es noch nicht diesen zu schreiben und bräuchte eure Unterstützung - komme da gerade nicht weiter.

Seht ihr in der CSV einen Fehler? Wie deklariert man den Write richtig? 

`"# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment # Change Noise,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,71,08,B51A,,,,IGN:3,,,,,,,,,,,,,,,,,,,,, *w,,,,71,08,B51A,,,,,,,,,,,,,,,,,,,,,,,,, r;w,,FluessterbetriebKompr,,,,,053b3428,,,D1B,,%,,,,,,,,,,,,,,,,,,, !include,hcmode.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,, !include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,"`

ebusctl read -f -c vwzio FluessterbetriebKompr 
30

ebusctl write -c vwzio FluessterbetriebKompr 60 
done

ebusctl read -f -c vwzio FluessterbetriebKompr 
30

Er nimmt den Schreibvorgang. In der LOG steht das:
2023-01-06 18:27:26.831 [main notice] write vwzio FluessterbetriebKompr: decode done

ABER: der Read danach zeigt, dass der Wert in der VWZIO nicht angekommen ist. Ist vllt. etwas bei QQ, ZZ noch zu ändern? bzw. eine andere CSV noch geschrieben werden aus der Perspektive eines anderen Senders?

Gruß karusell

closed 2023-01-07 by karusell

cyberthom42, 2023-02-02:
Bist du hier weiter gekommen? Ich versuche auch gerade das Konfigurations Menü zu dekodieren (B51A). Ich kann die Parameter identifizieren. Ich kann bei dir nicht ganz nachvollziehen, wie du von `7108b51a0405xx3428 / 0axx021byy001e003c0001` auf `*r,,,,31,08,B51A,,,,IGN:2,,,,,,tempv,,,,,,,,,,,,,,, r,,Parametername,,,,,053b3428,3b021b,,UIN,,%,,,,,,,,,,,,,,,,,,, ` gekommen bist.

Warum hast du 053b3428 genommen? Bei dir werden die Zahlen an der Position 05**xx**3428 doch auch mehr oder weniger hoch gezählt. Ich dachte, der relevante Teil wäre nur 3428... Und warum hast du 3b021b genommen?

Meine Werte sehen auch so aus, für Restförderhöhe z.B. 7108b51a0405xx341f / 0a99022d**3002**c80084030a (Wert=560)