john30/ebusd-configuration #298: Unknown MS cmd

Koky05, 2023-01-11:
I have Vaillant ecoCompact VSC with VR61 mixing module and F470 CalorMatic controler. I use eBus 3.0 and recaiving some unknown messages:

> [update notice] received unknown MS cmd: 1050b505053c00808101 / 00
> [update notice] received unknown MS cmd: 1050b5040126 / 0702020030018101
> [update notice] received unknown MS cmd: 1050b5040118 / 061d1d00bb0118
> [update notice] received unknown MS cmd: 1050b5040126 / 0702020030018201
> [update notice] received unknown MS cmd: 1050b505053c00808201 / 00
> [update notice] received unknown MS cmd: 1050b5040118 / 061d1d00be0118
> [update notice] received unknown MS cmd: 1050b505022900 / 00
> [update notice] received unknown MS cmd: 1050b50903291f00 / 031f0001
> [update notice] received unknown MS cmd: 1050b5040118 / 061d1d00bd0118
> [update notice] received unknown MS cmd: 1050b503020003 / 0affffffffffffffffffff
> [update notice] received unknown MS cmd: 1050b509040e010000 / 00
> [update notice] received unknown MS cmd: 1050b509040e200000 / 00
> [update notice] received unknown MS cmd: 1050b505021800 / 00
> [update notice] received unknown MS cmd: 1050b505023001 / 0101

Is it way how to translate these messages?

> version: ebusd 23.1.23.1
> update check: OK, device firmware 1[8419] available, broadcast.csv: different version available, memory.csv: different version available, vaillant/15.f47.csv: different version available, vaillant/50.v61.mc.csv: different version available, vaillant/bai.0010015600.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available, vaillant/iotestbmc.inc: different version available, vaillant/scan.csv: different version available, vaillant/tempsetpoints.inc: different version available, vaillant/timercc.inc: different version available, vaillant/timerhc.inc: different version available, vaillant/timerhwc.inc: different version available, vaillant/yield3d43.inc: different version avail
> device: 192.168.2.31:9999, enhanced, firmware 1.1[63aa]
> access: *
> signal: acquired
> symbol rate: 39
> max symbol rate: 294
> min arbitration micros: 1
> max arbitration micros: 150
> min symbol latency: 9
> max symbol latency: 132
> reconnects: 14
> masters: 5
> messages: 987
> conditional: 6
> poll: 681
> update: 10
> address 03: master #11
> address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0116;HW=9602", loaded "vaillant/bai.0010015600.inc" ([HW=9602]), "vaillant/08.bai.csv"
> address 10: master #2
> address 11: master #7
> address 15: slave #2, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102", loaded "vaillant/15.f47.csv"
> address 16: slave #7, scanned "MF=Vaillant;ID=B7000;SW=0120;HW=6202"
> address 26: slave, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102", loaded "vaillant/26.f47.2.csv"
> address 31: master #8, ebusd
> address 36: slave #8, ebusd, scanning
> address 50: slave, scanned "MF=Vaillant;ID=V6100;SW=0306;HW=1902", loaded "vaillant/50.v61.mc.csv"
> address f1: master #10
> address f6: slave #10, scanned "MF=Vaillant;ID=F4700;SW=0124;HW=6102", loaded "vaillant/f6.f47.3.csv"

26.f47.2.csv and f6.f47.3.csv are just copy of 15.f47.csv

Silverstar, 2023-01-31:
`vaillant/bai.0010015600.inc: different version available`
You are using a local copy of configurations instead of ebusd loading them from the web? Maybe the newer(?) files have these messages decoded. (Or do you have the latest versions and edited the files locally? But due to the amount of different files reported, I guess you have old versions local.)
Maybe not. I have a VRC470 on a VR39 adapter and have some unknown messages as well, so some are still left to be discovered.

Koky05, 2023-01-31:
Yes, I use local config files for different priority and for:

Silverstar, 2023-02-01:
What I was trying to say is, you are working with local copies; do you have the latest versions? Because `vaillant/15.f47.csv: different version available, vaillant/50.v61.mc.csv: different version available,` is in your log.
(These are the two units talking in your unknown messages)
So, if you did not edit the files, you obviously don't have the latest version, so try that.

Koky05, 2023-02-02:
I understand your point. But I monitor changes in configuration files and update my local copy. I made changes in priority "r[1-9]" for lines that I monitor often.

Silverstar, 2023-02-02:
Alright, sorry. So you could have a look at the common files like h(w)cmode were b504, b505 messages are defined. Maybe you need one of these included in your config.

Koky05, 2023-02-03:
In 08.bai is included hcmode.
Can you send me decoded names for that command for look where it is located, and I will try to include it?