john30/ebusd-configuration #301: Actual Humidity

pepsonEL, 2023-01-15:
Hi. I have controller VRC700 in my Vaillant. It show me in configuration menu > installer show me Actual Humidity in Room. 

How i can read this value by EBUS in Home Assistant ? I dont see this value in entity  from auto discovery.

chrizzzp, 2023-10-13:
Found it: 

Following definitions were added to the CSV of CTLV2 (defaults were not added but indicate the sections of the ctlv2.csv were the definition wass added):
```

*r,,,,,,B524,02000300,,,IGN:4,,,
r,,z1RoomHumidity,,,,,2800,,,EXP,,%,room humidity zone1

*r,,,,,,B524,02000301,,,IGN:4,,,
r,,z2RoomHumidity,,,,,2800,,,EXP,,%,room humidity zone2

*r,,,,,,B524,02000302,,,IGN:4,,,
r,,z3RoomHumidity,,,,,2800,,,EXP,,%,room humidity zone3
```

Wireheadbe, 2023-10-15:
> Found it:
>
> Following definitions were added to the CSV of CTLV2 (defaults were not added but indicate the sections of the ctlv2.csv were the definition wass added):
>
> ``` [...]

This seemingly doesn't work on the 700:
```
2023-10-15 15:52:19.964 [update error] unable to parse read 700 z1RoomHumidity from ff15b52406020003002800 / 00: ERR: invalid position
2023-10-15 15:52:19.964 [mqtt error] decode 700 z1RoomHumidity: ERR: invalid position
```
.. after adding
```
,,z1RoomHumidity,,,,,2800,,,EXP,,%,room humidity zone1
```
to 15.700.csv

stadid, 2024-04-25:
@Wireheadbe, @pepsonEL,
Have you moved forward with reading humidity for VRC700 using ebusd?

If someone of you **has internet gateway and native MyVaillant app** there are chances to get the format of the reading command on the bus, as some users of VRC700 report that they observed humidity readings in the MyVaillant app for the VRC700.

BTW: I've also tried the above command from 720 found by @chrizzzp, but without success.

arnaudpn, 2026-03-28:
Thanks @chrizzzp for the quick feedback.
I found the right definition (060009010700) here: https://github.com/john30/ebusd/discussions/1482#discussioncomment-16357483

```
root@b4d7ad18-ebusd:/# ebusctl define -r "r,700,RoomHumidity,room humidity,,15,b524,060009010700,ign,,IGN:4,,,,humidity,,EXP,,%,room humidity"
done

root@b4d7ad18-ebusd:/# ebusctl read -f RoomHumidity
49

root@b4d7ad18-ebusd:/# ebusctl read -f RoomHumidity
50
```