john30/ebusd-configuration #304: Hilfe benötigt (VWF 117/4 + VRC 720 + VR 921

Geg0r, 2023-01-26:
Hi all,

I just received my rev3.1 ebusd adapter and now I am trying since hours to connect my Vaillant VWF 117/4 to it.

`root@ebusd:/etc/ebusd/config# ebusctl info
version: ebusd 23.1.23.1
device: 192.168.1.107:9999, enhanced, firmware 1.1[8419].2[847f]
signal: acquired
symbol rate: 24
max symbol rate: 115
min arbitration micros: 1
max arbitration micros: 2
min symbol latency: 7
max symbol latency: 10
reconnects: 0
masters: 5
messages: 509
conditional: 0
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0308;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=72000;SW=0122;HW=7703", loaded "vaillant/15.720.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0308;HW=0403"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX2;SW=4027;HW=5703"

`

From my point of view this looks good but the results are really low... hardly any results are found...

`root@ebusd:/etc/ebusd/config# ebusctl find
720 AdaptHeatCurve = no data stored
720 BankHolidayEndPeriod = no data stored
720 BankHolidayStartPeriod = no data stored
720 ccTimer.Friday = no data stored
720 ccTimer.Monday = no data stored
720 ccTimer.Saturday = no data stored
720 ccTimer.Sunday = no data stored
720 ccTimer.Thursday = no data stored
720 ccTimer.Tuesday = no data stored
720 ccTimer.Wednesday = no data stored
720 ContinuosHeating = no data stored
720 currenterror = no data stored
720 CylinderChargeHyst = no data stored
720 CylinderChargeOffset = no data stored
720 Date = no data stored
720 DisplayedOutsideTemp = 0.125
720 errorhistory = no data stored
720 FrostOverRideTime = no data stored
720 GlobalSystemOff = no data stored
720 Hc1ActualFlowTempDesired = no data stored
720 Hc1AutoOffMode = no data stored
720 Hc1CircuitType = no data stored
720 Hc1ExcessTemp = no data stored
720 Hc1FlowTemp = 24.5
720 Hc1HeatCurve = no data stored
720 Hc1HeatCurveAdaption = no data stored
720 Hc1MaxFlowTempDesired = no data stored
720 Hc1MinFlowTempDesired = no data stored
720 Hc1MixerMovement = no data stored
720 Hc1PumpStatus = 0
720 Hc1RoomTempSwitchOn = no data stored
720 Hc1Status = no data stored
720 Hc1SummerTempLimit = no data stored
720 Hc2ActualFlowTempDesired = no data stored
[... 250 lines cut ...]
hmu YieldThisYear7 = no data stored
hmu YieldThisYear8 = no data stored
hmu YieldThisYear9 = no data stored
hmu YieldTotal = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.08  = Vaillant;HMU00;0308;0403
scan.08 id = 21;21;04;0010016422;0001;005082;N0
scan.15  = Vaillant;72000;0122;7703
scan.15 id = 21;21;03;0020260914;0953;006556;N3
scan.76  = Vaillant;VWZ00;0308;0403
scan.76 id =  (ERR: invalid position for 3176b5090124 / 00)
scan.f6  = Vaillant;NETX2;4027;5703
scan.f6 id = 21;21;06;0020260962;0938;009604;N2
`

I am mainly interessted in the hmu results like "hmu CurrentConsumedPower" but I am not able to receive this info.

What am I doing wrong?

Hopefully somebody can help.

Thanks and best regards
Geg0r

cyberthom42, 2023-02-02:
you need to actively read the parameters. 
Did you try:  `ebusctl read CurrentConsumedPower` ?

Geg0r, 2023-02-02:
Yes you are right this needs to be done and I change the 08.hmu.csv file replacing r with r1 to read the parameters every cycle. Data is now written to an influxes via mqtt but there are still a lot parameters missing

cyberthom42, 2023-02-03:
there are improved csv files around. I actually put my own locally as some of the pull requests are still not incorporated. Yesterday I also figured out how to decode the parameter from the configuration menu "Fachhandwerkerebene" of the arotherm. But I gess it should be similar to yours. You can have a look at my github repository, I will also upload my files there

Geg0r, 2023-02-03:
Yeah I also realized that there are different file flying around and I also created a local copy and putted all working parameters in my local file. But there are still a lot parameters missing. 
I still have 50 parameters which are shown in `ebusctl grab result` as unknown but I am struggeling with decoding them.
I will check your repo

cyberthom42, 2023-02-03:
Yesterday I found out how to decode and change the settings at the "Fachhandwerkerebene", maybe this helps so far. I will upload this soon:
`# B5 1A Fachhandwerker Konfiguration,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,"B51A","05FF34",,,IGN:3,,,,,,,,,,,,,,,,,,,,,
*w,,,,,,"B51A","06FF34",,,,,,,,,,,,,,,,,,,,,,,,
r;w,,Restfoerderhoehe,,,,,1f,,,UIN,,,,,,,,,,,,,,,,,,,,,
r;w,,hwcBetriebsart,,,,,44,,,UIN,,,Betriebsart Warmwasser 0=eco 1=normal 2=balance,,,,,,,,,,,,,,,,,,
`

cyberthom42, 2023-02-03:
How did you figure out that r1 means that to read the parameters every cycle. What is a cycle exactly (each read request?). I could not found a documentation about this.

Geg0r, 2023-02-03:
The r1 thing was mentioned by @Thoro here: 
[image]

Thoro, 2023-02-03:
@cyberthom42 Documentation is here: https://github.com/john30/ebusd/wiki/4.1.-Message-definition

highshot1, 2023-02-09:
Got the same configuration like you Geg0r. (vwf 117/4, VRC 720 + VR921)
thanks for your input. changed r to r1 and now i am getting few data... searching especially for brine temperatures.... still a long way to go to get there..