john30/ebusd-configuration #305: Heizstab aktivieren

cyberthom42, 2023-01-27:
Gibt es eine Möglichkeit den Heizstab über ebus zu aktivieren (für PV-Überschuss)? Theoretisch könnte man ja den Bivalenzpunkt hochsetzen, aber dafür habe ich keinen   ebusd-Eintrag gefunden. Evtl. kann man auch mit trival arbeiten, dazu habe ich aber erst recht nichts gefunden. Aber evtl. gibt es ja eine einfachere Möglichkeit?

edsub, 2023-02-04:
Looking for this feature as well! (but first have to get my setup running :-( )

hmax-72, 2023-02-22:
Would interested as well.
but for different purpose: turn on hot water preparation if energy is cheapest during the day (my tariff provides hourly stock prices).

M0nk3yOo, 2023-03-04:
Vielleicht hilft das schonmal weiter:

Ich habe dies per Heizungsumstellung und Logprüfung und sonstige Suche herausgefunden:

ebusctl write -c ehp BackupModeHwc withbackup
ebusctl write  -c hwc StorageTempDesiredStart 51
ebusctl write  -c hwc StorageTempDesiredStop 75
ebusctl write  -c hwc load on

Eraser-1, 2023-03-05:
Ich nutze die Boilerladung bei mir erfolgreich seit Jahren.
Dazu sende ich einen MQTT-Befehl mit dem Wert 6 auf ebusd/700/SFMode/set.

swi88, 2024-03-23:
Hi, mich würde auch interessieren wie man den Heizstab manuell an und ausschalten kann. Gibt es diesbezüglich etwas neues? 
Bisher habe ich nur gefunden, wie man die Leistung des Heizstabs begrenzen kann (FF331A)

swi88, 2024-03-25:
> Hi, hier der Befehl, den ich oben erwähnte
> 
> ebusctl write -c ehp BackupModeHwc withbackup

Danke. 
Ich besitze eine AroTherm Plus 55/6 (h08.hmu). Leider habe ich dafür noch keine entsprechenden Settings gefunden.

M0nk3yOo, 2024-03-25:
Ah okay, wir habe eine vailant, aber das genaue Modell weiß ich gerade nicht. 
vermutlich sind es dann wohl andere Parameter bei AroTherm(?)

Vielleicht weiß jemand anderes ja weiter