john30/ebusd-configuration #306: MF=Vaillant fails to parse answer: scan config 76: ERR: invalid position

grafenam, 2023-01-27:
This topic was raised in other discussions, but resolutions where not working for me.

I just received my ebusd adapter 3 and now I am trying since hours to connect my Vaillant VWF 88/4 + VRC700/5 and VR920 to it.

```
 $ ebusctl info
version: ebusd 23.1.23.1-3-gaf614ae1
device: /dev/ttyUSB0, enhanced, firmware 1.1[e85a].2[73c8]
signal: acquired
symbol rate: 79
max symbol rate: 169
min arbitration micros: 604
max arbitration micros: 2704
min symbol latency: 6
max symbol latency: 12
reconnects: 0
masters: 5
messages: 629
conditional: 49
poll: 0
update: 10
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VR920;SW=2706;HW=5703"
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0131;HW=5202", loaded "vaillant/e0.omu.csv"

```

```
2023-01-27 11:15:57.290 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2023-01-27 11:15:57.429 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2023-01-27 11:15:57.529 [update notice] received unknown MS cmd: 1076b5110101 / 09ff371002ff5c0000ff
2023-01-27 11:15:57.617 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2023-01-27 11:15:57.617 [main error] scan config 76: ERR: invalid position

```

Is it possible to get the scan working properly?

grafenam, 2023-02-04:
I've found the solution here https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files

mwildbolz, 2023-02-04:
@grafenam Did you find a configuration for the device with id 76 (VWZ00)?
Would be fine, if you could share it...

Thanks!

grafenam, 2023-02-05:
no, i started to debug. Do you have the same issue?

mwildbolz, 2023-02-05:
> no, i started to debug. Do you have the same issue?

Same issue on my side, also with exact same HW and SW version...

grafenam, 2023-02-11:
I'm still not able to find the right configuration. Any idea if a newer SW version could be provided by Vaillant?

Sethach, 2023-02-12:
I dont think that a never software from vaillant could solve this.
We have to create a fitting csv for this device.
By the way I am in a similar situation 
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0308;HW=0403"

grafenam, 2023-04-15:
It seems other users had this issue too:
https://github.com/john30/ebusd/issues/379#issuecomment-757459644
Anyone able to share the 76.vwz.csv file?

szflo, 2024-06-09:
I have same issue: 
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0308;HW=0403"

kjumybit, 2024-09-25:
Me too
```address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0308;HW=0403"```

grafenam, 2025-01-10:
to solve the issue and to avoid the error messages in the logs I added a current 76.vwz.csv [](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/76.vwz.csv) to my configuration folder and I deleted the existing 76.vwzio.csv 

```
# ./ebusctl info
version: ebusd 24.1.p20241110
update check: revision 24.1 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.hmu.csv: different version available, vaillant/15.700.csv: newer version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available
device: /dev/ttyACM0, serial
access: *
signal: acquired
symbol rate: 46
max symbol rate: 195
min arbitration micros: 1
max arbitration micros: 302
min symbol latency: 0
max symbol latency: 11
scan: finished
reconnects: 0
masters: 5
messages: 691
conditional: 49
poll: 215
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403", loaded "vaillant/76.vwz.csv"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0131;HW=5202"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX1;SW=4035;HW=5703"
```

hackentrick, 2025-01-12:
Hi @grafenam, thanks for solving this! It looks like I dont have the correct configuration.
My ebusd service is not loading the config for `address 76: slave #9`
Could you tell me what is wrong? Sorry, I'm new to ebus, and not sure where to look.
```

ebusctl info
version: ebusd 24.1.24.1
device: /dev/ttyACM0, serial high speed, enhanced, firmware 1.1[4414].1[4414]
signal: acquired
symbol rate: 23
max symbol rate: 175
min arbitration micros: 2
max arbitration micros: 9
min symbol latency: 4
max symbol latency: 6
scan: finished
reconnects: 0
masters: 5
messages: 544
conditional: 0
poll: 0
update: 16
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0308;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=72000;SW=0122;HW=7703", loaded "vaillant/15.720.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0308;HW=0403"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX2;SW=4035;HW=5703"

```

john30, 2025-01-14:
a 76.vwz is now part of the [next](https://ebus.github.io/next/) folder as of #330

closed 2025-01-14 by john30