john30/ebusd-configuration #307: VWL 75/5, VWL 78/5, VRC 720, VR 921, config not recognized

Nyrobee, 2023-01-28:
Trying to start ebusd on ebus adapter from Daniel Kucera. Adapter seems to connect but the bus messages give an indication of incomplete or unknown config files. I've tried the PR "support ctlv2" from janphkre but without result. After a while my system asks for maintenance. See attached log files for details. If somebody can point me in the right direction I would appreciate it.
[Ebusd_Vaillant 75_5.txt](https://github.com/john30/ebusd-configuration/files/10527981/Ebusd_Vaillant.75_5.txt)

Geg0r, 2023-02-02:
Can you show your /etc/default/Ebusd file?

Nyrobee, 2023-02-04:
Thanks for reply. The problem with system going in maintenance mode is solved by powering the ebusd adapter (it was relying on power from the ebus but in my case that wasn't enough clearly).
What do you mean by the ebusd file, I've Ebusd installed with Home assistant. So I'm not sure where I can find this ebusd file.
My log gives unable to load scan configs for HMU, VWZ, CTLV2 and NEXT2, I do see some values but most of them show as unknown.

SzosszeNET, 2023-02-23:
@Nyrobee I have the same issue - so you are saying the unknown messages that can be seen in your attached log went away by supplying external power to the adapter???

Nyrobee, 2023-02-26:
No, I can read the VR 720 with the pull request for CTLV2. The HMU I still can read only partially, VWZ and NEXT2 don't have CSV files on GitHub so unknown.
I made a mistake by putting config repository in wrong directory. 
The external power was needed because my bus went in maintenance mode after connecting the ebus adapter.