john30/ebusd-configuration #310: Local config clone gives errors

edsub, 2023-02-04:
I cloned the contents of the 'latest' folder of http://cfg.ebusd.eu/ to  /var/ebusd/config

When I run a --checkconfig I get
`ebusd -f --scanconfig=full -d /dev/ttyUSB-EBUS -c /var/ebusd/config --checkconfig
2023-02-04 22:54:04.982 [main notice] ebusd 23.1.23.1-3-gaf614ae1 performing configuration check...
en/vaillant/_templates.csv:46: ERR: element not found, field type TEMP in field 0
2023-02-04 22:54:04.991 [main error] error reading templates in en/vaillant: ERR: element not found, last error: en/vaillant/_templates.csv:46: ERR: element not found, field type TEMP in field 0
en/vaillant/15.b7v.csv:25: ERR: element not found, field type ENERGY4 in field 1
2023-02-04 22:54:04.997 [main error] error reading config files from /var/ebusd/config/: ERR: element not found, last error: en/vaillant/15.b7v.csv:25: ERR: element not found, field type ENERGY4 in field 1
2023-02-04 22:54:04.997 [main notice] found messages: 44 (0 conditional on 0 conditions, 0 poll, 4 update)
`

When I run the same but using the web resources, all looks ok. 
` ebusd -f --scanconfig=full -d /dev/ttyUSB-EBUS -c http://cfg.ebusd.eu/ --checkconfig
2023-02-04 22:57:38.975 [main notice] ebusd 23.1.23.1-3-gaf614ae1 performing configuration check...
2023-02-04 22:57:39.326 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)`

What am I doing wrong?

Geg0r, 2023-02-06:
I would recommend not to download everything from https://cfg.ebusd.eu
Just do a git clone of the repository and delete everything besides the ebusd-2.1.x folder.
Or just delete the libebus-0.1.x folder because its not needed and there seems to be a broken file in there (file mc2.csv)

edsub, 2023-02-06:
Thanks for the tip. This works!

closed 2023-02-06 by edsub