john30/ebusd-configuration #311: Devices not recognized (remain in 'scanning' in ebusctl info)

edsub, 2023-02-05:
Not sure if this an ebusd or an ebusd-configuration issue.

I have build ebusd from source on Ubuntu 22.04.01
Things seem to be running ok, have managed to get the potentionmeter on my Esera eBus to USB coupler and have nice "<aa"  values when running 
ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB-EBUS

When starting ebusd in order to scan  my devices, for some reason nothing (except for the ebus coupler) is recognized.

My lograwdata is attached

When I run ebusctl info, the result after few minutes is (and this remains!)
[ebusd_lograwdata.log](https://github.com/john30/ebusd-configuration/files/10611466/ebusd_lograwdata.log)
:
```
version: ebusd 23.1.23.1-8-g9c0af7b9
device: /dev/ttyUSB-EBUS
signal: acquired
symbol rate: 24
max symbol rate: 144
min arbitration micros: 114
max arbitration micros: 1477
min symbol latency: 5
max symbol latency: 7
reconnects: 0
masters: 4
messages: 15
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanning
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 52: slave, scanning
address 71: master #9
address 76: slave #9, scanning
```

edsub, 2023-02-06:
To add to the above: Can it be that my (Esera USB ebus) coupler is not capable of writing to the ebus?
I followed https://github.com/john30/ebusd/discussions/735#discussioncomment-4871690
And without "scanconfig" and just the csv's of the devices I (think I) need I now can reed some data.
I am a bit puzzled about the fact that my (new) Esera USB cannot read (dropped Esera a question) as they currently advertise the couplers ARE able to write....

Sethach, 2023-02-15:
Do you have any news on this? I have a similar problem, that the scanning does not end for some of my devices. But some are ok.
The strange thing with my older esera is, after a power off of the heating system, not the esera and ebusd, I got a lot more configuration from the bus

edsub, 2023-02-24:
I have read about the behaviour when restarting the heating system. But I prefer not to do that...
So waiting for ebusd adapter 3

edsub, 2023-09-21:
Currently have my ebusd v5 adapter, but no joy of progress....
On the contrary: I now also have a 
`[main error] scan config 15: ERR: read timeout `
once (in this case) my VR720 is scanned . . . .

Puzzled

john30, 2023-09-23:
isn't this related to https://github.com/john30/ebusd/discussions/1018#discussioncomment-7089152 ?
why do you duplicate so much...?