john30/ebusd-configuration #313: vSMART configuration

gvisol, 2023-02-12:
I have created a new file for vSMART configuration.
The aim is to requst a pull when it were confirmed
BTW, hcmode.inc configuration lines B510 00 are inconsistent with vSMART messages: 
# type (r[1-9];w;u)	circuit	name	[comment]	[QQ]	ZZ	PBSB	[ID]	field1	part (m/s)	datatypes/templates	divider/values	unit	comment																																																																			
*uw						B510																																																																										
uw		SetMode	Operation Mode				00			hcmode				flowtempdesired		temp1				hwctempdesired		temp1				hwcflowtempdesired		temp0						IGN:1				disablehc		BI0				disablehwctapping		BI1				disablehwcload		BI2						IGN:1				remoteControlHcPump		BI0				releaseBackup		BI1				releaseCooling		BI2				

File: 15.scc.csv
# type (r[1-9];w;u)	circuit	name	[comment]	[QQ]	ZZ	PBSB	[ID]	field1	part (m/s)	datatypes/templates	divider/values	unit	comment	field2	part (m/s)	datatypes/templates	divider/values	unit	comment	field3	part (m/s)	datatypes/templates	divider/values	unit	comment	field4	part (m/s)	datatypes/templates	divider/values	unit	comment	field5	part (m/s)	datatypes/templates	divider/values	unit	comment	field6	part (m/s)	datatypes/templates	divider/values	unit	comment	field7	part (m/s)	datatypes/templates	divider/values	unit	comment
#	SCC	vSMART																																															
uw	SCC	WriteModeSCC	Set Mode	10	08	B510				IGN:2			Unknown always 00 00	TempSalidaObjetivo		temp1				TempACSObjetivo		temp1						IGN:2			Unknown always ff ff	Flags0		UCH						IGN:2			Unknown always ff 00						
																																																	
*r				10	08	B511																																											
r		ReadModeSCC	Operation Mode				00	Temperaturasalida		temp				TemperaturaentradaACS		D1B			Puede ser la temperatura del agua de entrada?	Tempcircuitoagua		temp1			Va de 17-70 y 95-100	Flags1		UCH				Flags2		UCH						IGN:1				Flags3		UCH			
																																																	
*uw				10	08	B512																																											
uw		StatusCirPumpSCC	Status circulation pump				00			UCH	0=off;100=on																																						
# Current errors																																																	
*r				10	08	B503	00																																										
r		currenterror02	Current errors 02				02			errors																																							
r		currenterror03	Current errors 03				03			errors																																							
r		currenterror04	Current errors 04				04			errors

wimleers, 2023-02-19:
I have a vSMART too! In combination with a Vaillant ecoTEC pro VCW 286/5-3 A (which AFAICT is functionally identical to and just more powerful than a VCW 226/5-3 A).

I'm running `ebusd` using Docker and am using https://lectronz.com/products/ebus-to-wifi-adapter-module.

My output:
```
 a d m i n @ m i s t   e b u s d - d o c k e r   %   d o c k e r   c o m p o s e   u p 
 [ + ]   R u n n i n g   1 / 0 
  (?   C o n t a i n e r   e b u s d     R e c r e a t e d                                                                                                                                                                                                                                               0 . 1 s 
 A t t a c h i n g   t o   e b u s d 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 1 8 . 8 4 7   [ m a i n   e r r o r ]   i n v a l i d / u n k n o w n   a r g u m e n t   i n   e n v   ( i g n o r e d ) :   r e a d o n l y 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 1 8 . 9 4 2   [ m a i n   n o t i c e ]   e b u s d   2 3 . 1 . 2 3 . 1   s t a r t e d   w i t h   a u t o   s c a n   o n   d e v i c e   e s p - e b u s . l o c a l : 3 3 3 3 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 1 9 . 8 3 8   [ b u s   n o t i c e ]   b u s   s t a r t e d   w i t h   o w n   a d d r e s s   3 1 / 3 6 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 1 9 . 8 3 8   [ b u s   n o t i c e ]   s i g n a l   a c q u i r e d 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 2 0 . 8 0 6   [ b u s   n o t i c e ]   n e w   m a s t e r   1 0 ,   m a s t e r   c o u n t   2 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 2 0 . 8 3 6   [ b u s   n o t i c e ]   n e w   m a s t e r   0 3 ,   m a s t e r   c o u n t   3 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 2 0 . 8 3 6   [ u p d a t e   n o t i c e ]   r e c e i v e d   u n k n o w n   M S   c m d :   1 0 0 8 b 5 1 0 0 9 0 0 0 0 3 0 f f f f f f 0 0 f f 0 0   /   0 1 0 1 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 2 7 . 3 3 0   [ u p d a t e   n o t i c e ]   r e c e i v e d   u n k n o w n   M S   c m d :   1 0 0 8 b 5 1 1 0 1 0 0   /   0 8 6 c 0 1 1 0 0 0 0 0 0 0 0 0 0 1 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 2 7 . 6 3 2   [ u p d a t e   n o t i c e ]   r e c e i v e d   u n k n o w n   M S   c m d :   1 0 0 8 b 5 1 1 0 1 0 1   /   0 9 2 d 2 e 0 0 8 0 f f 3 0 0 0 0 0 f f 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 0 1 1   [ b u s   n o t i c e ]   s c a n   0 8 :   ; V a i l l a n t ; B A I 0 0 ; 0 2 0 4 ; 9 6 0 2 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 0 1 1   [ u p d a t e   n o t i c e ]   s t o r e   0 8   i d e n t :   d o n e 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 0 1 1   [ u p d a t e   n o t i c e ]   s e n t   s c a n - r e a d   s c a n . 0 8     Q Q = 3 1 :   V a i l l a n t ; B A I 0 0 ; 0 2 0 4 ; 9 6 0 2 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 0 1 1   [ b u s   n o t i c e ]   s c a n   0 8 :   ; V a i l l a n t ; B A I 0 0 ; 0 2 0 4 ; 9 6 0 2 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 5 2 6   [ m a i n   n o t i c e ]   r e a d   c o m m o n   c o n f i g   f i l e   v a i l l a n t / b r o a d c a s t . c s v 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 6 6 3   [ m a i n   n o t i c e ]   r e a d   c o m m o n   c o n f i g   f i l e   v a i l l a n t / s c a n . c s v 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 7 9 3   [ m a i n   n o t i c e ]   r e a d   c o m m o n   c o n f i g   f i l e   v a i l l a n t / g e n e r a l . c s v 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 9 0 7   [ u p d a t e   n o t i c e ]   r e c e i v e d   u n k n o w n   M S   c m d :   1 0 0 8 b 5 1 0 0 9 0 0 0 0 3 0 f f f f f f 0 0 f f 0 0   /   0 1 0 1 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 0 . 9 1 8   [ m a i n   n o t i c e ]   r e a d   s c a n   c o n f i g   f i l e   v a i l l a n t / 0 8 . b a i . c s v   f o r   I D   " b a i 0 0 " ,   S W 0 2 0 4 ,   H W 9 6 0 2 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 1 . 1 0 5   [ u p d a t e   n o t i c e ]   s e n t   s c a n - r e a d   s c a n . 0 8   i d   Q Q = 3 1 :   
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 1 . 2 7 0   [ u p d a t e   n o t i c e ]   s e n t   s c a n - r e a d   s c a n . 0 8   i d   Q Q = 3 1 :   
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 1 . 4 3 9   [ u p d a t e   n o t i c e ]   s e n t   s c a n - r e a d   s c a n . 0 8   i d   Q Q = 3 1 :   
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 1 . 6 0 7   [ u p d a t e   n o t i c e ]   s e n t   s c a n - r e a d   s c a n . 0 8   i d   Q Q = 3 1 :   2 1 ; 1 7 ; 3 9 ; 0 0 1 0 0 2 1 8 4 9 ; 1 3 0 0 ; 3 4 6 0 2 0 ; N 4 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 2 . 0 4 3   [ m a i n   n o t i c e ]   f o u n d   m e s s a g e s :   2 0 2   ( 2   c o n d i t i o n a l   o n   2 7   c o n d i t i o n s ,   0   p o l l ,   1 0   u p d a t e ) 
 e b u s d     |   2 0 2 3 - 0 2 - 1 8   1 5 : 5 4 : 3 2 . 2 0 8   [ u p d a t e   n o t i c e ]   s e n t   s c a n - r e a d   s c a n . 0 8   i d   Q Q = 3 1 :   2 1 ; 1 7 ; 3 9 ; 0 0 1 0 0 2 1 8 4 9 ; 1 3 0 0 ; 3 4 6 0 2 0 ; N 4 
[... cut ...]
  (?   C o n t a i n e r   e b u s d     S t o p p e d                                                                                                                                                                                                                                                   2 . 6 s 
 c a n c e l e d 
```

AFAICT making `ebusd` discover the config file you posted here should change all those `received unknown MS cmd` lines into actual information, right?

I'll try to get your CSV loaded now … (the UX of `ebusd` is rather unique 😅).

Would me enabling raw logging help you confirm things?

wimleers, 2023-02-20:
## Using `15.scc.csv` as provided by @gvisol 

I did:
1. `cd /Volumes/dev/ebusd`
2. `git clone git@github.com:john30/ebusd-configuration.git`
3. `cd ebusd-configuration/ebusd-2.1.x/en/vaillant`
4. Copy the CSV provided by @gvisol and paste it in the appropriate file: `pbpaste > 15.scc.csv— ` as instructed above
5. Instructed the `ebusd` docker container to know about this path by adding to `docker-compose.yaml`:
```
    volumes:
      - /Volumes/dev/ebusd/ebusd-configuration/ebusd-2.1.x/en:/etc/ebusd-configuration
```
6. And then using it:
```
      EBUSD_CONFIGPATH: /etc/ebusd-configuration
```
7. `docker compose up`

## Before
```
ebusd  | 2023-02-18 15:54:35.569 [bus notice] scan 15: ;21;20;32;0020197223;0932;016140;N2
ebusd  | 2023-02-18 15:54:35.707 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "scc00", SW0500, HW3903
ebusd  | 2023-02-18 15:54:35.707 [main error] scan config 15: ERR: element not found
```

## After
```
busd  | 2023-02-20 20:20:05.948 [bus notice] scan 15: ;Vaillant;SCC00;0500;3903
ebusd  | 2023-02-20 20:20:05.948 [update notice] store 15 ident: done
ebusd  | 2023-02-20 20:20:05.948 [update notice] sent scan-read scan.15  QQ=31: Vaillant;SCC00;0500;3903
ebusd  | 2023-02-20 20:20:05.948 [bus notice] scan 15: ;Vaillant;SCC00;0500;3903
ebusd  | 2023-02-20 20:20:06.118 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231323033323030
ebusd  | 2023-02-20 20:20:06.284 [update notice] sent scan-read scan.15 id QQ=31: 
ebusd  | 2023-02-20 20:20:06.454 [update notice] sent scan-read scan.15 id QQ=31: 
ebusd  | 2023-02-20 20:20:06.623 [update notice] sent scan-read scan.15 id QQ=31: 21;20;32;0020197223;0932;016140;N2
ebusd  | 2023-02-20 20:20:06.623 [bus notice] scan 15: ;21;20;32;0020197223;0932;016140;N2
ebusd  | 2023-02-20 20:20:06.680 [main error] error reading scan config file vaillant/15.scc.csv for ID "scc00", SW0500, HW3903: ERR: invalid argument, vaillant/15.scc.csv:14: ERR: invalid argument, missing field map
ebusd  | 2023-02-20 20:20:06.680 [main error] scan config 15: ERR: invalid argument
ebusd  | 2023-02-20 20:20:06.983 [update notice] received update-write bai SetMode QQ=10: auto;25.0;-;-;0;0;0;0;0;0
```

👆 Your `15.scc.csv` seems to have loaded, but it's still complaining. Perhaps I have a different vSMART model. I'm in Belgium — I have the impression Vaillant does slightly different products for each m

P.S.: pointed https://github.com/john30/ebusd/discussions/832 to this issue.

wimleers, 2023-02-20:
BTW, https://github.com/john30/ebusd/discussions/781 really helps connect the different parts of `ebusd`.

tim-devel, 2023-02-20:
Hi, thanks both for sharing this.  I have tried your CSV but I am getting the following error:

2023-02-20 20:34:02.641 [main error] error reading scan config file vaillant/15.scc.csv for ID "scc00", SW0500, HW3903: ERR: invalid argument, vaillant/15.scc.csv:14: ERR: invalid argument, missing field map

tim-devel, 2023-02-26:
@gvisol I still can't get this to work.  Can you upload as a pull request so we can test and comment please

gvisol, 2023-03-02:
> @gvisol I still can't get this to work. Can you upload as a pull request so we can test and comment please

Sure!
In fact the version running at home is newer than the one I've uploaded here. Allow me to have spare time and I'll update my progress.

tim-devel, 2023-04-02:
> > @gvisol I still can't get this to work. Can you upload as a pull request so we can test and comment please
> 
> Sure! In fact the version running at home is newer than the one I've uploaded here. Allow me to have spare time and I'll update my progress.

Hi, are you still planning to upload your changes?  I am excited to try them out

tim-devel, 2023-05-01:
> > @gvisol I still can't get this to work. Can you upload as a pull request so we can test and comment please
> 
> Sure! In fact the version running at home is newer than the one I've uploaded here. Allow me to have spare time and I'll update my progress.

@gvisol sorry to chase but are you still planning to upload your config as a pull request so we can test?

JensVanhooydonck, 2023-07-24:
> > @gvisol I still can't get this to work. Can you upload as a pull request so we can test and comment please
> 
> Sure! In fact the version running at home is newer than the one I've uploaded here. Allow me to have spare time and I'll update my progress.

Could you share your config file? Love to try this out.

iangregory, 2024-11-15:
@gvisol did you have any success identifying the values?

I believe `temp_1` is the flow temperature, and `temp_3` is the % modulation requested (though, I notice on your chart some negative values, I don't see them with my setup).

JensVanhooydonck, 2025-05-05:
Somebody who knows if it is possible to set the target temperature with the vsmart over ebusd ?