john30/ebusd-configuration #315: Decoded "vwzio" Hydraulikstation VWZ MEH 97/6

cyberthom42, 2023-02-13:
I decoded values for the VWZ MEH 97/6. I was mainly interested in the parameters for the heating rod (Heizstab) which can not be read and changed via 08.hmu. A separate csv for the VWZ is required. I created a csv, I will initiate the pull request (I need to find out how, it will be my first one) as soon I tested all the parameter. I am not sure id the includes are needed...
Here is a short draft  in case you want to give it a try, please comment your results:

```
##type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s)_1,datatypes/templates_2,divider/values_3,unit_4,comment_5,field3,part (m/s)_6,datatypes/templates_7,divider/values_8,unit_9,comment_10,field4,part (m/s)_11,datatypes/templates_12,divider/values_13,unit_14,comment_15,Column1

*r,,,,,,"B511",,,,,,,,,,,,,,,,,,,,,,,,,
#r,,State,,,,,07,energy,,UCH,,,,,,energy,,kWh,,onoff,,UCH,0=off;1=on,,,state,,UCH,0x01=ready;0x0b=error;0x09=heating;0x11=cooling;0x81=heating_water,,
r,,State,,,,,07,field1,,UCH,,,,field2,,UIN,,,,field3,,UCH,,,,field4,,UCH,,,,field5,,UIN,,,,field6,,UCH,,,

# B5 1A 32 Heatpump statistics Live Monitor,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,"B51A","05FF32",,,IGN:3,,,,,,,,,,,,,,,,,,,,,
r,,FlowPressure,,,,,3d,,,D2C,4,,,,,,,,,,,,,,,,,,,,
r,,HeizstabBetriebsStd,,,,,46,,,energy,,,,,,,,,,,,,,,,,,,,,
r,,HeizstabStromverbrGesamt,,,,,49,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,HeizstabSchaltvorg,,,,,4a,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,SchaltvorgängeVUV_WW,,,,,4b,,,UIN,,,,,,,,,,,,,,,,,,,,,
r,,Heizstab_Leistung,,,,,4c,,,D1B,10,kW,in kW,,,,,,,,,,,,,,,,,,
r,,HeizstabTemp,,,,,5c,,,D2C,,,,,,,,,,,,,,,,,,,,,

# B5 1A 33 Fachhandwerker Konfiguration,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,"B51A","05FF33",,,IGN:3,,,,,,,,,,,,,,,,,,,,,
*w,,,,,,"B51A","06FF33",,,,,,,,,,,,,,,,,,,,,,,,
r;w,,HeizstabLeistungsgrenze,,,,,1a,,,UIN,,kW,,,,,,,,,,,,,,,,,,,

# B5 1A 34 Fachhandwerker Konfiguration,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
#*r,,,,,,"B51A","05FF34",,,IGN:3,,,,,,,,,,,,,,,,,,,,,
#*w,,,,,,"B51A","06FF34",,,,,,,,,,,,,,,,,,,,,,,,
#r;w,,hwcBetriebsart,,,,,44,,,UCH,,,,,,,,,,,,,,,,,,,,,

# B5 14 
*r,,,,,,B514,,,,IGN:2,,,,,,,,,,,,,,,,,,,,,,
r,,HE_Vorlauftemperatur_Heizstab,T.1.130,,,,058203ffff,,,UIN,10,tempv,Vorlauftemperatur Heizstab,,,,,,,,,,,,,,,,,,,

!include,hcmode.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
!include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

satbeppo, 2023-02-25:
Great, thanks for your work.
but your Link above points to "https://github.com/john30/ebusd-configuration/issues/url"
I own a VWL75/6 too, how to integrate into my Ebusd config?

OleWissing, 2023-10-01:
> Great, thanks for your work. but your Link above points to "https://github.com/john30/ebusd-configuration/issues/url" I own a VWL75/6 too, how to integrate into my Ebusd config?

Me too!! Did you get it to work???

cyberthom42, 2024-01-16:
> Great, thanks for your work. but your Link above points to "https://github.com/john30/ebusd-configuration/issues/url" I own a VWL75/6 too, how to integrate into my Ebusd config?

Fixed :-)