john30/ebusd-configuration #319: WaterPressure and HeatingSwitch parameters are not requested

MityaDVG, 2023-02-17:
My parameters info:
version: ebusd 23.1.p20230106
update check: revision 23.1 available
device: tcp:192.168.2.34:9999
access: *
signal: acquired
symbol rate: 23
max symbol rate: 127
min arbitration micros: 5
max arbitration micros: 3146
min symbol latency: 0
max symbol latency: 82
reconnects: 0
masters: 2
messages: 56
conditional: 3
poll: 10
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0104;HW=7803", loaded "vaillant/bai.0010021961.inc" ([PROD='0010021963']), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

The HeatingSwitch and WaterPressure parameters are not requested when scanning from the bai.0010021961.inc file, although they are there. If I request commands via TCP requests for HeatingSwitch "read -s 31 -h 08b509030d f203" response is 0101 and WaterPressure "read -s 31 -h 08b509030d 0200" response is 03730a00. These responses appear in the addon logs and are recorded in the MQTT (I look at the MQTT explorer), but do not appear in the MQTT integration.

Kolbi, 2026-01-30:
I can confirm that manual call

`ebusctl read -s 31 -h 08b509030d 0200`

 is also working manually for me. But they are not automatically requested.
@john30 any idea why?

pepsonEL, 2026-01-30:
Please tell me how you change it ? and what you change(give me your line)
I use ebusd ad addon in home assistant and i also search solution how change it.

Kolbi, 2026-01-30:
I'm using HA in a docker instance and even ebusd - i have no experience with addon :)

My line looks like:
`filter-name = status|temp|humidity|yield|count|energy|power|runtime|hours|starts|mode|curve|^load$|^party$|sensor|timer|pressure`

Maybe you can install vscode as an addon and then you need to find the path to the ebusd addon and /etc/ebusd/mqtt-hassio.cfg