john30/ebusd-configuration #325: unable to load scan config

SzosszeNET, 2023-02-24:
Hi there,

I'm suspecting it's missing CSV's somewhere but not fully understaning some of the issues. Sure I can see there is no file starting with 76, but I see 05, 08 and 15 yet the latest container seems to be complaining about them too. Happy to try to figure it out but would need some pointers pleaes if possible.

Kind regards,

[ebusd1.csv](https://github.com/john30/ebusd-configuration/files/10819981/ebusd1.csv)

SzosszeNET, 2023-02-24:
I found just this - not sure if helps

ebusctl info
version: ebusd 23.1.23.1
update check: OK
device: esp-ebus.majoros.me:3333
signal: acquired
symbol rate: 98
max symbol rate: 108
min arbitration micros: 3
max arbitration micros: 4
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 4
messages: 3
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

SzosszeNET, 2023-02-24:
Okay, did seem to make some progress using the CTLV2 fix, however it seems that ebusd makes the VR921 inaccessible (it stops providing data and loose the ability to control the installation), also on the controller (assume the VWZIO?) I see "communication fault". Any thoughts, pointers please?

closed 2023-03-15 by SzosszeNET

JanDetrez, 2023-03-23:
I had similar communication issues when I connected my wifi eBUS adapter. It pulled too much current from the eBUS network, causing it to break down. I solved it by providing 24V on the VCC and GND pins on my adapter. Alternatively a USB power adapter could be used. This is my adapter https://lectronz.com/products/ebus-to-wifi-adapter-module