john30/ebusd-configuration #335: Vaillant: scan.76 id =  (ERR: invalid position for 3176b5090124 / 00)

teclab-at, 2023-05-03:
Hi there,

I bought https://adapter.ebusd.eu/v31/ and now testing.

Calling ebusctl find gives me this:

```
…
…
omu Timer.Thursday = no data stored
omu Timer.Tuesday = no data stored
omu Timer.Wednesday = no data stored
omu ZustandOAWLogic = no data stored
scan id = no data stored
scan.08  = Vaillant;HMU00;0308;0403
scan.08 id = 21;17;15;0010016422;0001;005417;N3
scan.15  = Vaillant;70000;0419;4603
scan.15 id = 21;17;17;0020218357;0082;027150;N1
scan.76  = Vaillant;VWZ00;0308;0403
scan.76 id =  (ERR: invalid position for 3176b5090124 / 00)
scan.e0  = Vaillant;OMU00;0131;5202
scan.e0 id = 21;17;17;0010016715;0001;006077;N7
```
Should I care on this error? Because it seems to me that I am missing some values on the switching cycle which I can’t find listed.
Log file is attached.
[ebusd.log](https://github.com/john30/ebusd-configuration/files/11391083/ebusd.log)

thx for support!

wimleers, 2023-05-11:
What Vaillant devices do you have?

`http://mist.local:8080/data/scan.76` returns this for me:
```
{
  "scan.76": {
    "messages": {
      "": {
        "name": "",
        "passive": false,
        "write": false,
        "lastup": 1683663282,
        "zz": 118,
        "fields": {
          "MF": {
            "value": "Vaillant"
          },
          "ID": {
            "value": "VWZ00"
          },
          "SW": {
            "value": "0522"
          },
          "HW": {
            "value": "5103"
          }
        }
      },
      "id": {
        "name": "id",
        "passive": false,
        "write": false,
        "lastup": 1683763260,
        "zz": 118,
        "fields": {
          "prefix": {
            "value": "21"
          },
          "year": {
            "value": "22"
          },
          "week": {
            "value": "33"
          },
          "product": {
            "value": "0010022070"
          },
          "supplier": {
            "value": "3100"
          },
          "counter": {
            "value": "006482"
          },
          "suffix": {
            "value": "N8"
          }
        }
      }
    }
  },
  "global": {
…
  }
}
```

teclab-at, 2023-05-11:
aroCOLLECT VWL 11/4 SA
flexoTHERM VWF 117/4
multiMATIC 700 VRC 700

don't know how you did this http://mist.local:8080/data/scan.76 ... I do not have an http server enabled in ebusd ...

thx

wimleers, 2023-05-16:
Okay so `Vaillant;VWZ00;0308;0403` would seem to be how a flexoTHERM's controller is identified versus `Vaillant;VWZ00;0522;5103` for a aroTHERM split uniTOWER's controller.

smartboart, 2023-07-09:
Was anybody able to read and control vaillant Arotherm Plus with sensocomfort ?

mrgrlscz, 2023-07-12:
hello, also have this in my logs: `[bus notice] scan 76: ;Vaillant;VWZ00;0522;5103`
looking for a way to fetch data from it

my config (FRANCE=SaunierDuval, VAILLANT Group):
HeatPump GeniaSet Split (https://www.saunierduval.fr/particulier/nos-produits/geniaset-split-45440.html)
MiSet Thermostat (SRT 380?) (got it on ebusd by renaming "15.700.csv" to "15.bass2.csv")

wimleers, 2023-08-14:
Please join us at https://github.com/john30/ebusd-configuration/pull/330.