john30/ebusd-configuration #342: Some BC messages not parsed

vadito, 2023-07-09:
Hi all,
I got my v5 adapter this week and started working on getting my Ariston furnace up and running. I'm using a csv found on github for a similar(but not the same) Ariston device as a base.
For most parts my modified CSV works fine but I'm struggling with parsing some BC messages which I can see in the ebusd log but are claimed to be unknown.
I did some tests with longer/shorter messages, but multiple 8B longs simply don't work for me. What am I missing?

```
Message in ebus log:

[update notice] received unknown BC cmd: 37fe200e086271b40064002c01
[update notice] received unknown BC cmd: 37fe200e086126e00190018a02

Definitions in CSV:

r,energymgr,z1_day_temp_bcast,Z1 Day temperature,,fe,200e,6271,,s,SIN,10,°C
r,energymgr,dhw_comfort_temp,DHW Comfort Temp,,fe,200e,6126,dhw_comfort_temp,,SIN,10,°C
```

In the same CSV these ones work fine
```
r,energymgr,dhw_comfort_mode,DHW Comfort Mode,,fe,200e,c028,dhw_comfort_mode,,BCD,0=off;1=timed;2=always_on
r,energymgr,dhw_thermal_cleansing_mode,DHW Thermal Cleansing Function,,fe,200e,0b20,,m,onoff
```

I did some additional test definitions, these random 7 and 12B work

```
2023-07-09 17:26:44.541 [update info] received BC cmd: 37fe207107003b171a061700
2023-07-09 17:26:44.541 [update notice] received unknown BC cmd: 37fe207107003b171a061700

$ebusctl define -r 'r,energymgr2,match_test,Match test,,fe,2071,003b,match_test,s,SIN,10,C'`

2023-07-09 17:36:44.992 [update info] received BC cmd: 37fe207107003b171a061700
2023-07-09 17:36:44.992 [update notice] received read energymgr2 match_test QQ=37: 667.9

```

```
2023-07-09 17:44:49.854 [update info] received BC cmd: 37fe20100c0c1900000d1900000e190000
2023-07-09 17:44:49.854 [update notice] received unknown BC cmd: 37fe20100c0c1900000d1900000e190000

$ebusctl define -r 'r,energymgr2,match_test2,Match test2,,fe,2010,0c19,match_test2,s,SIN,10,C'

2023-07-09 17:45:29.961 [update info] received BC cmd: 37fe20100c0c1900000d1900000e190000
2023-07-09 17:45:29.961 [update notice] received read energymgr2 match_test2 QQ=37: 0.0
```

This test definition for 8B doesn't parse:

```
2023-07-09 17:48:50.708 [update info] received BC cmd: 37fe2010080591000006910000
2023-07-09 17:48:50.708 [update notice] received unknown BC cmd: 37fe2010080591000006910000

$ebusctl define -r 'r,energymgr3,match_test3,Match test3,,fe,2010,0591,match_test3,s,SIN,10,C'

2023-07-09 17:49:35.970 [update info] received BC cmd: 37fe2010080591000006910000
2023-07-09 17:49:35.970 [update notice] received unknown BC cmd: 37fe2010080591000006910000
```

wrongisthenewright, 2024-03-22:
It's a problem I already noticed on Bridgenet protocol. I haven't found a reason. I think it's related to how the message is sent on the bus from some device (maybe incorrect CRC??) 
You can try to open an issue on the Ebusd GitHub repo.
If you need I can add my experience.

DaeMonSxy, 2025-02-08:
**vadito: may I request your ariston.csv?**
Iam struggling to get AristonGenusOne working as expected( for example not getting consumptions at all)
thanks

vadito, 2025-02-12:
I've attached it, although consumption is not something I looked into.
[ariston.csv](https://github.com/user-attachments/files/18762107/ariston.csv)