john30/ebusd-configuration #347: Vaillant VRC 710 missing energy consumption data

GLSSoftware, 2023-08-03:
Hello John,

I bought the eBUSd adapter v5 to monitor the energy consumption of my vaillant heater. I don't need internal control data for my project, only energy consumtion.  

I am using the following configuration:
- eBUS Adaptger v5
- Vaillant EcoTEC VC 20
- Controller VRC 710 (sensoDirect)
- Home Assistant via MQTT

The heater is recognized as
`scan 08: ;Vaillant;BAI00;0807;7603`

The adapter and the communication with Home Assistant ran right away on the first try. Many thanks to the entire team for the great work!

I get a lot of data like FlowTemp, HcStarts, ValveMode etc.
Unfortunately, all data on energy consumption are missing. But I can call up all the needed data directly on the heater's touch display. This means to me that the data must be available somewhere deep in the system. 
The log in HA shows the following errors:

```
[mqtt error] decode bai PowerValue: ERR: invalid position
[update error] unable to parse poll-read bai PrAPSCounter from 3108b509030df200 / 00: ERR: invalid position
[mqtt error] decode bai PrAPSCounter: ERR: invalid position
[update error] unable to parse poll-read bai PrEnergyCountHc1 from 3108b509030df600 / 00: ERR: invalid position
[main error] scan config 05: ERR: read timeout
[mqtt error] decode bai PrEnergyCountHc1: ERR: invalid position
[update notice] received unknown BC cmd: 10feb5550406010001
[update notice] received unknown MS cmd: 1008b5550100 / 0103
[update error] unable to parse poll-read bai PrEnergyCountHc2 from 3108b509030df800 / 00: ERR: invalid position
```

I assume that either the formatting of the data is wrong or that the data isn't in the expected variables but in some other ebus values.

Question: what can I do to get the energy data?
What kind of information can I provide the team with in order to adjust the config files if necessary?

kind regards
Gerold

john30, 2023-08-15:
you could check for unknown messages in the logging, but it is rather unlikely that the emm will push them to the bus without reason (i.e. someone requesting it)

of course looking at other 15.*.csv files and especially the energy values in there might give a hint to where the value can be read. you can then try by using `define "..."` and a subsequent read, or use the `hex` cmd directly.

another approach is using the [readallvaillantregisters.sh](https://github.com/john30/ebusd/blob/master/contrib/scripts/readallvaillantregisters.sh) script on the address 15 lets say today, store the output to a text file, do the same again after at least a day and at least some measurable energy was consumed (as counters sometimes are only updated once per day), and then compare the two.

laserrichi, 2023-10-04:
Has anyone here made any further progress? I also have the same situation
i have wrote here, but I can't get any further:
https://github.com/john30/ebusd/discussions/980

about the readallvaillantregisters.sh script... 
i become 0 = ERR: command not enabled
where do I have to copy it and run it? Does this then call the ebusctl?

Anaximelis, 2023-10-18:
> Has anyone here made any further progress? I also have the same situation i have wrote here, but I can't get any further: [john30/ebusd#980](https://github.com/john30/ebusd/discussions/980)
> 
> about the readallvaillantregisters.sh script... i become 0 = ERR: command not enabled where do I have to copy it and run it? Does this then call the ebusctl?

In order to run the script, [enablehex ](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#hex)must be activated on your ebusd.

laserrichi, 2023-10-18:
Ok thanks nuw it runs

Anaximelis, 2023-10-18:
@laserrichi do you have more traffic on your VRC 710 bus than the three standard messages? I had dealt with the matter, similar to you. Am then but come from it, because nothing is to log with.

laserrichi, 2023-11-07:
i have it here post: https://github.com/john30/ebusd/discussions/980